networkupstools/nut #1649: Cross-platform build tool/system that works on Windows

Anutrix, 2022-09-04:
Has there ever been thought of moving to cross-platform build tool like scons, gradle, ant, etc. instead of the sh script that's currently used.

Is there already a ticket for this?

jimklimov, 2022-09-04:
I suppose gradle and ant (and maven) are java-oriented and so care little for C/C++ builds (and their pkgconfig files, or legacy scripts, or autodetection fallbacks). Not sure about scons, didn't use it.

Maybe cmake could cut it, but would be quite an endeavour and not sure that the dozens of projects that NUT relies on would be esily wrangled to play along.

In fact the core system used in NUT builds is the venerable GNU autotools/autoconf which is cross-platform for decades. And it got NUT all around the server and (POSIX) desktop and embedded world with whatever C and Make implementations they have. Windows was always an outlier of sorts, and just recently came closer into the fold at last. With the likes of mingw, msys, cygwin, git-bash etc. shell scripts are not too unwieldy on Windows either. If some build scenarios (e.g. VS tooling) would require an alternative set of recipes, someone would have to maintain them in sync with the true cross-platform builds. If there is a system that works everywhere (likely cmake of what I hear) - Linux, Solarish systems, BSDs, MacOS, maybe HPUX/AIX still used by some, so there is one set of recipes to maintain, and you endeavour to write something on par with current web of makefiles for libs and docs and programs, that could replace it wholesale - might be interesting :)

The `ci_build.sh` is recent syntactic sugar to avoid mile-long build option lists for typical dev/CI iterations in a way reproducible easily in a terminal on whatever OS, and to automate some CI scenarios like builds with different SSL and USB implementations.

Anutrix, 2022-09-04:
> Keep in mind that NUT tries to remain usable on systems still (and already) running, so roughly anything built this millenium (OS version, CPU arch and bitness wise) - if it built/ran NUT before, it should still with new releases. C99, weirdness of embedded, and all that...

Cool. Makes sense.
Is there a list of platforms we support officially? If not we should make one. Contributors can add more platforms/OSs they've used and/or tested on.

This might be harder to generalize given how many distributions/flavors of Linux exist but might be simpler/must for Windows platforms.
Windows has quite a reputation for breaking compatibility at low level OS APIs(not sure if much of it will affect this project).
Windows NT 5.1(Windows XP) and newer should be our goal for now IMHO. For older Windows stuff if needed, we can slowly improve compatibility over time.

This will help us decide on choosing an alternative build system or any other Windows tool/library if and when needed in the future.