networkupstools/nut #1665: UPS Samurai SMART 1500 LCD

MatejKovacic, 2022-09-18:
Hi, I have **UPS Samurai SMART 1500 LCD** connected to Linux ARM device (OrangePi running Armbian). It seems some generic UPS device. `nut-scanner -U` says:

```
Scanning USB bus.
[nutdev1]
    driver = "nutdrv_atcl_usb"
    port = "auto"
    vendor = "MEC"
```

If I go to `/etc/nut/ups.conf` and use `nutdrv_atcl_usb` driver, UPS is not detected (NUT says not available). The same is with `nutdrv_qx` driver. `upsc samurai@localhost` says `Error: Driver not connected`.

`lsusb` says:

```
  idVendor           0x0001 Fry's Electronics
  idProduct          0x0000
  bcdDevice            1.00
  iManufacturer           1 MEC
  iProduct                2 MEC0003
```

In my `/etc/nut/ups.conf` I have: 

```
pollinterval = 1
maxretry = 3

[samurai]
    driver = nutdrv_atcl_usb
    port = auto
    desc= "UPS Samurai SMART 1500 LCD"
    vendorid = 0001
    productid = 0000
    product = MEC0003
    vendor = MEC
    bus = 007
```

Any help?

System:
- OS: Armbian 22.05.3 Bullseye with Linux 5.15.48-sunxi
- Network UPS Tools - UPS driver controller 2.7.4
- Network UPS Tools - 'ATCL FOR UPS' USB driver 1.1 (2.7.4)

jimklimov, 2022-09-18:
No-names are hard to guess. For this "very non-random yet arbitrary" chip ID, a frequent fit might be nutdrv_qx (and then pick subdrivers).

If you can, try a custom build of modern NUT (e.g. from github).

MatejKovacic, 2022-09-19:
OK, in `/etc/nut/ups.conf` I put this:

```
[samurai]
    driver = megatec
    port=auto
    desc = "UPS Samurai SMART 1500 LCD"
```
However, when I say `sudo upsdrvctl start` I get this:
```
Network UPS Tools - UPS driver controller 2.7.4
Can't start /lib/nut/megatec: No such file or directory
```

MatejKovacic, 2022-09-19:
I tried sometning else, since I found mention of UPSilon 2000 software in https://networkupstools.org/stable-hcl.html. In `/etc/nut/ups.conf` I put this:
```
[samurai]
driver = blazer_usb 
port = auto
langid_fix= 0x409 
vendorid = 0001
productid = 0000
desc = "UPS Samurai SMART 1500 LCD"
```

Now I run `sudo upsdrvctl start`:

```
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Megatec/Q1 protocol USB driver 0.12 (2.7.4)
Supported UPS detected with megatec protocol
Vendor information unavailable
No values provided for battery high/low voltages in ups.conf

Using 'guestimation' (low: 20.800000, high: 26.000000)!
Battery runtime will not be calculated (runtimecal not set)
```

But nevertheless, I get this broadcast message from nut: `UPS samurai@localhost is unavailable`.

Any other idea?

jimklimov, 2022-09-19:
Try `driver = nutdrv_qx`

jimklimov, 2022-09-19:
Ok, so it seems the driver did recognize the device. Do you have the data server `upsd` (probably wrapped as `nut-server` service) running - and last started after you defined a `samurai` section?

MatejKovacic, 2022-09-19:
Thanks, I did this, just replaced `blazer_usb` with `nutdrv_qx` (the other options are the same, including `langid_fix`), and now I get this:

```
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
Using protocol: Q1 0.07
Can't autodetect number of battery packs [-1/27.50]
Battery runtime will not be calculated (runtimecal not set)
```

However, I still get this message `UPS samurai@localhost is unavailable`.

P. S. I really appreciate you are helping me with this guys.

jimklimov, 2022-09-19:
The `langid_fix` depends on device (dialect of the protocol it implemented) so can't say - try with and without it, dump both `upsc` results into files, `diff` them...

As for `blazer*` - generally they are among several obsoleted drivers, a superset of which is `nutdrv_qx` going forward ...

MatejKovacic, 2022-09-21:
OK, I did this (removed `langid_fix`, but using `nutdrv_qx`). And there is no difference in the output of `upsc samurai@localhost`. So I guess, i do not need this `langid_fix`.

Anyway, when I restarted everything, I got the "standard" message `Can't autodetect number of battery packs [-1/27.50]. Battery runtime will not be calculated (runtimecal not set)`, but I guess that is expected.

Now I have some other question... :) 

I write `upscmd -l samurai` and get this:
```
Instant commands supported on UPS [samurai]:

beeper.toggle - Toggle the UPS beeper
load.off - Turn off the load immediately
load.on - Turn on the load immediately
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start - Start a battery test
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```
Now, I try `upscmd samurai test.battery.start.quick` and get this:
```
Username (matej): monuser
Password: 
Unexpected response from upsd: ERR ACCESS-DENIED
```
I am using username and password from the file `/etc/nut/upsd.users`:
```
[monuser]
  password = mypassword
  admin master
```

Oh, and just one additional question. I am not getting battery percentage, just `battery.voltage`variable (currently 27.50 V). So I guess, I need to compute (estimate) battery percentage or running time by hand?

jimklimov, 2022-09-21:
Regarding `upsd.users`, there are two types of entities defined there: accounts permitted to use certain (or all) "actions" and/or "instcmds", and accounts used for monitoring (with one or another `upsmon` role). See for example in https://github.com/networkupstools/nut/blob/master/conf/upsd.users.sample 
I am puzzled how you got `admin master` there for example - random trying? :)

In your case, you are trying to invoke an `instcmd` with an account not explicitly permitted to use those.

As for the "standard message", per issue #1279 discussion and analysis (and PR #1652 attempting to address it - testing welcome) it seems that the auto-detection of battery packs amount relied on a certain voltage portion range around the detected (or assumed) nominal. Probably it considers 27.50V to be far outside the `'guestimation' (low: 20.800000, high: 26.000000)` range, so it is not deemed as one pack; similar counts for other numbers also do not fit.

A known (guessed, reported, overridden) battery pack count ("1" in your case) along with `battery.voltage`, in absence of a native `battery.charge` should cause the driver to calculate a guess. If you also request the `runtimecal` option, it should try to guess the remaining runtime based on load and speed of discharge.

See also https://networkupstools.org/docs/man/nutdrv_qx.html (or https://networkupstools.org/historic/v2.7.4/docs/man/nutdrv_qx.html for your older package) about the tweakables that you may need to set, possibly measure for your device to do so.

MatejKovacic, 2022-09-22:
Well, I was (blindly) following some guide on the internet... :) Anyway, I really appreciate your help. Now I can stop beeping, run some battery tests, etc. I will try to play with driver options. I also noticed, that UPSilon 2000 is showing the temperature - this is something I am not getting anywhere. UPSilon is also showing battery percentage, but not battery voltage, so I guess they are estimating percentage from that info also.

I wrote a small article about that on my blog (it is in Slovenian language, but you can use Google Translate): https://telefoncek.si/2022/09/2022-09-21-priklop-ups-na-homeassistant/ and also send it to the company which is producing (I mean rebranding) this device. With my comment that the openness sometimes costs nothing, on the contrary, it can even bring a competitive advantage to the company. We will see if something changes... :)

jimklimov, 2022-09-22:
Good to hear! For kicks, try running the driver with higher debug mode (up to 6 `-D` options to `nutdrv_qx`, with other copies of the driver stopped) so it would report the protocol lines it sees - usually there is obe with half a dozen floating-point numbers that the dialect subdriver then interprets into voltages, temps, etc.

See e.g. #1623 for a recent example.

And you can fiddle with `subdriver=...` option manually, to see if there is one better fit for your device than the first match that worked automatically.

jimklimov, 2022-09-22:
BTW in your article screenshots, temperature is 0*C (32*F) so I can't say it really reports that. Did you see (but not screenshot) other values too?

MatejKovacic, 2022-09-22:
> BTW in your article screenshots, temperature is 0_C (32_F) so I can't say it really reports that. Did you see (but not screenshot) other values too?

No, so it seems it does not really have support for that. It seems device is almost fully supported in NUT, just battery percentage is missing.

As I mentioned in article, there was just a small problem with HomeAssistant integration (entity name was autodetected as `sensor.samurai_input_line_frequency`, but after I changed that, everything started working.

MatejKovacic, 2022-09-27:
Update. Yesterday a seller called me and they said they really like my article. They asked me if they can send a link to the article to their other customers (of course!). And they also have other UPS models that support SNMP and they were asking me if NUT has a support for that (of course it has).

jimklimov, 2022-09-27:
Nice! Maybe they'll give one to you for a spin ;)

At least the IETF standard MIB should be supported for the basics. There are many others for advanced vendor-specific concepts like per-outlet management, metering, device daisy-chaining and whatnot.