networkupstools/nut #1668: `make install` installs files outside of prefix (leading to build failure)

fxcoudert, 2022-10-03:
Configuring nut 2.8.0 with:

```
./configure --disable-dependency-tracking --prefix=/home/linuxbrew/.linuxbrew/Cellar/nut/2.8.0 --localstatedir=/home/linuxbrew/.linuxbrew/var --sysconfdir=/home/linuxbrew/.linuxbrew/etc/nut --with-statepath=/home/linuxbrew/.linuxbrew/var/state/ups --with-pidpath=/home/linuxbrew/.linuxbrew/var/run --with-openssl --with-serial --with-usb --without-avahi --without-cgi --without-dev --without-doc --without-ipmi --without-libltdl --without-neon --without-nss --without-powerman --without-snmp --without-wrap --with-udev-dir=/home/linuxbrew/.linuxbrew/Cellar/nut/2.8.0/lib/udev
```

leads to `make install` trying to install files outside of the prefix:

```
   /usr/bin/mkdir -p '/usr/lib/tmpfiles.d'
   /usr/bin/install -c -m 644 nut-common.tmpfiles '/usr/lib/tmpfiles.d'
  /usr/bin/install: cannot create regular file '/usr/lib/tmpfiles.d/nut-common.tmpfiles': Permission denied
  make[3]: *** [Makefile:622: install-systemdtmpfilesDATA] Error 1
  make[3]: Leaving directory '/tmp/nut-20220922-12902-u3lle/nut-2.8.0/scripts/systemd'
  make[2]: *** [Makefile:693: install-am] Error 2
  make[2]: Leaving directory '/tmp/nut-20220922-12902-u3lle/nut-2.8.0/scripts/systemd'
  make[1]: *** [Makefile:502: install-recursive] Error 1
  make[1]: Leaving directory '/tmp/nut-20220922-12902-u3lle/nut-2.8.0/scripts'
  make: *** [Makefile:634: install-recursive] Error 1
```

This fails because `/usr/lib` is not user-writable. `make install` should not install files outside of the `--prefix` path passed to `configure`.

Seen in Homebrew CI: https://github.com/Homebrew/homebrew-core/pull/101035

jimklimov, 2022-10-03:
Wondering now how our CI does it on linux. OTOH maybe passes the configure parameters for systemd paths (by default detects them from OS tools...) and probably you forget to add DESTDIR to make install? Or we forget to handle it here?

fxcoudert, 2022-10-03:
`DESTDIR` is different from `--prefix`. Once `--prefix` is set, paths should typically be inside the prefix. `DESTDIR` allows to prepend a path to the prefix during install, while still having configured with a system path.

jimklimov, 2022-10-03:
Did you configure the path options for systemd pieces? Prefix is default for parts of the software project. However integrations with third party layout (systemd, augeas, udev, etc.) require installing into their known paths, or the files may as well not exist otherwise - won't be found by that other software in random locations.

jimklimov, 2022-10-03:
Also if you want to make an installation for e.g. "proto area" before packaging, you use layout (prefix, sysconfdir, ...) like on target OS and a DESTDIR to where you can write the isolated set of files. 
Many people install custom built NUT into their system by `make && sudo make install` and expect it to work out of the box (see putting systemd units ibto systemd locations).

jimklimov, 2022-10-03:
BTW, out of curiosity - what packaging is that? Distro-agnostic Homebrew built for arbitrary Linux (like pkgsrc and many other sidekicks)?

jimklimov, 2023-01-11:
In relation to #1708 now extending the NUT configuration reports to clarify where it plans to make a mess for third-party integrations.

closed 2023-01-11 by jimklimov