networkupstools/nut #1669: Powercom BNT-600AP (ProductID: 0004) on FreeBSD

andros-ua, 2022-10-04:
`FreeBSD 13.1-RELEASE-p2 GENERIC amd64`

`ugen0.2: <POWERCOM Co.,LTD HID UPS Battery> at usbus0, cfg=0 md=HOST spd=LOW (1.5Mbps) pwr=ON (100mA)`

/usr/local/etc/nut/ups.conf
```
[mypcm]
driver = usbhid-ups
port = auto
desc = "BNT-600AP"
```

`/usr/local/libexec/nut/usbhid-ups -DDDDD -a mypcm`
```
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] main_arg: var='driver' val='usbhid-ups'
   0.000026     [D3] main_arg: var='port' val='auto'
   0.000040     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000046     [D3] main_arg: var='desc' val='BNT-600AP'
   0.000059     [D1] debug level is '5'
   0.000406     [D5] send_to_all: SETINFO device.type "ups"
   0.000422     [D2] Initializing an USB-connected UPS with library libusb-1.0.0 (API: 0x1000102) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.000431     [D1] upsdrv_initups (non-SHUT)...
   0.000685     [D2] Checking device 1 of 1 (0D9F/0004)
   0.129090     [D1] nut_libusb_open get iManufacturer failed, retrying...
   0.260078     [D1] nut_libusb_open get iManufacturer failed, retrying...
   0.388056     [D1] nut_libusb_open get iManufacturer failed, retrying...
   0.516057     [D1] nut_libusb_open get iProduct failed, retrying...
   0.644063     [D1] nut_libusb_open get iProduct failed, retrying...
   0.772076     [D1] nut_libusb_open get iProduct failed, retrying...
   0.900057     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   1.028159     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   1.156059     [D1] nut_libusb_open get iSerialNumber failed, retrying...
   1.156075     [D2] - VendorID: 0d9f
   1.156079     [D2] - ProductID: 0004
   1.156083     [D2] - Manufacturer: unknown
   1.156086     [D2] - Product: unknown
   1.156090     [D2] - Serial Number: unknown
   1.156093     [D2] - Bus: 000
   1.156096     [D2] - Device: unknown
   1.156100     [D2] - Device release number: 0001
   1.156103     [D2] Trying to match device
   1.156107     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   1.156117     [D3] match_function_regex: matching a device...
   1.156122     [D2] Device matches
   1.156125     [D2] Reading first configuration descriptor
   1.156202     [D3] libusb_kernel_driver_active() returned 0
   1.156209     [D2] Claimed interface 0 successfully
   1.156215     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   1.224015     [D2] Unable to get HID descriptor (No device)
   1.224033     [D3] HID descriptor length (method 1) -1
   1.224038     [D4] i=0, extra[i]=09, extra[i+1]=21
   1.224047     [D3] HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 c6 03
   1.224054     [D3] HID descriptor length (method 2) 966
   1.224058     [D2] HID descriptor length 966
   1.288110     [D2] Unable to get Report descriptor: Input/output error
   1.288171     [D2] libusb1: No appropriate HID device found
   1.288178     libusb1: Could not open any HID devices: insufficient permissions on everything
   1.288183     No matching HID UPS found
```
The ups was checked on windows and is in working condition.
The issue was reproduced on two FreeBSD machines.

jimklimov, 2022-10-04:
Is `ugen` configured to let NUT use the device node? As a quick check, try running the driver with `user = root` option so it remains almighty, to rule out protocol incompatibilities.

andros-ua, 2022-10-04:
I've tried this one also listing permissions in dev
`/usr/local/libexec/nut/usbhid-ups -DDDDD -a mypcm -u root`
```
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D3] main_arg: var='driver' val='usbhid-ups'
   0.000028     [D3] main_arg: var='port' val='auto'
   0.000041     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000049     [D3] main_arg: var='desc' val='BNT-600AP'
   0.000062     [D1] Built-in default or configured user for drivers 'uucp' was ignored due to 'root' specified on command line
   0.000067     [D1] debug level is '5'
   0.000365     [D5] send_to_all: SETINFO device.type "ups"
   0.000378     [D2] Initializing an USB-connected UPS with library libusb-1.0.0 (API: 0x1000102) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.000383     [D1] upsdrv_initups (non-SHUT)...
   0.000614     [D2] Checking device 1 of 8 (0000/0000)
   0.000716     [D2] - VendorID: 0000
   0.000724     [D2] - ProductID: 0000
   0.000728     [D2] - Manufacturer: ATI
   0.000733     [D2] - Product: OHCI root HUB
   0.000736     [D2] - Serial Number: unknown
   0.000741     [D2] - Bus: 004
   0.000745     [D2] - Device: unknown
   0.000749     [D2] - Device release number: 0100
   0.000753     [D2] Trying to match device
   0.000757     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.000768     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.000773     [D2] Device does not match - skipping
   0.000788     [D2] Checking device 2 of 8 (0000/0000)
   0.000818     [D2] - VendorID: 0000
   0.000823     [D2] - ProductID: 0000
   0.000827     [D2] - Manufacturer: ATI
   0.000831     [D2] - Product: OHCI root HUB
   0.000835     [D2] - Serial Number: unknown
   0.000840     [D2] - Bus: 003
   0.000844     [D2] - Device: unknown
   0.000848     [D2] - Device release number: 0100
   0.000852     [D2] Trying to match device
   0.000857     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.000864     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.000868     [D2] Device does not match - skipping
   0.000878     [D2] Checking device 3 of 8 (0000/0000)
   0.000907     [D2] - VendorID: 0000
   0.000914     [D2] - ProductID: 0000
   0.000919     [D2] - Manufacturer: ATI
   0.000923     [D2] - Product: EHCI root HUB
   0.000927     [D2] - Serial Number: unknown
   0.000930     [D2] - Bus: 005
   0.000934     [D2] - Device: unknown
   0.000938     [D2] - Device release number: 0100
   0.000943     [D2] Trying to match device
   0.000947     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.000953     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.000957     [D2] Device does not match - skipping
   0.000967     [D2] Checking device 4 of 8 (0000/0000)
   0.000995     [D2] - VendorID: 0000
   0.001000     [D2] - ProductID: 0000
   0.001004     [D2] - Manufacturer: ATI
   0.001008     [D2] - Product: OHCI root HUB
   0.001012     [D2] - Serial Number: unknown
   0.001015     [D2] - Bus: 006
   0.001020     [D2] - Device: unknown
   0.001024     [D2] - Device release number: 0100
   0.001028     [D2] Trying to match device
   0.001032     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.001038     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.001043     [D2] Device does not match - skipping
   0.001053     [D2] Checking device 5 of 8 (0000/0000)
   0.001080     [D2] - VendorID: 0000
   0.001085     [D2] - ProductID: 0000
   0.001089     [D2] - Manufacturer: ATI
[... 68 lines cut ...]
   1.349811     [D2] libusb1: No appropriate HID device found
   1.349818     libusb1: Could not open any HID devices: insufficient permissions on everything
   1.349824     No matching HID UPS found
```

`ll /dev/usb`
```
total 0
crw-------  1 root  operator  0x31 Oct  5 00:49 0.1.0
crw-------  1 root  operator  0x52 Oct  5 00:49 0.1.1
crw-rw----  1 root  uucp      0x7f Oct  5 00:49 0.2.0
crw-------  1 root  operator  0x81 Oct  5 00:49 0.2.1
crw-------  1 root  operator  0x33 Oct  5 00:49 1.1.0
crw-------  1 root  operator  0x4f Oct  5 00:49 1.1.1
crw-------  1 root  operator  0x35 Oct  5 00:49 2.1.0
crw-------  1 root  operator  0x69 Oct  5 00:49 2.1.1
crw-------  1 root  operator  0x37 Oct  5 00:49 3.1.0
crw-------  1 root  operator  0x50 Oct  5 00:49 3.1.1
crw-------  1 root  operator  0x39 Oct  5 00:49 4.1.0
crw-------  1 root  operator  0x4e Oct  5 00:49 4.1.1
crw-------  1 root  operator  0x3b Oct  5 00:49 5.1.0
crw-------  1 root  operator  0x51 Oct  5 00:49 5.1.1
crw-------  1 root  operator  0x3d Oct  5 00:49 6.1.0
crw-------  1 root  operator  0x53 Oct  5 00:49 6.1.1
```

gdt, 2022-10-09:
I use NetBSD not FreeBSD and all my UPS units use actual serial ports.   Yes, the 90s called....  So take this with a grain of salt.

Looking at the logs, it seems like trouble in the USB code, perhaps not in nut.  In particular `nut_libusb_open get iManufacturer failed, retrying...` is concerning.   This doesn't really smell like permissions as nut is able to open the device.  But, keep in mind that (at least on NetBSD) there is /dev/usb and /dev/ugen, and I think the first is for enumeration and the second for interacting.

It is very suspicious that there are not device/manufacturer for the hubs.  However usbdevs -v on my NetBSD box (2014 dell) shows similar, so i guess it's just actually goofy, not a bug in your reading them.

Looking at the vendor, it seems to make sense; see https://devicehunt.com/view/type/usb/vendor/0D9F.

The parsing of the HID descriptor looks off.  966 bytes seems way too big, even though I know zip about HID descriptors.   When you ran the same debugging on windows, how was the HID parsed?  Or does that not use  libusb?   Have you tried any other HID debugging/printing programs?

I would do three things next:
  1) Look at the dmesg lines at boot, when this is probably probed and decided not to match and thus assigned to ugen.
  2) Run usbdevs to enumerate the device tree and similar inspection programs, as hard as you can.
  3) Use ktrace to run the driver, and then kdump, and look at the usb syscalls.  Then probably rebuild libusb-1.0 with debugging and also nut and run it under gdb, unless you figure out permissions issues.

gdt, 2022-10-09:
Also this seems like a clue something is wrong

```
   1.281593     [D2] Unable to get HID descriptor (No device)
   1.281615     [D3] HID descriptor length (method 1) -1
```

I would also post on freebsd-users (however that's spelled) and ask if anyone is successfully using *any* HID UPS, and if so to share the same debug info with you and compare.