networkupstools/nut #1672: Unable to enter background: Out of memory

martijndierckx, 2022-10-08:
I'm trying to get NUT up and running within docker.
Using an APC Back-UPS BX1600MI, connected via ESXi -> PhotonOS VM -> nut 2.80 on Alpine 3.16.2 in Docker con

The driver seems to start successfully:
```
+ /usr/sbin/upsdrvctl start
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
kill: No such process
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x0c
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x22
HIDParse: LogMax is less than LogMin. Vendor HID report descriptor may be incorrect; interpreting LogMax -1 as 255 in ReportID: 0x40
Using subdriver: APC HID 0.98
```

But upsd doens't want to start:
```
+ /usr/sbin/upsd
Network UPS Tools upsd 2.8.0
fopen /var/run/nut/upsd.pid: No such file or directory
Could not find PID file '/var/run/nut/upsd.pid' to see if previous upsd instance is already running!

listening on 0.0.0.0 port 3493
Connected to UPS [ups]: usbhid-ups-ups
Unable to enter background: Out of memory
```

Unless I run it with the -D option so it stays in the foreground:
```
+ /usr/sbin/upsd -D
Network UPS Tools upsd 2.8.0
   0.000000     fopen /var/run/nut/upsd.pid: No such file or directory
   0.000023     Could not find PID file '/var/run/nut/upsd.pid' to see if previous upsd instance is already running!

   0.000243     [D1] debug level is '1'
   0.000335     listening on 0.0.0.0 port 3493
   0.000663     Connected to UPS [ups]: usbhid-ups-ups
   0.001677     Running as foreground process, not saving a PID file
  31.832109     User monitor@127.0.0.1 logged into UPS [ups]
  43.037353     User monitor@127.0.0.1 logged out from UPS [ups]
```

After which I can successfully use upsc to read all values.

Any idea what is causing the "Unable to enter background: Out of memory" error? (Can't be a RAM issue since there's plenty of available RAM on the VM)

martijndierckx, 2022-10-08:
Solved it for now by running:
`/usr/sbin/upsd -F &`

It works, but is a bit hacky :-)

jimklimov, 2022-10-08:
Never saw such a case TBH. Maybe there are OS limits on forking and/or memory allotment (ulimit or other quotas)? Is NUT running in a Docker container - can that have limits set?

Maybe try running it with many `-D` options, in case some clues appear from debug trace, or via strace, ptrace, truss, dtrace etc. if that environment offers any such insight into syscalls. Or gdb (maybe via some IDE like Netbeans) if you know how to step with it and view variables etc.

I'm curious if it is some real ENOMEM from the OS, or misdiagnostics (mismatch of error code in headers used from what the OS emits), some bitness issue, etc.

As for `upsd -F` - it was added in 2.8.0 to let systemd avoid extra forking to track the daemon. Also `upsd -F &` makes it easier to run custom builds and debug them in the terminal, and then `fg` to `Ctrl+C` it away :)

thedanbob, 2023-02-15:
I ran into this myself a couple of days again. Weird thing is, the [image I use](https://github.com/instantlinux/docker-tools/tree/main/images/nut-upsd) was working just fine until the last docker update.

Unfortunately, adding a bunch of `-D`s to upsd didn’t really produce anything useful:
```
nut-upsd  | Network UPS Tools - UPS driver controller 2.8.0
nut-upsd  | Network UPS Tools - Generic HID driver 0.47 (2.8.0)
nut-upsd  | USB communication driver (libusb 1.0) 0.43
nut-upsd  | Using subdriver: CyberPower HID 0.6
nut-upsd  | Network UPS Tools upsd 2.8.0
nut-upsd  |    0.000000 Ignoring invalid pid number 0
nut-upsd  |    0.000322 [D3] listen_add: added 0.0.0.0:3493
nut-upsd  |    0.000380 [D1] debug level is '5’
nut-upsd  |    0.000417 [D3] setuptcp: try to bind to 0.0.0.0 port 3493
nut-upsd  |    0.000490 listening on 0.0.0.0 port 3493
nut-upsd  |    0.001049 Connected to UPS [ups]: usbhid-ups-ups
nut-upsd  |    0.002905 [D2] user_add_action: adding 'login' for upsmon
nut-upsd  |    0.002942 [D2] user_add_action: adding 'master' for upsmon
nut-upsd  |    0.002976 [D2] user_add_action: adding 'fsd' for upsmon
nut-upsd  |    0.003151 Unable to enter background: Out of memory
nut-upsd  |    0.003309 [D1] driver_free: forgetting UPS [ups] (FD 5)
```
Fortunately the `-F &` workaround did the trick to get it working at least.

jimklimov, 2023-02-15:
The "Invalid PID 0" seems to stem from entrypoint.sh:

````
# Issue #15 - change pid warning message from "No such file" to "Ignoring"
echo 0 > /var/run/nut/upsd.pid && chown $USER.$GROUP /var/run/nut/upsd.pid
echo 0 > /var/run/upsmon.pid
````

Not sure about "Out of memory", maybe docker forbids forking and/or detaching from grandparent terminal?

jthiltges, 2024-08-21:
I ran into this issue as well. Using strace showed a `fork()` call failing with ENOMEM.

It turned out that the container environment had a very high file ulimit:
```
# ulimit -n
1073741816
```

Setting a lower value in the docker-compose.yml got the service working as expected.
```
    ulimits:
      nofile: 2048
```

jimklimov, 2024-08-22:
Curious. Thanks for the report! :)

zolbarna, 2025-11-09:
I ran the same issue on docker.

Same container was running just fine in docker since 2022 on a different host. 

docker run .... .... .... --ulimit nofile=2048:2048 saved the day. Thanks @jthiltges