networkupstools/nut #1675: NextUPS soon or later nut lose the UPS

Pristep, 2022-10-11:
Hello I am beginner I love your NUT project which works on another manufacturer (EATON) but I am facing issue with "Next Ups" manufacturer.
Soon or later I get : `UPS ups@localhost is unavailable`

I am linux beginner I hope my issue is complete ?
Thanks by advance for the help you will provide !

System :
`Linux NUT 4.19.0-22-amd64 #1 SMP Debian Buster 10 4.19.260-1 (2022-09-29) x86_64 GNU/Linux`

Nut version :
```
  Installed: 2.7.4-8
  Candidate: 2.7.4-8
  Version table:
*** 2.7.4-8 500
        500 http://deb.debian.org/debian buster/main amd64 Packages
        100 /var/lib/dpkg/status

```
UPS :		
`NEXT UPS Systems Mantis II 2000 RT2U
`
sudo /lib/nut/bcmxcp_usb -DDD -a ups
```
Network UPS Tools - BCMXCP UPS driver 0.31 (2.7.4)
USB communication subdriver 0.22
0.0	Error: UPS [ups] is for driver usbhid-ups, but I'm bcmxcp_usb!
```

sudo systemctl status nut-server
```
● nut-server.service - Network UPS Tools - power devices information server
   Loaded: loaded (/etc/systemd/system/nut-server.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2022-10-10 11:36:27 CEST; 23h ago
 Main PID: 577 (upsd)
    Tasks: 1 (limit: 2357)
   Memory: 5.6M
   CGroup: /system.slice/nut-server.service
           └─577 /lib/nut/upsd

Oct 11 05:45:35 NUT upsd[577]: Send ping to UPS [ups] failed: Resource temporarily unavailable
Oct 11 05:45:37 NUT upsd[577]: Connected to UPS [ups]: usbhid-ups-ups
Oct 11 05:45:53 NUT upsd[577]: Data for UPS [ups] is stale - check driver
Oct 11 06:13:25 NUT upsd[577]: Send ping to UPS [ups] failed: Resource temporarily unavailable
Oct 11 06:13:27 NUT upsd[577]: Connected to UPS [ups]: usbhid-ups-ups
Oct 11 06:13:43 NUT upsd[577]: Data for UPS [ups] is stale - check driver
Oct 11 06:41:15 NUT upsd[577]: Send ping to UPS [ups] failed: Resource temporarily unavailable
Oct 11 06:41:17 NUT upsd[577]: Connected to UPS [ups]: usbhid-ups-ups
Oct 11 06:41:33 NUT upsd[577]: Data for UPS [ups] is stale - check driver
Oct 11 07:09:05 NUT upsd[577]: Send ping to UPS [ups] failed: Resource temporarily unavailable
```

sudo systemctl status nut-monitor
```
● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
   Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; vendor preset: enabled)
   Active: active (running) since Mon 2022-10-10 11:36:27 CEST; 23h ago
 Main PID: 580 (upsmon)
    Tasks: 2 (limit: 2357)
   Memory: 3.8M
   CGroup: /system.slice/nut-monitor.service
           ├─579 /lib/nut/upsmon
           └─580 /lib/nut/upsmon

Oct 11 10:44:44 NUT upsmon[580]: UPS [ups@localhost]: connect failed: Connection failure: Connection timed out
Oct 11 10:45:38 NUT systemd[1]: /lib/systemd/system/nut-monitor.service:6: PIDFile= references path below legacy directory /var/run/, updating /var/run/nut/upsmon.pid → /run/nut/upsmon.pid; plea
Oct 11 10:45:38 NUT systemd[1]: nut-monitor.service: Supervising process 580 which is not our child. We'll most likely not notice when it exits.
Oct 11 10:45:38 NUT systemd[1]: /lib/systemd/system/nut-monitor.service:6: PIDFile= references path below legacy directory /var/run/, updating /var/run/nut/upsmon.pid → /run/nut/upsmon.pid; plea
Oct 11 10:45:38 NUT systemd[1]: nut-monitor.service: Supervising process 580 which is not our child. We'll most likely not notice when it exits.
Oct 11 10:45:39 NUT systemd[1]: /lib/systemd/system/nut-monitor.service:6: PIDFile= references path below legacy directory /var/run/, updating /var/run/nut/upsmon.pid → /run/nut/upsmon.pid; plea
Oct 11 10:45:39 NUT systemd[1]: nut-monitor.service: Supervising process 580 which is not our child. We'll most likely not notice when it exits.
Oct 11 10:47:03 NUT upsmon[580]: UPS [ups@localhost]: connect failed: Connection failure: Connection timed out
Oct 11 10:47:03 NUT upsmon[580]: UPS ups@localhost is unavailable
Oct 11 10:49:22 NUT upsmon[580]: UPS [ups@localhost]: connect failed: Connection failure: 
Connection timed out
```

Often upsc ups return :
```
Init SSL without certificate database
Error: Data stale
```

And :
```
Init SSL without certificate database
battery.charge: 100
battery.runtime: 4410
battery.type: PbAc
battery.voltage: 55
battery.voltage.nominal: 48
device.mfr: PPC
device.model: HID UPS
device.serial: 717322500085
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: Phoenixtec/Liebert HID 0.3
driver.version.internal: 0.41
ups.load: 9
ups.mfr: PPC
ups.model: HID UPS
ups.productid: ffff
ups.serial: 717322500085
ups.status: OL CHRG
ups.vendorid: 06da
```

upsdrvctl start result :
```
Network UPS Tools - UPS driver controller 2.7.4
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
Using subdriver: Phoenixtec/Liebert HID 0.3
```

jimklimov, 2022-10-11:
> sudo /lib/nut/bcmxcp_usb -DDD -a ups

Why bcmxcp_usb if your driver is usbhid-ups?

Generally UPSes may get disconnected by an USB bus reset or chip reset (mobo, hubs along the way, device) including buggy/noisy cabling, firmware reboots etc. Some devices do tend to get lost more often than others.

The usbhid-ups should try to reconnect, though you may have to ensure that it is killed if it cannot reconnect (e.g. script detection of that state being too long). NUT master branch had some recent improvements in this area, if you'd manage a custom build.

Pristep, 2022-10-12:
> 
You right which command should I run

jimklimov, 2022-10-12:
The errors above (especially "Got disconnected by another driver") indicate that the other drivers were not stopped, and compete for access to device. If they are under systemd or similar facility, was the service stopped? Likewise if you (like many others) have a custom cron script checking that the driver is alive and restarting if not - was that neutered while you experiment?

The "libusb_get_interrupt: Connection timed out" happens with some chips, although I am not aware of good ways (nor statistics) to predict that this method should not be used with particular devices. However in such cases, a driver flag option "pollonly" may help - using a different communications method. So may well be that it is the fix you need.

I am also a bit concerned about this:
````
421.065031 failed to claim USB device: could not claim interface 0: Device or resource busy
421.065241 detached kernel driver from USB device...
````
it could mean that the kernel (or default OS handler) held the USB HID device node and did not immediately allow access to it; however further log lines indicate it did allow the access. Still, one more thing to keep in mind is to check that udev/ugen/... settings allow NUT user and programs to take over that device node in the OS. At least if other ideas do not help.

Pristep, 2022-10-13:
> The errors above (especially "Got disconnected by another driver") indicate that the other drivers were not stopped, and compete for access to device. If they are under systemd or similar facility, [...]

Thanks for the answer but you lost me I am linux beginner I don't have any script running by cron or anything.
Its a fresh Debian 10 VM I have installed nut and sudo.
Then I have copied the conf I had from my working nut server.

sudo cat /etc/nut/ups.conf 
maxretry = 3
[ups]
        driver = usbhid-ups
        #driver = bcmxcp_usb
        port = auto
        desc = "Mantis RT2U"

sudo cat /etc/nut/upsd.conf
        LISTEN 127.0.0.1 3493
        LISTEN 192.168.200.60 3493

sudo cat /etc/nut/upsd.users
        [admin]
                password = @dm!n-Pri
                actions = SET
                instcmds = ALL
        [upsmon_local]
                password  = L0c@l4Pr!
                upsmon master
        [upsmon_remote]
                password  = R3M0te4Pr!
                upsmon slave
        [monuser]
                password  = secret
                upsmon slave

sudo cat /etc/nut/upsmon.conf
        MONITOR ups@localhost 1 upsmon_local L0c@l4Pr! master
        MINSUPPLIES 1
        SHUTDOWNCMD "/sbin/shutdown -h +0"
        POLLFREQ 10
        POLLFREQALERT 5
        HOSTSYNC 15
        DEADTIME 20
        POWERDOWNFLAG /etc/killpower
        RBWARNTIME 43200
        NOCOMMWARNTIME 300
        FINALDELAY 5

sudo cat /etc/nut/nut.conf
        MODE=netserver

jimklimov, 2022-10-14:
> MONITOR ***@***.*** 1 upsmon_local ***@***.***! master

(Quick comment: I hope the passwords in the sample above are examples... or
soon will be ;) )

Pristep, 2022-10-17:
> (Quick comment: I hope the passwords in the sample above are examples... or soon will be ;) )
> […](#)
I have edited the answer you right It wasnt smart again ...
I will change it locally !
But i am still facing the issue :(

jimklimov, 2022-10-19:
"Fresh VM" is a clue - is USB passed through from the host system? Does the host know to pass this device (probably by VID:PID, possibly also bus/port/... identifiers) "completely" without barging into the protocol dialog?

For now, check if this addition to avoid interrupts would "just help":
````
# /etc/nut/ups.conf
maxretry = 3
[ups]
    driver = usbhid-ups
    #driver = bcmxcp_usb
    port = auto
    desc = "Mantis RT2U"
    pollonly
````
IIRC, that's the syntax for it.

Pristep, 2022-10-24:
> "Fresh VM" is a clue - is USB passed through from the host system? Does the host know to pass this device (probably by VID:PID, possibly also bus/port/... identifiers) "completely" without barging [...]

Ok it worked one weeke without any error and both services remains running with green circle.
Thanks you.
I have a last question because you helped me much, maybe you can help with one last issue.
I would like to test the nut server like there is a power outage but it seems the UPS doesn't support any command.
```
upscmd -l ups
Instant commands supported on UPS [ups]:

```
So I think I can't test it by the safe way ?
The command : upsmon -c fsd just turn off the nut srv ...

jimklimov, 2022-10-25:
The typical approach for safe graceful-shutdown tests is to redefine `SHUTDOWNCMD` to something that logs a "gonna shutdown" message (instead of `/sbin/shutdown...`) and restart the `upsmon` daemon - but this only tests that/how/when this service would react to the external events (whether your commands or real outage).

However as part of such testing, the UPS is usually not powering the real load but some mock of it (lamps, monitors, one PSU of redundantly-powered servers, etc.) draining the battery - an approach which you can use for "safe" testing of commands like immediate power-off.