networkupstools/nut #1677: RFE: Add a way for drivers to pass commands to device via CLI

jimklimov, 2022-10-12:
As highlighted in #1673 discussion, there may be a practical need to send shutdown start/abort commands to locally managed UPS and similar devices, while the driver daemon and `upsd` are no longer running.

This RFE is about investigation if this ability exists. If not, it should be commonly implemented (in `drivers/main.c` probably) and documented.

user01010111, 2026-09-20:
I believe #2686 supplies the facility requested here: with the driver daemon stopped, `driver -a upsname -x sdcommands=shutdown.stop -k -F` runs the selected command directly, without `upsd`, provided the driver and device support it. With an active driver daemon, `-k` first requests that daemon's configured shutdown action, so that is a different case.

I verified configured overrides, unknown-command rejection and fallback ordering against an isolated serial simulator using a fresh build of master at `a66c0090`. I have prepared a `nutupsdrv(8)` example and `ups.conf(5)` cross-reference. Would that documentation complete #1677?

AI assistance: OpenAI Codex, using `gpt-6-astra` with `xhigh` reasoning effort.

jimklimov, 2026-09-23:
@user01010111 : yeah, it seems that feature does solve the problem so catching up with documentation would be the cherry on top :)

user01010111, 2026-09-23:
Thanks @jimklimov : the documentation update is now merged in [#3672](https://github.com/networkupstools/nut/pull/3672), so I think this issue can now be closed 👌

closed 2026-09-24 by jimklimov