networkupstools/nut #1686: Manpage missing for `clone-outlet` driver

jimklimov, 2022-10-24:
Need to discover and document the difference between mostly-similar sibling drivers:
* https://github.com/networkupstools/nut/blob/master/drivers/clone-outlet.c
* https://github.com/networkupstools/nut/blob/master/drivers/clone.c

aquette, 2022-10-24:
AFAIR, one clones the whole ups, while the other, just a specific outlet.
Intended for upsmon per outlet protection (test discussed with Arjen ; could be done in upsmon too).
I'm unsure why Arjen made 2 drivers, instead of one though, it's more than a decade old. Worth simplifying probably

jimklimov, 2024-09-10:
I think I've unraveled why there are two drivers here, and also found that I had a misconception that they would report only the specific outlet's info. In fact, both report the original "real" driver's information, with only some data points modified (primarily `ups.status`) based on a particular outlet's information.

My guess is that a larger deployment with several lower-priority devices collected on a manageable outlet of an UPS or ePDU would likely see several drivers set up on the server actually capable of interactions with the UPS and running the NUT data server `upsd`:

* a "real" driver talking to the UPS;
* a `clone` driver talking to the "real" driver and issuing outlet power-off (or power-cycle) based on relatively high thresholds for remaining battery charge and/or runtime of the actual UPS (or explicit instant commands), with such operations first setting the respective timers for the outlet on the "real" driver, and the "FSD" flag among states of the virtual UPS status;
* possibly a `clone-outlet` driver which is read-only and interprets the outlet timer values as triggers for "FSD" or "OFF" flags reported among states of the virtual UPS status.

With this approach, the lower-priority systems collected on such outlet would run the NUT `upsmon` client to `MONITOR` the virtual UPS presented by the read-only `clone-outlet` driver and shut down as soon as the "FSD" flag is raised.

The `clone` driver responsible for outlet power state changes would not normally be monitored directly (e.g. to avoid unfortunate direct shutdown requests from those clients).

Perhaps we want a concept for drivers to only be accessible to a certain login (including an absent one, to avoid direct `upsrw`/`upscmd` management ability).

jimklimov, 2024-09-10:
@arjendekorte : could you please chime in, if I got your 15-year-old idea about right? :)

PR #2618 (2024-09-11): Add man page for `clone-outlet` driver
Closes: #1686

Also resyncs a bit of code style differences, to minimize comparison noise.

TODO:
* [x] Incorporate comments from issue
* [x] Revise (test?) what `clone` does exactly
closed 2024-09-11 by jimklimov