networkupstools/nut #1687: Windows nut-Windows-v2.8.0-alpha3.tar.gz link is broken

joachim-heck, 2022-10-27:
Dear all,

I'm struggeling to build the https://github.com/networkupstools/nut/releases/tag/v2.8.0-Windows version on my own.
Is there any way that somebody might fix the broken download link from the download page (https://networkupstools.org/download.html)
http://www.networkupstools.org/source/Win/nut-Windows-v2.8.0-alpha3.tar.gz ? 

Thanks and regard!

jimklimov, 2022-10-27:
That was a relatively old snapshot by now. The code was merged since, so you can start with building `master` branch right away. Hopefully docs would lead you how, depending on the platform you build on (linux+mingw or windows+msys2) see docs/config-prereqs.txt or scripts/Windows/README respectively.

Recent CI builds should be available on Appveyor for a month after the build ran; see links in NUT website footer.

Note that not *everything* was completed for Windows, known differences documented in issue #5 and those linked to it - and help would be welcome ;)

joachim-heck, 2022-10-28:
Thanks for your support - Finally we have an running build! 

**We managed it like that:**
- Installed Windows MSYS2 - MINGW64
- Installed required dependencies: git, build toolchain, GCC, LibUSB,..
- git clone of the master branch
- run: `./configure --with-usb`
- run: `make install-win-bundle DESTDIR=/output-folder`

Unfortunately I'm using an EATON UPS in combination with the LibUSB 1.2.6.0 which is working like charm on my linux machine, **but doesn't want to connect on the windows build.** The old NUT Version was working on windows but the returned values where partially incorrect and fewer values where collected and returned. The EATON is connected correctly and detected - also accessible with the eaton tools.  

**Console-Output:**
```
Network UPS Tools - Generic HID driver 0.49 (2.8.0-Windows-55-g6538df32e)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] debug level is '2'
   0.001217     [D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.008066     [D1] upsdrv_initups (non-SHUT)...
   2.648681     [D2] Checking device 1 of 5 (0B1F/03E9)
   2.650760     [D2] - VendorID: 0b1f
   2.652079     [D2] - ProductID: 03e9
   2.653508     [D2] - Manufacturer: unknown
   2.655120     [D2] - Product: unknown
   2.656570     [D2] - Serial Number: unknown
   2.658212     [D2] - Bus: 001
   2.659397     [D2] - Device: unknown
   2.660813     [D2] - Device release number: 0100
   2.662618     [D2] Trying to match device
   2.665398     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   2.668322     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   2.672318     [D2] Device does not match - skipping
   2.674244     [D2] Checking device 2 of 5 (0B1F/03E8)
   2.676190     [D1] Failed to open device (0B1F/03E8), skipping: Operation not supported or unimplemented on this platform
   2.681766     [D2] Checking device 3 of 5 (0463/FFFF)
   2.760148     [D2] - VendorID: 0463
   2.761392     [D2] - ProductID: ffff
   2.762896     [D2] - Manufacturer: EATON
   2.764443     [D2] - Product: Eaton 9SX
   2.765955     [D2] - Serial Number: GC36L47026
   2.767691     [D2] - Bus: 001
   2.768876     [D2] - Device: unknown
   2.770287     [D2] - Device release number: 0202
   2.773241     [D2] Trying to match device
   2.774883     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   2.777807     [D2] Device matches
   2.779104     [D2] Reading first configuration descriptor
   2.781234     [D2] Claimed interface 0 successfully
   2.783112     [D2] Unable to get HID descriptor (Invalid parameter)
   2.785526     [D1] Eaton device v2.02. Using full report descriptor
   2.788847     [D2] Unable to retrieve any HID descriptor
   2.790950     [D2] Checking device 4 of 5 (8086/A1AF)
   2.792899     [D1] Failed to open device (8086/A1AF), skipping: Operation not supported or unimplemented on this platform
   2.797354     [D2] Checking device 5 of 5 (064F/2AF9)
   2.799764     [D1] Failed to open device (064F/2AF9), skipping: Input/Output Error
   2.802438     [D2] libusb1: No appropriate HID device found
   2.805581     libusb1: Could not open any HID devices: insufficient permissions on everything
   2.808935     No matching HID UPS found
```

**From my point of view this is somehow related to this part in the codebase of libusb1:** 
https://github.com/networkupstools/nut/blob/035ae79dc28a04c443f110bd7046aab3c504319b/drivers/libusb1.c

- Initially `row 449` fails
- Then `row 507` sets rdlen = rdlen1 which is still -1 at this point 
- then `row 515` checks whether rdlen is < 0 which to debug output a continue to next device
- which shouldn't happen

[image]

**Is this behaviour intentended?**

closed 2022-10-28 by joachim-heck