networkupstools/nut #1693: AEG Protec C support

V1pr, 2022-11-01:
Hi,

I've an AEG Protec C 2000 UPS, which has serial. Currently it's working on my Debian 11 with Generex UPSMAN over a serial-to-usb adapter. Is it possible to use the UPS with NUT somehow?

Thanks,

        Tom

jimklimov, 2022-11-03:
Hard to say, first time I've heard of that model or software.

According to https://networkupstools.org/stable-hcl.html there is support for some AEG devices... if yours is similar (i.e. talks Megatec Qx protocol family) - with serial port you could try older `blazer_ser` or newer `nutdrv_qx` drivers.

jimklimov, 2022-11-03:
> Is it possible to use the UPS with NUT somehow?

Also the general answer to such a broad question is "Of course, but you may need to get vendor specs for the protocol or reverse-engineer it, and write a NUT driver!" :)