networkupstools/nut #1695: usbhid-ups: shutdown not working on Powercom BNT-800AP

Vladdrako, 2022-11-03:
NUT 2.8.0 27 Apr 2022

Shutdown commands on Powercom BNT-800AP do nothing.

ups.conf
<pre>
[powercom]
desc = Powercom BNT-800AP Schuko
driver = usbhid-ups
port = auto
productid = 0004
vendorid = 0d9f
</pre>
upsc powercom
<pre>
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 30
battery.date: 2014/01/15
battery.runtime: 800
battery.temperature: 25.0
battery.type: PbAc
device.mfr: POWERCOM Co.,LTD
device.model: HID UPS Battery
device.serial: 004-0D9F-000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.productid: 0004
driver.parameter.synchronous: auto
driver.parameter.vendorid: 0d9f
driver.version: reboot-22899-gedd727a917
driver.version.data: PowerCOM HID 0.6
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.26 (API: 0x1000109)
input.frequency: 50.0
input.voltage: 230.0
input.voltage.nominal: 220
output.frequency: 50.0
output.voltage: 230.0
output.voltage.nominal: 220
ups.date: 2014/01/15
ups.delay.shutdown: 20
ups.delay.start: 60
ups.load: 20
ups.mfr: POWERCOM Co.,LTD
ups.model: HID UPS Battery
ups.productid: 0004
ups.serial: 004-0D9F-000
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: 0
ups.timer.start: 771120
ups.vendorid: 0d9f
</pre>

/lib/nut/usbhid-ups -DDD -a powercom -k
<details>
<pre>
root@MI-R3P:~# /lib/nut/usbhid-ups -DDD -a powercom -k
Network UPS Tools - Generic HID driver 0.47 (reboot-22899-gedd727a917)
USB communication driver (libusb 1.0) 0.43
   0.000000	[D3] main_arg: var='desc' val='Powercom'
   0.000158	[D3] main_arg: var='driver' val='usbhid-ups'
   0.000202	[D3] main_arg: var='port' val='auto'
   0.000269	[D3] main_arg: var='productid' val='0004'
   0.000356	[D3] main_arg: var='vendorid' val='0d9f'
   0.001028	[D1] debug level is '3'
[... 401 lines cut ...]
   0.190934	[D3] instcmd: using Path 'UPS.PowerSummary.DelayBeforeShutdown'
   0.191011	[D3] powercom_shutdown_nuf: value = (null), command = 5400
   0.201172	[D3] Report[set]: (3 bytes) => 0f 00 54
   0.201296	[D3] instcmd: SUCCEED

   0.201415	[D1] upsdrv_cleanup...
</pre>
</details>

upscmd -l powercom

<pre>
Instant commands supported on UPS [powercom]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.toggle - Toggle the UPS beeper
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
test.battery.start.quick - Start a quick battery test
</pre>

beeper.disable, beeper.enable, test.battery.start.quick works fine. The rest - no(?)

Vladdrako, 2022-11-03:
root@MI-R3P:~# /lib/nut/usbhid-ups -DDDDDD -u root -a powercom
Network UPS Tools - Generic HID driver 0.47 (reboot-22899-gedd727a917)
USB communication driver (libusb 1.0) 0.43
   0.000000	[D1] Built-in default or configured user for drivers 'root' was ignored due to 'root' specified on command line
   0.000527	[D3] main_arg: var='desc' val='Powercom'
   0.000764	[D3] main_arg: var='driver' val='usbhid-ups'
   0.000867	[D3] main_arg: var='port' val='auto'
   0.001003	[D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.001109	[D3] main_arg: var='productid' val='0004'
   0.001241	[D5] send_to_all: SETINFO driver.parameter.productid "0004"
   0.001384	[D3] main_arg: var='vendorid' val='0d9f'
   0.001525	[D5] send_to_all: SETINFO driver.parameter.vendorid "0d9f"
   0.001902	[D1] debug level is '6'
   0.004037	[D5] send_to_all: SETINFO device.type "ups"
   0.004176	[D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.004313	[D1] upsdrv_initups (non-SHUT)...
   0.012370	[D2] Checking device 1 of 11 (0080/A001)
   0.014419	[D2] - VendorID: 0080
   0.014549	[D2] - ProductID: a001
   0.014639	[D2] - Manufacturer: Western Digital
   0.014750	[D2] - Product: WDC WD30EZRX-00DC0B0
   0.014825	[D2] - Serial Number: WMC1T1872664
   0.014902	[D2] - Bus: 002
   0.015003	[D2] - Device: unknown
   0.015078	[D2] - Device release number: 3102
   0.015151	[D2] Trying to match device
   0.015251	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.015363	[D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.015498	[D2] Device does not match - skipping
   0.015659	[D2] Checking device 2 of 11 (0D9F/0004)
   0.022825	[D2] - VendorID: 0d9f
   0.022942	[D2] - ProductID: 0004
   0.023028	[D2] - Manufacturer: POWERCOM Co.,LTD
   0.023102	[D2] - Product: HID UPS Battery
   0.023170	[D2] - Serial Number: 004-0D9F-000
   0.023244	[D2] - Bus: 001
   0.023316	[D2] - Device: unknown
   0.023391	[D2] - Device release number: 0002
   0.023498	[D2] Trying to match device
   0.023599	[D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.023813	[D3] match_function_regex: matching a device...
   0.024026	[D2] Device matches
   0.024159	[D2] Reading first configuration descriptor
   0.024427	[D3] libusb_kernel_driver_active() returned 0
   0.024630	[D2] Claimed interface 0 successfully
   0.024780	[D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.025957	[D3] HID descriptor, method 1: (9 bytes) => 09 21 00 01 00 01 22 e4 03
   0.026076	[D3] HID descriptor length (method 1) 996
   0.026189	[D4] i=0, extra[i]=09, extra[i+1]=21
   0.026293	[D3] HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 e4 03
   0.026404	[D3] HID descriptor length (method 2) 996
   0.026482	[D2] HID descriptor length 996
   0.061536	[D2] Report Descriptor size = 996
   0.061711	[D3] Report Descriptor: (996 bytes) => 05 84 09 04 a1 01 09 24 a1 02 85 01 09 fe
   0.061862	[D3]  79 01 75 08 95 01 15 00 26 ff 00 b1 22 85 02 09 ff 79 02 b1 22 05 85 85 05
   0.062044	[D3]  09 89 79 04 b1 22 85 07 09 8b b1 22 85 08 09 2c 25 03 b1 22 85 09 09 83 25
   0.062779	[D3]  64 b1 22 09 67 b1 22 85 0e 09 66 81 22 09 66 b1 a2 85 0b 09 8c b1 22 85 0c
   0.063093	[D3]  09 29 b1 a2 85 0d 09 85 75 10 27 ff ff 00 00 b1 a2 85 37 09 68 81 22 09 68
   0.063256	[D3]  b1 a2 05 84 85 0f 09 57 75 10 16 00 00 27 ff ff 00 00 b1 a2 85 10 09 56 75
   0.063475	[D3]  20 16 00 00 27 9f 86 01 00 b1 a2 85 11 09 40 75 10 16 64 00 26 f0 00 b1 22
   0.063645	[D3]  85 12 09 30 75 10 16 00 00 26 40 01 b1 a2 85 2c 09 36 75 08 25 5a b1 a2 85
   0.063863	[D3]  13 09 5a 75 08 15 00 25 03 b1 a2 09 02 a1 02 05 85 85 0a 09 44 75 01 15 00
   0.064032	[D3]  25 01 81 a2 09 44 b1 a2 09 45 81 a2 09 45 b1 a2 09 d0 81 a2 09 d0 b1 a2 09
   0.064228	[D3]  d1 81 a2 09 d1 b1 a2 09 42 81 a2 09 42 b1 a2 09 43 81 a2 09 43 b1 a2 09 4b
   0.064459	[D3]  81 a2 09 4b b1 a2 09 db 81 a2 09 db b1 a2 05 84 09 68 81 a2 09 68 b1 a2 09
   0.064669	[D3]  69 81 a2 09 69 b1 a2 09 73 81 a2 09 73 b1 a2 09 65 81 a2 09 65 b1 a2 95 03
   0.064888	[D3]  81 01 b1 01 95 01 06 86 ff 09 80 81 a2 09 80 b1 a2 c0 c0 05 84 09 12 a1 00
   0.065090	[D3]  85 15 09 58 75 08 25 66 b1 a2 85 16 09 70 75 08 25 03 b1 a2 05 85 85 17 09
   0.065308	[D3]  85 75 10 27 ff ff 00 00 b1 a2 85 18 09 66 75 08 15 00 25 64 b1 a2 05 84 85
   0.065530	[D3]  19 09 40 75 10 16 00 0c 26 00 30 b1 22 85 1a 09 30 75 10 16 00 09 26 00 24
[... 1672 lines cut ...]
   0.631219	[D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   0.631435	[D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   0.631637	[D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   0.631891	[D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   0.632069	[D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   0.632220	[D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   0.632520	[D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   0.632782	[D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
   0.632964	[D3]  00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
[... cut ...]
   0.634070	[D3] Report[buf]: (3 bytes) => 37 20 03
   0.634151	[D5] PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   0.634228	[D5] Unit = 00000000, UnitExp = 0
   0.634300	[D5] Exponent = 0
   0.634416	[D5] hid_lookup_path: 00840004 -> UPS
   0.634558	[D5] hid_lookup_path: 00840024 -> PowerSummary
   0.634682	[D5] hid_lookup_path: 00850068 -> RunTimeToEmpty
   0.634782	[D2] Path: UPS.PowerSummary.RunTimeToEmpty, Type: Input, ReportID: 0x37, Offset: 0, Size: 16, Value: 800
   0.634933	[D1] Quick update...
</pre>
</details>

jimklimov, 2022-11-03:
I wonder if this is among models that only turn off programmatically if currently on battery (e.g. presence of wall power precludes a poweroff and/or causes it to start back up quickly).

Likewise, to test `load.on*` commands you may have to get it into the "off" state first (e.g. by button)?

Haven't used this model so just guessing... but I had several Powercom's (simple and BNT) about 20 years ago and with serial port - NUT did work with those for poweroff and reboot.

Vladdrako, 2022-11-03:
> I wonder if this is among models that only turn off programmatically if currently on battery (e.g. presence of wall power precludes a poweroff and/or causes it to start back up quickly).

Switched the UPS to the battery and tried shutdown.return - nothing happened. usbhid-ups didn't recognize the UPS on this mode.

> Likewise, to test `load.on*` commands you may have to get it into the "off" state first (e.g. by button)?

Powered off UPS by button. No result after load.on command. Just OK.

All of this I tested on router with OpenWRT OS.
Now I connected UPS to the PC.

/lib/nut/usbhid-ups -DDDDDD -u root -a powercom2 -k

<details>
<pre>
Network UPS Tools - Generic HID driver 0.47 (2.8.0)
USB communication driver (libusb 1.0) 0.43
   0.000000     [D1] Built-in default or configured user for drivers 'nut' was ignored due to 'root' specified on command line
   0.000055     [D3] do_global_args: var='maxretry' val='3'
   0.000101     [D3] main_arg: var='driver' val='usbhid-ups'
   0.000108     [D3] main_arg: var='port' val='auto'
   0.000113     [D5] send_to_all: SETINFO driver.parameter.port "auto"
   0.000125     [D1] debug level is '6'
   0.000451     [D5] send_to_all: SETINFO device.type "ups"
   0.000460     [D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.43')
   0.000465     [D1] upsdrv_initups (non-SHUT)...
   0.004712     [D2] Checking device 1 of 9 (04D9/A124)
   0.006251     [D2] - VendorID: 04d9
   0.006264     [D2] - ProductID: a124
   0.006273     [D2] - Manufacturer: HOLTEK
   0.006279     [D2] - Product: USB-HID Keyboard
   0.006283     [D2] - Serial Number: AP0000000003
   0.006287     [D2] - Bus: 002
   0.006290     [D2] - Device: unknown
   0.006295     [D2] - Device release number: 0100
   0.006308     [D2] Trying to match device
   0.006311     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.006320     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.006324     [D2] Device does not match - skipping
   0.006335     [D2] Checking device 2 of 9 (0D9F/0004)
   0.013016     [D2] - VendorID: 0d9f
   0.013030     [D2] - ProductID: 0004
   0.013034     [D2] - Manufacturer: POWERCOM Co.,LTD
   0.013038     [D2] - Product: HID UPS Battery
   0.013042     [D2] - Serial Number: 004-0D9F-000
   0.013046     [D2] - Bus: 002
   0.013058     [D2] - Device: unknown
   0.013062     [D2] - Device release number: 0002
   0.013065     [D2] Trying to match device
   0.013069     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.013076     [D3] match_function_regex: matching a device...
   0.013081     [D2] Device matches
   0.013085     [D2] Reading first configuration descriptor
   0.013092     [D3] libusb_kernel_driver_active() returned 0
   0.013105     [D2] Claimed interface 0 successfully
   0.013109     [D3] nut_usb_set_altinterface: skipped libusb_set_interface_alt_setting(udev, 0, 0)
   0.013895     [D3] HID descriptor, method 1: (9 bytes) => 09 21 00 01 00 01 22 e4 03
   0.013907     [D3] HID descriptor length (method 1) 996
   0.013912     [D4] i=0, extra[i]=09, extra[i+1]=21
   0.013918     [D3] HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 e4 03
   0.013923     [D3] HID descriptor length (method 2) 996
   0.013936     [D2] HID descriptor length 996
   0.046420     [D2] Report Descriptor size = 996
   0.046443     [D3] Report Descriptor: (996 bytes) => 05 84 09 04 a1 01 09 24 a1 02 85 01 09 fe
   0.046451     [D3]  79 01 75 08 95 01 15 00 26 ff 00 b1 22 85 02 09 ff 79 02 b1 22 05 85 85 05
   0.046458     [D3]  09 89 79 04 b1 22 85 07 09 8b b1 22 85 08 09 2c 25 03 b1 22 85 09 09 83 25
   0.046463     [D3]  64 b1 22 09 67 b1 22 85 0e 09 66 81 22 09 66 b1 a2 85 0b 09 8c b1 22 85 0c
   0.046471     [D3]  09 29 b1 a2 85 0d 09 85 75 10 27 ff ff 00 00 b1 a2 85 37 09 68 81 22 09 68
   0.046478     [D3]  b1 a2 05 84 85 0f 09 57 75 10 16 00 00 27 ff ff 00 00 b1 a2 85 10 09 56 75
   0.046484     [D3]  20 16 00 00 27 9f 86 01 00 b1 a2 85 11 09 40 75 10 16 64 00 26 f0 00 b1 22
[... 3021 lines cut ...]
   0.101473     [D5] hid_lookup_usage: DelayBeforeShutdown -> 00840057
   0.101477     [D4] string_to_path: depth = 3
   0.101482     [D5] send_to_all: SETINFO ups.delay.shutdown "20"
   0.101487     [D5] send_to_all: SETFLAGS ups.delay.shutdown RW STRING
   0.101491     [D5] send_to_all: SETAUX ups.delay.shutdown 8
   0.101496     [D5] hid_lookup_usage: UPS -> 00840004
   0.101501     [D5] hid_lookup_usage: PowerSummary -> 00840024
   0.101506     [D5] hid_lookup_usage: DelayBeforeShutdown -> 00840057
   0.101510     [D4] string_to_path: depth = 3
   0.101515     [D3] Report[buf]: (3 bytes) => 0f 00 00
   0.101519     [D5] PhyMax = 0, PhyMin = 0, LogMax = 65535, LogMin = 0
   0.101524     [D5] Unit = 00000000, UnitExp = 0
   0.101528     [D5] Exponent = 0
   0.101532     [D2] Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x0f, Offset: 0, Size: 16, Value: 0
   0.101538     [D3] powercom_shutdown_fun: value = 0, buf = 0
   0.101543     [D5] send_to_all: SETINFO ups.timer.shutdown "0"
   0.101548     [D5] hid_lookup_usage: UPS -> 00840004
   0.101553     [D5] hid_lookup_usage: PowerSummary -> 00840024
   0.101558     [D5] hid_lookup_usage: PCMDelayBeforeShutdown -> 00ff0057
   0.101562     [D4] string_to_path: depth = 3
   0.101567     [D5] hid_lookup_usage: UPS -> 00840004
   0.101572     [D5] hid_lookup_usage: PowerSummary -> 00840024
   0.101577     [D5] hid_lookup_usage: PCMDelayBeforeShutdown -> 00ff0057
   0.101581     [D4] string_to_path: depth = 3
   0.101586     [D5] hid_lookup_u

tannisroot, 2023-05-18:
<details>
  <summary>upsc powercom</summary>

  ```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 30
battery.date: 2014/01/15
battery.runtime: 2100
battery.type: PbAc
device.mfr: POWERCOM Co.,LTD
device.model: HID UPS Battery
device.serial: 004-0D9F-000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: PowerCOM HID 0.5
driver.version.internal: 0.41
input.frequency: 50.0
input.voltage: 226.0
input.voltage.nominal: 240
output.frequency: 50.0
output.voltage: 226.0
output.voltage.nominal: 240
ups.beeper.status: disabled
ups.date: 2014/01/15
ups.delay.shutdown: 20
ups.delay.start: 60
ups.load: 7
ups.mfr: POWERCOM Co.,LTD
ups.model: HID UPS Battery
ups.productid: 0004
ups.serial: 004-0D9F-000
ups.status: OL
ups.test.result: Done and passed
ups.timer.shutdown: 0
ups.timer.start: 740160
ups.vendorid: 0d9f

  ```
</details>

<details>
  <summary>/lib/nut/usbhid-ups -DDD -a powercom -k (connected to power)</summary>

  ```
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000	debug level is '3'
   0.086292	upsdrv_initups...
   0.312622	Checking device (1D6B/0003) (006/001)
   0.340449	- VendorID: 1d6b
   0.340458	- ProductID: 0003
   0.340461	- Manufacturer: unknown
   0.340467	- Product: unknown
   0.340470	- Serial Number: unknown
   0.340473	- Bus: 006
   0.340477	- Device release number: 0515
   0.340481	Trying to match device
   0.340489	Device does not match - skipping
   0.340614	Checking device (10C4/EA60) (005/004)
   0.340631	- VendorID: 10c4
   0.340635	- ProductID: ea60
   0.340639	- Manufacturer: unknown
   0.340643	- Product: unknown
   0.340647	- Serial Number: unknown
[... 920 lines cut ...]
   0.682841	find_nut_info: unknown info type: load.off.delay
   0.682845	Initiating UPS shutdown
   0.682848	upsdrv_shutdown...
   0.682852	instcmd(shutdown.return, [NULL])
   0.682857	powercom_shutdown_nuf: value = (null), command = 5400
   1.032947	Report[set]: (3 bytes) => 0f 00 54
   1.032964	upsdrv_cleanup...
  ```
</details>

upscmd -l powercom
```
Instant commands supported on UPS [powercom]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.toggle - Toggle the UPS beeper
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
test.battery.start.quick - Start a quick battery test
```
As with the other guy all the other commands work but not the shutdown ones.
The UPS does however shutdown properly with the vendor's proprietary tool, UPSMON.

closed 2024-07-14 by jimklimov