networkupstools/nut #1696: blazer_usb not working with upsonic UPS

joesitsolutions, 2022-11-06:
Hi I can't seem to get my ups connected to NUT.,
I've tried different drivers/ports/configs and cant seem to get it working.
Please see the following text dump:
[https://pastebin.com/2e2Y1zC7](url)
Not 100% sure what I'm doing but it seems weird that it won't work no matter what.
I'm using CentOS 7 on a potato PC as a NUT server for testing.

jimklimov, 2022-11-06:
Typically, running the driver binary with higher debug level can expose what it is trying to do and how exactly it fails. In your case, it would be like `blazer_usb -a Upsonic -DDDDDD` (after stopping an earlier running driver, if any).

I am not sure if `blazer_usb` is or is not a best-fit driver for that device, at least with modern NUT a `nutdrv_qx` might fare better.
The `nut-scanner` mostly matches by product/vendor IDs and suggest drivers that are known to support those -- and generic unregistered ID numbers (shady OEM vendors) may have arbitrary functionality behind them.

Generally, USB related problems are often about OS permissions (may your packaged `nut` driver account access the Linux device node for that connection?) and this is typically managed by `udev` rules file that says "give VendorID:ProductID X:Y to user Z" a lot of times over (or a different tech for same purpose on other OSes).

NUT v2.7.2 was released over 8 years ago, so maybe newer iterations added support for the Megatec Qx protocol dialects that one was missing and that might be or not be needed for your device.

Current NUT (github master) is tested on CentOS 7 among its CI targets, so you can give it a shot - revise `docs/config-prereqs.txt` for tools and third-party libs you might need (doc is geared towards building as much as possible however - you might want to skip on html/pdf generation or CGI bits that tend to pull in much of X11 and TeX), and figure out `configure` script options from your distro's packaging recipes that could help a drop-in replacement. 

For first shots however, the default paths (/usr/local/ups/...) might be better to exactly avoid a conflict and not overwrite anything should you want to keep the original packaged version, so you might want to just specify the same packaged user and group to run as, and `--sysconfdir=/etc/ups` (or what is appropriate there) to see same config files.

Thinking of udev mentioned above, a typical `sudo make install` would rewrite some systems-integration files - including systemd unit definitions and udev rules files. In case of recent NUT, it is named and structured differently (udev.hwdb) - I wonder if with older OS it would be recognized at all, and if having two rules files for same VID:PID pairs would confuse the system. Just something to keep in mind for troubleshooting.

Anyhow, after such installation you'd need to `systemctl restart udev` (or something like that), and probably unplug-replug your USB cable for the UPS to get it recognized and permissions assigned.

joesitsolutions, 2022-11-06:
So an update. I gave the machine a reboot and retried both `blazer_usb` and `nutdrv_qx` and I don't think either worked. Both seem to output a bunch of text and fail. I also don't seem to have `udev` on the machine. I feel like I may need to install it but am unsure. Please see the following pastebin with the results from my testing. [PasteBinLink](https://pastebin.com/zX0gb1pQ)
Thanks for the advice thus far. The UPS units I have are quite old and may not be supported (I have replaced the batteries a few times and they're running well).

jimklimov, 2022-11-06:
Thanks for the debugs. Seems the device does talk some strange variant of Qx protocol. Seems garbled, so either a dialect I haven't seen yet, or wire/connector/... is noisy, or some other process reads part of the bytes (at least, outputs of two competing readers on serial ports seem awkward sort of like this).

Still, with many changes and new subdrivers in nutdrv-qx over the years, a newer build has a chance to help.

joesitsolutions, 2022-11-08:
I have a set of weird UPS units for sure. I can set this in settings also:
[image]
I'm not sure if perhaps another setting could yield different results but I'll have a tinker later. The cable is a USB 2.0 A-B Printer Cable but there is also a weird megatec console connector that I can try to connect.