networkupstools/nut #1707: Device dump for Maruson TAC-LV1K

jlg89, 2022-11-14:
[Maruson__TAC-LV1K__nutdrv_qx__2.7.4__01.txt](https://github.com/networkupstools/nut/files/10006778/Maruson__TAC-LV1K__nutdrv_qx__2.7.4__01.txt)

jimklimov, 2022-11-15:
Thanks! From cursory look at the data, seems the device support is partial: `ups.temperature` is not set (and yet not omitted), and `battery.voltage` is not adjusted by `battery.packs` to report something around 36V in this case, I guess. The latter was researched in #1279 (PR #1652 started but not satisfactorily completed/tested so far).

jlg89, 2022-11-15:
Right, it's pretty basic. I had to use a lot of the workarounds at https://networkupstools.org/docs/man/nutdrv_qx.html to get sane battery values.