networkupstools/nut #1712: Unable to start nut-driver after upgrade to Fedora 37

andrewvillano, 2022-11-22:
nut-driver.service as well as upsdrvctl is unable to start any of my snmp-ups devices individually or collectively. 

These are the errors I'm seeing:

[root@xxxxx run]# upsdrvctl start nutdev7
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic SNMP UPS driver 1.21 (2.8.0)
writepid: fopen /var/run/snmp-ups-nutdev7.pid: Permission denied
Detected Smart-UPS 2200 on host 10.100.2.49 (mib: apcc 1.6)
[nutdev7] Warning: excessive poll failures, limiting error reporting (OID = .1.3.6.1.4.1.318.1.1.1.9.2.3.1.5.1.1.3)
[nutdev7] Warning: excessive poll failures, limiting error reporting (OID = .1.3.6.1.4.1.318.1.1.1.9.3.3.1.6.1.1.1)

Fatal error: unable to create listener socket

bind /var/run/snmp-ups-nutdev7 failed: Permission denied

Exiting.
Driver failed to start (exit status=1)
[root@xxxxx run]# 

Thanks in advance.

jimklimov, 2022-11-22:
That seems to have been raised just recently on the mailing list and fedora bug tracker, and boils down to wrong packaging (no rights to write into `/var/run`, there should be a NUT-owned subdirectory):

* https://alioth-lists.debian.net/pipermail/nut-upsuser/2022-November/012997.html
*  https://bugzilla.redhat.com/show_bug.cgi?id=2127269

Thanks to @svarshavchik for letting us know.

jimklimov, 2022-11-25:
FYI: After a bit more digging in current codebase, I can say that NUT unprivileged daemons (drivers and upsd) default to using the `ALTPIDPATH` in fact -- which in turn defaults to `STATEPATH` e.g. `/var/state/ups` since both daemon types exchange pipe files there so may write there, but some package recipes `configure` it to e.g. `/var/run/nut`.

This can be seen by code lines preparing a `pidfn` (usually) with `altpidpath()` (see `common/common.c`) in `drivers/main.c`, `drivers/upsdrvctl.c` and `server/upsd.c`.

And then it gets messy in same `common/common.c` with:
* `writepid()` which *optionally* uses `PIDPATH` (as defined, no `./nut` subdir) if file `name` is not absolute, and with
* `sendsignal()` which combines `PIDPATH` (as defined, no `./nut` subdir) and the `progname` to wrap `sendsignalfn()` (which reads PID from that absolute filename and signals the process in OS-dependent manner).

These latter usages imply that `PIDPATH` is expected to be writable by `nut` processes and should not be the root-secured system location (e.g. `/var/run` directly) unless NUT tools and daemons run as `root`; can be `/tmp` however.

I'll post a clean-up PR to clarify this in `configure` script comments and docs, and "reference" init-scripts and packaging templates which use `$PIDPATH/nut` to confuse matters more.

Also linking to #123 which creeps out to other big discussions on PID files :)