networkupstools/nut #1733: winNUT-Client Error/Bug

Samir-driod, 2022-12-15:
[Crash_Report_14-12-2022_21-34-11.txt](https://github.com/networkupstools/nut/files/10239931/Crash_Report_14-12-2022_21-34-11.txt)

I have an Error When I want to start a Client. I reinstall it and it didn't work.
Can En-one help.

Link To Crash_Report: 
[Crash_Report_14-12-2022_21-34-11.txt](https://github.com/networkupstools/nut/files/10239931/Crash_Report_14-12-2022_21-34-11.txt)

jimklimov, 2022-12-16:
WinNUT (which variant BTW?) is a community project, so can't say much about it. Maybe post on their issue tracker too...

FWIW did you try NUT for Windows builds (e.g. recent master from Appveyor CI)?

jimklimov, 2022-12-21:
@Samir-driod : seems like https://github.com/gawindx/WinNUT-Client/issues/138 is about the same issue - and a workaround.

closed 2022-12-21 by jimklimov