networkupstools/nut #1735: Windows 10 Nut Server

Samir-driod, 2022-12-21:
I have the "APC bk650". I try on my Linux Server and it works on first time. I reinstall everything(nut Server, drivers and uninstall the ups app). I check the config, It was fine.
When I wanna to run it I have this:

 Network UPS Tools - UPS driver controller Windows-v2.6.5-5-7-g72f380c
Network UPS Tools - Generic HID driver 0.38 (Windows-v2.6.5-5-7-g72f380c)
USB communication driver 0.32
No matching HID UPS found
Driver failed to start (exit status=1)

I don't know what i need to do.
Can Anyone help, please

jimklimov, 2022-12-21:
Gor comparison, what version NUT do you have on Linux server? The 2.6.5 based Windows release (I believe that's what you want to keep as the NUT data server?) are over a dozen years old now so might just not know about data points including VendorID:ProductID for that UPS. You can try to specify those IDs in ups.conf section for the device.

The NUT for Windows effort was revived so current master branch can be built (numerous incompletenesses are documented) and recent master builds can be found on Appveyor CI, but work stalled at reviving the installer - in particular to add the privileged USB driver files. So for now "as is" it may help with Windows clients and serve networked UPSes, less so with directly attached devices.

Samir-driod, 2022-12-22:
When I wanna to use "nut-scanner.exe" it give me this: 

Cannot load USB library (libusb) : The specified module could not be found.. USB search disabled.
Cannot load SNMP library (libnetsnmp) : The specified module could not be found.. SNMP search disabled.
Cannot load XML library (libneon) : The specified module could not be found.. XML search disabled.
Cannot load NUT library (libupsclient) : The specified module could not be found.. NUT search disabled.

Can You Help me

jimklimov, 2022-12-22:
Interesting... did you unpack that tarball from Appveyor CI, keeping the directory structure and all, and run the program from there? Specifically, we are not talking now about the antique build from 2.6.5-based release?

* https://ci.appveyor.com/project/nut-travis/nut/build/artifacts

jimklimov, 2023-01-02:
Hm, it seems that for the tarballs from Appveyor some DLL names are not getting matched (numbered version suffixes for SNMP and NEON DLLs are not what is expected by code - will fix). However two others (USB and NUT DLLs) are linked by nut-scanner okay.

PR #1751 (2023-01-02): nutscan-init.c: adjust WIN32 shared library detection for SNMP and NEON
Closes: #1735

Note: libnetsnmp is not bundled in mingw, but can be built additionally as documented in NUT sources - and automated for Appveyor CI builds.
closed 2023-01-02 by jimklimov