networkupstools/nut #1744: riello_usb multiple identical ups USB issue

shgubar, 2022-12-26:
I have several Riello UPS connected via USB to RPi4. Only one of them works, that is, the problem with all the others. At the same time, I know which bus they are on, as well as the device number.

Is it possible to make device identification for this driver (I saw other posts with similar problems, but for other drivers)?

### **UPS info:**
Riello SEP 1000VA ER

### **RPi4 info:**
```
System: Raspbian Buster
Linux rasp4 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
```

### **NUT Driver info:**
```
riello_usb -h (compiled from nut-master):
Network UPS Tools - Riello USB driver 0.07 (2.8.0.1)
```
### **lsusb:**
```
Bus 001 Device 004: ID 04b4:5500 Cypress Semiconductor Corp. HID->COM RS232 Adapter
Bus 001 Device 003: ID 04b4:5500 Cypress Semiconductor Corp. HID->COM RS232 Adapter
```

### **ups.conf:**
```
[riello1000]
         driver = riello_usb
         port = auto
         #port = /dev/bus/usb/001/003
         desc = "Riello SEP 1000 ER"
         #device = 003
         #vendor = 04b4
         #productid = 5500
         pollinterval = 15
 
[rps1000]
        driver = riello_usb
        #port = /dev/bus/usb/001/004
        port = auto
        desc = "Riello SEP 1000 ER NP"
        #device = 004
        #vendor = 04b4
        #productid = 5500
        pollinterval = 15
```

### **riello_usb -a riello1000 -DD:**

```
Network UPS Tools - Riello USB driver 0.07 (2.8.0.1)
Warning: This is an experimental driver.
Some features may not function correctly.

    0.000000 [D1] debug level is '2'
    0.001899 [D1] Succeeded to become_user(nut): now UID=1002 GID=1002
    0.112263 [D2] Checking device (1D6B/0003) (002/001)
    0.181240 [D1] libusb_open get iManufacturer failed, retrying...
    0.181305 [D1] libusb_open get iManufacturer failed, retrying...
    0.181357 [D1] libusb_open get iManufacturer failed, retrying...
    0.181407 [D1] libusb_open get iProduct failed, retrying...
    0.181462 [D1] libusb_open get iProduct failed, retrying...
    0.181512 [D1] libusb_open get iProduct failed, retrying...
    0.181560 [D1] libusb_open get iSerialNumber failed, retrying...
    0.181607 [D1] libusb_open get iSerialNumber failed, retrying...
    0.181658 [D1] libusb_open get iSerialNumber failed, retrying...
    0.181709 [D2] - VendorID: 1d6b
    0.181756 [D2] - ProductID: 0003
    0.181799 [D2] - Manufacturer: unknown
    0.181842 [D2] - Product: unknown
    0.181884 [D2] - Serial Number: unknown
    0.181931 [D2] - Bus: 002
    0.181977 [D2] - Device: 001
    0.182020 [D2] - Device release number: 0510
    0.182062 [D2] Trying to match device
    0.182109 [D2] Device does not match - skipping
    0.182565 [D2] Checking device (04B4/5500) (001/004)
    0.189681 [D2] - VendorID: 04b4
    0.189738 [D2] - ProductID: 5500
    0.189787 [D2] - Manufacturer: Cypress Semiconductor
    0.189837 [D2] - Product: USB to Serial
    0.189885 [D2] - Serial Number: unknown
    0.189934 [D2] - Bus: 001
    0.189982 [D2] - Device: 004
    0.190033 [D2] - Device release number: 0001
```

By the way, when '_nut-server_' is started, files are created only for one (the last one from the list UPS):
```
srw-rw---- 1 nut nut 0 дек 26 14:17 riello_usb-rps1000=
-rw-r--r-- 1 nut nut 4 дек 26 14:17 riello_usb-rps1000.pid
```
As far as I understand, it should be created for everyone UPS.

Please help to connect multiple devices.
Thanks.

jimklimov, 2022-12-26:
Regarding the second question - I suppose logs would show that one driver started earlier, looked for "auto" port by pid/vid and took it. Another was a bit too late, selected same port and could not take it (device busy) so aborted and did not leave a pipe file.

As for other selection options, I think libusb does not care about device paths (or not on all OSes, or libusb versions?) so picking devices historically went by checking a list of discovered device descriptors suggested by libusb and somehow matching one to take. Generally up to 6-7 fields can be checked, depending on driver and what the devices expose (vid/pid, names of the same, model, bus, serial number...)

jimklimov, 2022-12-26:
It might make sense to add an option to keep iterating if a device is busy, if a user knows they have two of the same - but the result is not deterministic unless you have a unique value to match. And then you won't need an option.

In some cases you do care to know if e.g. one or two UPSes are alive for the dual-PSU server but don't really care which one is down. So PRs are welcome, should not be hard to add such option :)

shgubar, 2022-12-27:
Thanks Jim for your reply.

In my opinion, a driver file should be created for each connected device, because otherwise there is no way to read that the device is in the system.

Regarding adding an iteration, it wouldn't be bad, but can it loop, then simply no device will be connected and the program will go to the disabled state. By the way, I still don't understand how to do it correctly, where to look and what to add.

Regarding the options: I think it would be a good idea to add them to this particular driver to be able to separate each device. This is how, for example, it is done for the driver - _nutdrv_qx_, and here is the discussion of the problem: https://github.com/networkupstools/nut/issues/1174

jimklimov, 2022-12-27:
> in my opinion, a driver file should be created for each connected device, because otherwise there is no way to read that the device is in the system.

Agreed. My point is that *maybe* according to detection method used currently, it can't tell you have two devices. Vid:Pid matched - check. Busy? abort!

shgubar, 2022-12-27:
From what I can see, _Vid = 04b4_, _Pid = 5500_ are the same for both. And for some reason that is not clear to me, when starting the driver, the first one is connected, the one with _Device ID 003_ (**/dev/bus/usb/001/003**), but the second one (**rsp1000**) is created from the list - _riello_usb-rps1000.pid_.

In my opinion, if indicate the number of the device for each, regardless of the fact that they are both on the same bus, then each should be connected in the form of an iteration.

jimklimov, 2023-01-02:
Linking to #1414 and #1565 to keep related bugs together :)

jimklimov, 2023-01-02:
Well, in the config posted above I do see `device` lines  but they are commented away - so unless something changed invisibly to me, they are not used in matching.

shgubar, 2023-01-02:
> Well, in the config posted above I do see `device` lines but they are commented away - so unless something changed invisibly to me, they are not used in matching.

This is just to show that I tried to try to add these options, but I get that the driver doesn't support it

jimklimov, 2023-01-04:
So thanks for nudging me to revise/retrace the code more carefully, I hope the PR above (merged now) should fix at least the part about driver options for USB matching.

shgubar, 2023-01-04:
So now I can try to download the updated branch (nut-master) and I should see the changes?
OK, I'll try and report the result, probably already in this ticket because I see you closed https://github.com/networkupstools/nut/issues/1754.

jimklimov, 2023-01-05:
Now you can also check https://github.com/jimklimov/nut/tree/issue-1756 (source of the PR above) which includes recent `master` branch fixes and adds an `allow_duplicates` flag that you can set in each device config section (in ups.conf) where you are feeling adventurous enough.

Would be interesting to know if it actually works on a system with two real similar devices (chaotically for "who is who", but without crashes).

shgubar, 2023-01-05:
Just yesterday, I tried a new brunch with the following configuration `ups.conf`:

```
[riello1000]
        driver = riello_usb
        port = auto
        desc = "Riello SEP 1000 ER"
        vendorid = 04b4
        productid = 5500
        bus = 001
        device = 004
        pollinterval = 15

[rps1000]
        driver = riello_usb
        port = auto
        desc = "Riello SEP 1000 ER NP"
        vendorid = 04b4
        productid = 5500
        bus = 001
        device = 003
        pollinterval = 15
```

I restarted the service several times, as well as the entire Raspberry - everything works as it should.
It has been working for more than 12 hours.

The question is, what exactly does this parameter give - `allow_duplicates` - how might this affect my configuration?

P.S.: if you need to test and give a result - I will do it without problems.

jimklimov, 2023-01-05:
Sounds great, thanks for the report!

So if you only used the configuration above, it means the fix from #1763 works and allows `riello_usb` to recognize and use the common USB matching parameters, and by the bus/device values it uniquely identifies your UPSes (hopefully the numbering would remain stable until you re-plug something, change kernel drivers, etc.) 

Matching by serial number might be more reliable, hopefully the empty string can be requested like `serial = "^$"` for one of your devices which did not serve it, to differentiate from the other device which did.

Hopefully you would not need `allow_duplicates` in production now, but for testing purposes it would look something like this config:
````
[riello1000]
        driver = riello_usb
        port = auto
        desc = "Riello SEP 1000 ER"
        vendorid = 04b4
        productid = 5500
        allow_duplicates
        #bus = 001
        #device = 004
        pollinterval = 15

[rps1000]
        driver = riello_usb
        port = auto
        desc = "Riello SEP 1000 ER NP"
        vendorid = 04b4
        productid = 5500
        allow_duplicates
        #bus = 001
        #device = 003
        pollinterval = 15
````

Now each of the two driver processes would only match the `vendorid=04b4 && productid=5500` as they did before the fix. This seems to be a common situation for devices that do not have (unique) serial numbers, have same vendor and product IDs and strings, and matching by bus/device is not stable or desirable.

Here like when you started with this problem, one of the NUT driver processes would get the first listed device, another would be denied access to it (because Resource Busy), but thanks to new `allow_duplicates` option it would not abort but would continue iterating and would find the second matching device.

Downside is that this depends on race between two driver starts, so you do not really know which driver instance monitors which physical device - it may be different for every restart - but depending on deployment situation you might not care. "I have a server with 2 PSUs fed from different UPSes; as long as any one is online, all is good".

shgubar, 2023-01-05:
Just uploaded the latest brunch. Changed mine `ups.conf` as you indicated:
```
Jan  5 18:16:10 rasp4 nut-driver@riello1000[19193]: Fatal error: 'allow_duplicates' is not a valid flag for this driver.
Jan  5 18:16:10 rasp4 nut-driver@rps1000[19194]: Fatal error: 'allow_duplicates' is not a valid flag for this driver.
```

shgubar, 2023-01-05:
> Matching by serial number might be more reliable, hopefully the empty string can be requested like `serial = "^$"` regex for one of your devices which did not serve it, to differentiate from the other device which did.

Both show:
`Serial Number: unknown`

jimklimov, 2023-01-05:
As for `allow_duplicates` - it is not yet merged to master, CI hiccupped so retrying; see https://github.com/networkupstools/nut/pull/1770 and its source branch for now... may get to master in a few hours though. Which "brunch" did you taste?

Sad about Serial field - guess the driver gets whatever ot reports to upsc screenshots from some other report than what is checked for generic device matching...

shgubar, 2023-01-06:
Please, tell me how to see the brunch number/code in the folder `nut-master`?

jimklimov, 2023-01-06:
`git branch` and `git log -1`

To get new commits from GitHub (recent days were quite intensive in merging new features), you can update the workspace with `git pull --all`

shgubar, 2023-01-06:
I did not use `git` during the entire testing period.
This is how I downloaded:
`wget https://github.com/networkupstools/nut/archive/refs/heads/master.zip`

shgubar, 2023-01-06:
Just installed via `git`, apparently I used:

```
commit 48d8be34ee989fa1b89055a89776928b7f6a6dd8 (HEAD -> master, origin/master, origin/HEAD)
Merge: 1e2a46275 ef9857965
Author: Jim Klimov <jim@jimklimov.com>
Date: Fri Jan 6 00:29:08 2023 +0100

     Merge pull request #1773 from efuss/qx-masterguard
    
     Add experimental prefix missed in 7ef2874e81
```

jimklimov, 2023-01-06:
Ok, that seems quite recent, but was the master branch so not the one with https://github.com/jimklimov/nut/tree/issue-1756 (PR #1770 was still open at the moment of this writing - but passed CI so merging now).

shgubar, 2023-01-09:
Jim, look at the situation, if I use `allow_duplicates`, each UPS really connects now, but if the first one is riello1000 (as it should be) in `ups.conf` - then everything works as it should, but if I install `riello1000` the second one and `rps1000` first, the sequence changes and the indicators are already incorrect.
It is more profitable for me to use `bus` and `device`.
Thanks for your help.

jimklimov, 2023-01-09:
Thanks for the report.

So can I take it as confirmation that without special matchers (bus, device) the new `allow_duplicates` option does allow two physical devices to be managed - which otherwise are seen as "identical" (when only vendorid/productid are matched) and so second driver did not start? And that this is not quite deterministic (configuration section names do not reliably map to certain devices)? These aspects are expected, so if it does work like this - the goals of #1756/#1770 are achieved.

And yes, "proper" deterministic matching where possible is "more profitable". Alas, not with all OSes and/or devices it is possible, so this fallback seemed useful to make the best outcome from a bad situation.

jimklimov, 2023-01-09:
Yes, thanks - this and other PRs related to the issue were merged to master, so its current codebase should be useful for the tests :)

jimklimov, 2023-01-09:
Great, thanks for testing! :)

closed 2023-01-09 by jimklimov