networkupstools/nut #1745: Powerwalker VI 1000 RT doesnt works

warp735, 2022-12-26:
According to the compatibility list, blazer_usb should be used. Then there is no data / connection at all. If I go to autodetect (unraid server), comes usbhid_ups and also data. However, I get a rhythm every 5 minutes that the UPS has failed. If I switch off the power, the message only comes up after 5 minutes.

jimklimov, 2022-12-26:
Which NUT version? 2.8.0/master or antique?

It is possible that different devices (or firmware capability generations) are marketed under the same name. Blazer would be a Megatec Qx protocol, so nutdrv_qx could also work. It is very different from usbhid though.

warp735, 2022-12-27:
Is the version 2.7.1
With usbhid the right data or realistic values ​​come. Only the connection is probably not kept clean.
I can't select nutdrv_qx with the unraid plugin.

jimklimov, 2022-12-27:
Ugh, 2.7.1 is what, a decade old? Can you (as a user/community member) nag unraid to step into this millenium? ;)