networkupstools/nut #1747: Tripplite SMART1500RM2UN missing output data

timjp692, 2022-12-28:
First time posting, I did not see any current issues reporting this when I searched.

Missing output voltage and output power

# Device dump generated by ./nut-ddl-dump.sh on Wed, 28 Dec 2022 03:11:37 +0000
# upsrw output:
# [battery.charge.low]
# Remaining battery level when UPS switches to LB (percent)
# Type: STRING
# Maximum length: 5
# Value: 10
#
# [input.transfer.high]
# High voltage transfer point (V)
# Type: STRING
# Maximum length: 5
# Value: 0.0
#
# [input.transfer.low]
# Low voltage transfer point (V)
# Type: STRING
# Maximum length: 5
# Value: 0.0
#
# [ups.delay.shutdown]
# Interval to wait after shutdown with delay command (seconds)
# Type: STRING
# Maximum length: 10
# Value: 20
#
# [ups.delay.start]
# Interval to wait before (re)starting the load (seconds)
# Type: STRING
# Maximum length: 10
# Value: 30
# upscmd output:
# Instant commands supported on UPS [ups]:
#
# beeper.disable - Disable the UPS beeper
# beeper.enable - Enable the UPS beeper
# beeper.mute - Temporarily mute the UPS beeper
# beeper.off - Obsolete (use beeper.disable or beeper.mute)
# beeper.on - Obsolete (use beeper.enable)
# load.off - Turn off the load immediately
# load.off.delay - Turn off the load with a delay (seconds)
# load.on - Turn on the load immediately
# load.on.delay - Turn on the load with a delay (seconds)
# reset.watchdog - Reset watchdog timer
# shutdown.reboot - Shut down the load briefly while rebooting the UPS
# shutdown.return - Turn off the load and return when power is back
# shutdown.stayoff - Turn off the load and remain off
# shutdown.stop - Stop a shutdown in progress
# test.battery.start.deep - Start a deep battery test
# test.battery.start.quick - Start a quick battery test
# test.battery.stop - Stop the battery test

# upsc output:
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 30
battery.runtime: 1378
battery.temperature: 30.9
battery.type: PbAC
battery.voltage: 40.9
battery.voltage.nominal: 36.0
device.mfr: Tripp Lite
device.model: TRIPP LITE SMART1500RM2UN
device.serial: 2726DY0SM887600561
device.type: ups
driver.name: usbhid-ups
[... 104 lines cut ...]
  12.364402     [D5] Exponent = 0
  12.364440     [D2] Path: UPS.PowerSummary.PresentStatus.FullyCharged, Type: Feature, ReportID: 0x32, Offset: 19, Size: 1, Value: 1
  12.364478     [D5] hu_find_infoval: found fullycharged (value: 1)
  12.364543     [D5] process_boolean_info: fullycharged
  12.364584     [D3] Report[buf]: (4 bytes) => 32 00 00 09
  12.364630     [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
  12.364696     [D5] Unit = 00000000, UnitExp = 0
  12.364732     [D5] Exponent = 0
  12.364777     [D2] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x32, Offset: 20, Size: 1, Value: 0
  12.364833     [D5] hu_find_infoval: found !chrg (value: 0)
  12.364890     [D5] process_boolean_info: !chrg
  12.364931     [D3] Report[buf]: (4 bytes) => 32 00 00 09
  12.364984     [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
  12.365009     [D5] Unit = 00000000, UnitExp = 0
  12.365047     [D5] Exponent = 0
  12.365083     [D2] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x32, Offset: 21, Size: 1, Value: 0
  12.365127     [D5] hu_find_infoval: found !dischrg (value: 0)
  12.365188     [D5] process_boolean_info: !dischrg
  12.365217     [D3] Report[buf]: (4 bytes) => 32 00 00 09
  12.365250     [D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
  12.365312     [D5] Unit = 00000000, UnitExp = 0
  12.365378     [D5] Exponent = 0
  12.365429     [D2] Path: UPS.PowerSummary.PresentStatus.FullyDischarged, Type: Feature, ReportID: 0x32, Offset: 22, Size: 1, Value: 0
  12.365476     [D5] hu_find_infoval: found !depleted (value: 0)
  12.365525     [D5] process_boolean_info: !depleted

jimklimov, 2023-01-02:
CC @dzomaya @aquette

timjp692, 2023-01-18:
This might also be an issue with the UPS, I do not see a load value in the web interface (WEBCARDLX)
I also do not see it being read from the USB interface. I am going to try and find a good time to do a ups reboot.

timjp692, 2023-02-15:
You guys can close this if you want. Seems the ups likes to stop reporting load to even the network card. Thanks for all the work you do for Nut.