networkupstools/nut #1749: CyberPower VP1200ELCD password required

Daverkex, 2022-12-31:
Test command:
```
root@server[~]# upscmd -l ups
Instant commands supported on UPS [ups]:

beeper.disable - Disable the UPS beeper
beeper.enable - Enable the UPS beeper
beeper.mute - Temporarily mute the UPS beeper
beeper.off - Obsolete (use beeper.disable or beeper.mute)
beeper.on - Obsolete (use beeper.enable)
load.off - Turn off the load immediately
load.off.delay - Turn off the load with a delay (seconds)
load.on - Turn on the load immediately
load.on.delay - Turn on the load with a delay (seconds)
shutdown.return - Turn off the load and return when power is back
shutdown.stayoff - Turn off the load and remain off
shutdown.stop - Stop a shutdown in progress
test.battery.start.deep - Start a deep battery test
test.battery.start.quick - Start a quick battery test
test.battery.stop - Stop the battery test
```
Try a battery test:
```
root@server[~]# upscmd ups test.battery.start.deep
Username (root): 
Password: 
Unexpected response from upsd: ERR PASSWORD-REQUIRED
```

Using the most common login:
```
root@server[~]# upscmd -u cyber -p cyber ups test.battery.start.deep
Unexpected response from upsd: ERR ACCESS-DENIED
```
I tried other login combinations without luck. Does someone know the right credentials?

jimklimov, 2022-12-31:
Um, what credentials did *you* set up in `upsd.users` file? (and did you secure it on FS so `upsd` dies not ignore it)?

Daverkex, 2023-01-05:
Hi @jimklimov as you can see in my last execution I pass directly the credentials. My question isn't about NUT configuration directly, it's mostly about the default credentials for this UPS unit. Do you think it's important the `upsd.users` and the permissions to run the command directly with the credentials in the prompt?

jimklimov, 2023-01-05:
There are no credentials of an UPS unit, as a hardware box. At least not for serial/usb links (your post does not detail further). Even if SNMP or NetXML or other protocols with auth are used, that's a minor detail of driver configuration - how it talks to device.

NUT client programs like upscmd deal with NUT auth - which you define, see docs or *.conf.sample files, please.