networkupstools/nut #1750: Installation docs depreciated

kondas, 2023-01-01:
The documentation (INSTALL.nut) is depreciated. The mentioned "configure" script is not available anymore. The Install / Build procedure is not possible following the instructions. Please provide updated instructions for manual install.

jimklimov, 2023-01-01:
As elsewhere, you use autoconf to generate `configure` script for your system from `configure.ac` template (result is included in tarballs and matches the instruction, to reduce build-time dependencies but risking portability to an extent). In NUT this is automated by `autogen.sh` helper script.

kondas, 2023-01-01:
Thank You for the clarification, i'm not familiar with `autoconf` (yet). I will make a suggestion to modify INSTALL.nut to be more comprehendable for the layman like me in a PR soon. Happy new year!

jimklimov, 2023-01-02:
Note: I've updated the text - see https://github.com/networkupstools/nut/pull/1752/files

If you think further (yet compact) layman refining would be useful, please do post a PR :)

PR #1752 (2023-01-02): Update docs wording and hints per user feedback
Closes: #1750 
Closes: #1721
Relates-to: #1736
closed 2023-01-02 by jimklimov