networkupstools/nut #1754: The problem is working with several UPS at the same time, connected via USB (Instructions, Setup, Logs)

shgubar, 2023-01-02:
Hi to all

I've been trying to figure out for a few days what I'm doing wrong with the installation and subsequent setup of version 2.8.0. The result is the same every time.

At first, I even thought that the problem was in the riello_usb driver, that it had no parameters: bus, device, and so on. Even created a post about it - https://github.com/networkupstools/nut/issues/1744. But then I found out that the package can work without them.

I believe that this post will be relevant to many people, because many problems that I have had to deal with, have not been written about anywhere. I used the search, but I did not find the answer, only by the method of my mistakes and reading the logs, I discovered the necessary actions to prevent problems with the NUT.

I have several UPSs connected to a Raspberry Pi4 4GB via USB. The installation was performed on a clean system by Buster. The only thing that I immediately changed for my needs was the Shell (tcsh), but I also used the default (bash). The behavior is as follows when I start it for the first time - everything works as it should: it finds both devices, drivers are connected, all services are started. But it is enough to reboot only once, as the struggle for the USB port immediately begins. Below I will show all the necessary data.

Immediately the question:
1. Am I doing everything right when installing the package?
2. Am I setting it up correctly to run NUT?
3. Why does each driver "search" for the same port after system reboot?

### **First about the system.**

**cat /sys/firmware/devicetree/base/model**
`Raspberry Pi 4 Model B Rev 1.1
`

**uname -a**
`Linux rasp4 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux
`

**cat /etc/os-release**
```
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
```

**lsusb**
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 04b4:5500 Cypress Semiconductor Corp. HID->COM RS232 Adapter
Bus 001 Device 003: ID 04b4:5500 Cypress Semiconductor Corp. HID->COM RS232 Adapter
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

### **Models of my UPSs.**
```
Name: Riello Sentinel Pro 1000 ER
Firmware: SWM073-01-00 & SWM073-01-01
ProductID: 5500
VendorID: 04b4
```

### **Next, I want to show step by step how I install and configure NUT. The type of instruction is similar to what is written in the Wiki, but a little easier (although I tried to install it as it was proposed).**

**1. Install packages:**
`	apt-get update && apt-get install -y build-essential autoconf gettext libusb-dev libtool libgd-dev
`

**2. Get 'nut-master':**
```
	wget https://github.com/networkupstools/nut/archive/refs/heads/master.zip
	unzip master.zip
	cd nut-master
```

**3. Create new group 'nut':**
`	groupadd --system nut
`

[... 332 lines cut ...]
Jan  2 14:28:42 rasp4 nut-server[647]: Running as foreground process, not saving a PID file
Jan  2 14:28:42 rasp4 nut-server[647]: User monitor@127.0.0.1 logged into UPS [riello1000]
Jan  2 14:28:42 rasp4 upsd[647]: Running as foreground process, not saving a PID file
Jan  2 14:28:42 rasp4 upsd[647]: User monitor@127.0.0.1 logged into UPS [riello1000]
Jan  2 14:28:42 rasp4 nut-monitor[648]: Poll UPS [riello1000@localhost] failed - Driver not connected
Jan  2 14:28:42 rasp4 nut-monitor[648]: Communications with UPS riello1000@localhost lost
Jan  2 14:28:42 rasp4 nut-monitor[648]: Network UPS Tools upsmon 2.8.0.1
Jan  2 14:28:42 rasp4 systemd[1]: systemd-update-utmp-runlevel.service: Succeeded.
Jan  2 14:28:42 rasp4 systemd[1]: Started Update UTMP about System Runlevel Changes.
Jan  2 14:28:42 rasp4 systemd[1]: Startup finished in 2.220s (kernel) + 14.033s (userspace) = 16.254s.
Jan  2 14:28:43 rasp4 riello_usb[646]: sock_connect: enabling asynchronous mode (auto)
Jan  2 14:28:47 rasp4 nut-monitor[648]: Poll UPS [riello1000@localhost] failed - Driver not connected
Jan  2 14:28:47 rasp4 nut-monitor[648]: UPS riello1000@localhost is unavailable
Jan  2 14:28:47 rasp4 nut-monitor[648]: wall: cannot get tty name: Неприменимый к данному устройству ioctl
Jan  2 14:28:47 rasp4 nut-monitor[648]: Network UPS Tools upsmon 2.8.0.1
Jan  2 14:28:52 rasp4 nut-monitor[648]: Poll UPS [riello1000@localhost] failed - Driver not connected
Jan  2 14:28:55 rasp4 systemd[1]: nut-driver@riello1000.service: Service RestartSec=15s expired, scheduling restart.
Jan  2 14:28:55 rasp4 systemd[1]: nut-driver@riello1000.service: Scheduled restart job, restart counter is at 1.
Jan  2 14:28:55 rasp4 systemd[1]: Stopped Network UPS Tools - device driver for riello1000.
Jan  2 14:28:55 rasp4 systemd[1]: Starting Network UPS Tools - device driver for riello1000...
```

Please, I'm really begging for someone's feedback, I'm just stuck and don't know how to proceed. Any information I can provide.

Thanks for paying attention, everyone

jimklimov, 2023-01-02:
Thanks for the very detailed write-up :) I think I'll comment as I read it, going along...

One thing that seems odd is about `upsmon.conf` MONITOR lines:
````
MONITOR riello1000@localhost 1 monitor ***** primary
MONITOR rps1000@localhost 2 monitor ***** secondary
````

The numbers are about how many power sources of the current computer are fed by that device. Unless your rPi sits on an ATS/STS, it probably has one PSU so would be "1" on one UPS (if fed from it) and "0" on other(s). Although Pi's are fun to set up, a HAT like PiJuice might add other power inputs (different USB, solar/battery, etc.) so it is feasible that a Pi really is fed from several different UPSes at once... Unless they somehow conflict on its power bus and burn it - not sure about the wiring there ;-\

Another odd thing nearby is `RUN_AS_USER ups` - earlier you created a `nut` user, so I suppose `ups` does not exist and/or directory permissions (for PID file etc.) were not tweaked for one; or did you not list some details after all?

jimklimov, 2023-01-02:
Also note the permissions warnings about files potentially with passwords and other sensitive data (ups.conf, upsd.conf, upsd.users, upsmon.conf), they should only be accessible to daemon accounts - usually as RW for `root` user and RO for `nut` group. It may be possible that NUT daemons would refuse to use files that are not secure.

jimklimov, 2023-01-02:
Also wondering how this worked before first reboot:
````
	systemctl enable nut-driver@riello1000.service
	systemctl enable nut-driver@rps1000.service
````

Such units would be generated (and enabled unless you pre-masked them) by `nut-driver-enumerator` script or service.

I guess you experimented several times so a copy of that service ran and picked up edition events of the ups.conf file?..

jimklimov, 2023-01-02:
Overall, it is really encouraging that at least initially the two drivers did run together and detected different instances of the devices (was it deterministic though - did "riello1000" always grab "SWM073-01-00", and did "rps1000" always grab "SWM073-01-01"; did it reliably succeed at all - if you ran the experiment several times?)

* *Maybe* libusb does not follow a deterministic detection/listing order for some reason, so in the "first reboot" two drivers got differently sorted lists of USB devices to look at - so for each one the first hit happened to be unique?

Alas, the behavior in the end of the post seems to match my expectations from detection logic described in #1744 - that to monitor several devices you need to match some unique set of their properties. In your case serial numbers are different (empty in one, nontrivial in the other) so that may help.

Another possibility is that `udev` or similar facility did not start (or do its job properly) before NUT drivers ran for the last posted reboot, so the USB device nodes were held by the kernel (or another default OS handler) at that moment.

jimklimov, 2023-01-02:
Would some progress on #1756 address the problem? At least it is something we *can* try to avoid the stalemate...

shgubar, 2023-01-02:
> One thing that seems odd is about `upsmon.conf` MONITOR lines:
> ```
> MONITOR riello1000@localhost 1 monitor ***** primary
> MONITOR rps1000@localhost 2 monitor ***** secondary
> ```

Sorry, it's my fault. Thanks for correct.

> The numbers are about how many power sources of the current computer are fed by that device. Unless your rPi sits on an ATS/STS, it probably has one PSU so would be "1" on one UPS (if fed from it) [...]

The connection is as follows: I have two UPS - this RPi is connected to the power supply, as well as a gateway, a switch and Wi-Fi to the first, and more energy-intensive devices are connected to the second. Both UPSs are connected via USB to the RPi, where I want to take measurements.

> The `RUN_AS_USER ups` should not be needed since it points to same user name as compiled in (via `--with-user ups` config option).

Agreed.

shgubar, 2023-01-02:
> Also note the permissions warnings about files potentially with passwords and other sensitive data (ups.conf, upsd.conf, upsd.users, upsmon.conf), they should only be accessible to daemon accounts - [...]

I tried to replace it - the situation remained unchanged.

shgubar, 2023-01-02:
> Also wondering how this worked before first reboot:

I never did without rebooting:

systemctl start <all services>
- nut.target
- nut-driver.target

> ```
> 	systemctl enable nut-driver@riello1000.service
> 	systemctl enable nut-driver@rps1000.service
> ```
> Such units would be generated (and enabled unless you pre-masked them) by `nut-driver-enumerator` script or service.

Immediately after installing the package, these services are visible - without _any_ startup - so it was done as above.

> I guess you experimented several times so a copy of that service ran and picked up edition events of the ups.conf file?..

I'm not sure if I understood you correctly, but - yes, of course I experimented for several days, each time installing a clean system and setting up the configs again and again

shgubar, 2023-01-02:
> Overall, it is really encouraging that at least initially the two drivers did run together and detected different instances of the devices (was it deterministic though - did "riello1000" always grab [...]

I can confidently say that the first time - there was always a correct connection, the names corresponded to the required indicators. After reboots - then it was already different: several reboots - no driver connected, rebooted again - connected, but the indicators did not match the names of the UPS, rebooted several times - they did. In other words, everything goes chaotically.

shgubar, 2023-01-02:
> Would some progress on #1756 address the problem? At least it is something we _can_ try to avoid the stalemate...

I still don't understand what exactly I should do/check.

jimklimov, 2023-01-02:
> I still don't understand what exactly I should do/check.

At a minimum - confirm that such option makes sense? It will probably still be chaotic - which device is known as "riello1000" or "rps1000" from one boot to another. But at least it is a step forward if you want the two UPS states monitored without caring for their identity - which matches not providing any uniquely identifying data into their configurations.

At a maximum, propose a PR :) Though in fact I hope to get to it some day...

jimklimov, 2023-01-02:
Also it might help to look into higher debug-verbosity messages from the drivers being started and trying to match their devices when you provide valid `device` and/or `serial` matching values. Do they complain about a mismatch when it should have been a hit? Why (e.g. could not read that info, etc.)?

Note you can use `debug_min = 6` or similar in ups.conf (global config or a device section) to facilitate such log collection.

shgubar, 2023-01-02:
Of course, if there was an identification of each device - bus, device - it would really solve the problem. But as I already noted (https://github.com/networkupstools/nut/issues/1744) - the driver does not support these parameters.

jimklimov, 2023-01-02:
Looking at code I am fairly sure it does. But lacking a device I can't run and step through it to prove that - or to catch it by hand where it calls something different than what I expect from reading the C words :\

shgubar, 2023-01-03:
> Also wondering how this worked before first reboot:
>
> ```
> 	systemctl enable nut-driver@riello1000.service
> 	systemctl enable nut-driver@rps1000.service
> ```
>
> Such units would be generated (and enabled [...]

Checked. Fresh install (not reboot). When enable only nut-driver.target & nut.target then start only nut.target - run all services without:
nut-driver@riello1000.service
nut-driver@rps1000.service

shgubar, 2023-01-03:
> Also it might help to look into higher debug-verbosity messages from the drivers being started and trying to match their devices when you provide valid `device` and/or `serial` matching values. Do [...]

### **ups.conf**

```
[riello1000]
        driver = riello_usb
        port = auto
        desc = "Riello SEP 1000 ER"
        debug_min = 6
        device = 003
        pollinterval = 15

[rps1000]
        driver = riello_usb
        port = auto
        desc = "Riello SEP 1000 ER NP"
        debug_min = 6
        device = 004
        pollinterval = 15
```

### **syslog**

```
Jan  3 21:22:23 rasp4 nut-driver@riello1000[6530]: Network UPS Tools - Riello USB driver 0.07 (2.8.0.1)
Jan  3 21:22:23 rasp4 nut-driver@riello1000[6530]: Warning: This is an experimental driver.
Jan  3 21:22:23 rasp4 nut-driver@riello1000[6530]: Some features may not function correctly.
Jan  3 21:22:23 rasp4 nut-driver@riello1000[6530]: Fatal error: 'device' is not a valid variable name for this driver.
Jan  3 21:22:23 rasp4 nut-driver@riello1000[6530]: Look in the man page or call this driver with -h for a list of
Jan  3 21:22:23 rasp4 nut-driver@riello1000[6530]: valid variable names and flags.
Jan  3 21:22:23 rasp4 nut-driver@rps1000[6531]: Network UPS Tools - Riello USB driver 0.07 (2.8.0.1)
Jan  3 21:22:23 rasp4 nut-driver@rps1000[6531]: Warning: This is an experimental driver.
Jan  3 21:22:23 rasp4 nut-driver@rps1000[6531]: Some features may not function correctly.
Jan  3 21:22:23 rasp4 nut-driver@rps1000[6531]: Fatal error: 'device' is not a valid variable name for this driver.
Jan  3 21:22:23 rasp4 nut-driver@rps1000[6531]: Look in the man page or call this driver with -h for a list of
Jan  3 21:22:23 rasp4 nut-driver@rps1000[6531]: valid variable names and flags.
```

shgubar, 2023-01-03:
What was done:

```
	systemctl enable nut.target
	systemctl enable nut-driver.target
	systemctl enable nut-driver@riello1000.service
	systemctl enable nut-driver@rps1000.service
```

Then enter only: `systemctl start nut.target` -> look at the file 'success.log' -> `systemctl stop nut.target` -> `systemctl start nut.target` -> look at the file 'failed.log' 

[success.log](https://github.com/networkupstools/nut/files/10339455/success.log)
[failed.log](https://github.com/networkupstools/nut/files/10339454/failed.log)

jimklimov, 2023-01-03:
Thanks!.. so it bypasses usb-common init template, will check... (Part of why there is a long-standing quest to combine all drivers for similar media into a few, ideally one, to avoid surprises like this).

shgubar, 2023-01-03:
Is there anything else I can do/check something to solve my problem?

shgubar, 2023-01-03:
By the way, I found out that `systemctl enable nut-driver.target` there is no need to `enable` - just leave it to `disable` because `nut.target` starts all services inclusively `nut-driver.target`.

But must be enabled - `nut-driver@riello1000.service` and `nut-driver@rps1000.service` - otherwise the drivers will not start.

In the end:
```
	systemctl enable nut.target
	systemctl enable nut-driver@riello1000.service
	systemctl enable nut-driver@rps1000.service
```

After that only enough - `systemctl start/stop nut.target`

jimklimov, 2023-01-03:
Normally, the `nut-driver-enumerator` would generate the drop-in configurations and enable the instantiated `nut-driver@SOMETHING` units (or remove them if the section is deleted from `ups.conf`). I wonder if NDE is getting enabled on your system though (maybe custom-build instructions need an update)...

jimklimov, 2023-01-03:
> Is there anything else I can do/check something to solve my problem?

It might be quicker if you can also go over riello_usb.c sources (and whatever libusbX.c or usb-common.c the build pulls in) to check where it misses the `addvar` for `device` that `usbhid-ups.c` has (and maybe other USB drivers). I'm planning to, but currently finishing up some other tickets.

jimklimov, 2023-01-04:
Hello, hopefully the fix from https://github.com/jimklimov/nut/tree/issue-1754-riello-device should let custom-built NUT discover the riello (and some other) USB devices with more variables to consider.

jimklimov, 2023-01-04:
Looking for some more clues in "Next, a normal system reboot and this is what the system log shows" part, I think I could have gotten the reason for driver abortions (so one runs at a time) inverted.

I wrote that one starts first, another sees the device as busy and aborts.

In fact it may be that the latest newly started grabs the matched device (intentionally, to try and resume monitoring if an older process froze somehow), and then the other driver which is not frozen but used same libusb device entry says "Got disconnected by another driver: Device or resource busy" and aborts.

I see your `rps1000` instance started first, immediately saying "device busy" just after `riello1000` began starting (and aborts); however it quickly restarts again without a systemd complaint, which puzzles me. Then `riello1000` says "device busy" (and aborts), systemd stops its unit, and revives it 15 sec later. Your log there stops just when it happens - but I would somewhat expect that the two drivers keep changing places every 15 sec or so, as they start and grab the same device. (If they would refuse using the device because it is busy in the first place, while initializing, you would instead see the same driver restarting every 15 sec or so.)

jimklimov, 2023-01-05:
...or maybe not: here I'm trying to start another driver for my device, and it is not hijacking it (did not kill the competitor either, different state paths used for test):

````
   0.002567     [D2] Checking device 5 of 9 (0463/FFFF)
   0.424085     [D2] - VendorID: 0463
   0.424098     [D2] - ProductID: ffff
   0.424099     [D2] - Manufacturer: EATON
   0.424101     [D2] - Product: Ellipse ECO
   0.424105     [D2] - Serial Number: 000000000
   0.424109     [D2] - Bus: 003
   0.424114     [D2] - Device: unknown
   0.424117     [D2] - Device release number: 0100
   0.424119     [D2] Trying to match device
   0.424122     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.424151     [D3] match_function_regex: matching a device...
   0.424223     [D2] Device matches
   0.424226     [D2] Reading first configuration descriptor
   0.424234     [D3] libusb_kernel_driver_active() returned 0
   0.424239     [D2] failed to claim USB device: Resource busy
   0.424241     [D2] Kernel driver already detached
   0.424244     [D2] failed to claim USB device: Resource busy
   0.424246     [D2] Kernel driver already detached
   0.424248     [D2] failed to claim USB device: Resource busy
   0.424250     [D2] Kernel driver already detached
   0.424252     [D2] failed to claim USB device: Resource busy
   0.424254     [D2] Kernel driver already detached
   0.424258     Can't claim USB device [0463:ffff]@0/0: Entity not found
````

PR #1763 (2023-01-04): Fix common USB matching options for riello, richcomm and nutdrv_atcl
Should fix #1754 and #1744 among other issues. Thanks to @shgubar for convincing me that the bug is there, and to trace the code with more attention.

Note: "bcmxcp_usb" driver suffers a similar problem but was not easy to fix in a similar fashion (conflicts with symbols in the built `LIBUSB_IMPL` of the day, and impacts nut-scanner build per different dependency chain), so attempt to fix it along with others was shelved.

A side note from debugging: the regex-matching in usb-common.c is not logged in full detail because arra [... cut]