networkupstools/nut #1769: Project Release Cycle Plan Needed

biergaizi, 2023-01-04:
First, apologize for the non-technical nature of this issue report. But I believe resolving this issue is crucial to downstream users given its importance. It's also worth reposting this issue to the mailing list for discussion.

Last week, I've received yet another user E-mail inquiry on the lack of support of newer UPS2000 devices in my driver. The support was in fact already added by me back in Jul 3, 2022 - now 5 months has passed. I realized that it would be completely impractical if users have to wait for more than one year for a trivial 1-line code change - and the downstream distros would only delay the release further to 1.5 years - But this is exactly the case given the current state of the project.

This raises the question of the plan of future project release cycle.

What can we do in the future so that a user doesn't need to wait for a year for new device supports? Some possible plans:

1. Mainline/Stable Model - The stable branch only carries bugfixes, new Device IDs, and other trivial changes, possibly cherry-picked from the development board. They're released as frequently as necessary, and due to the lack of breaking change, downstream distros can also upgrade the packages as soon as possible, without the long Debian-style unstable to stable transition.

2. Fixed release cycle - No matter how many milestones are reached, use mandatory release schedule of once or twice a year, so support for new devices and fixes won't be postponed indefinitely. 

3. Other plans?

jimklimov, 2023-01-04:
Hello,

  Good question - I hoped to deal them out twice a year or so, but it
(toxically, in hindsight) stumbled on trying to adddress a few bugs
reported in 2.8.0 to deliver fixes as 2.8.1 :\

  I've dealt with projects that port stuff to a release/stable branch, and
at least for a small team like ours it is a much more difficult
endeavour/overhead than trying to keep "master" clean enough that after
almost any merged PR an end-user delivery can be made (so the complexity of
variable-quality code lives on feature branches that are merged when
perfect^H^H^H^H^H^H^H good enough).

  And for people who do actually follow the suggestion to try building NUT
from github because their distro is a year behind our latest release (and
ship one 6-7 years old instead) and lacks stuff fixed after 2.7.4 or even
2.8.0, it usually "just works" so the model is viable. Sometimes new corner
cases are uncovered and new bugs to fix are recognized.

Jim

awused, 2023-05-02:
At this point it's been over a full year from the last release. Some projects just go off of git head and distros tend pull those in periodically (albeit usually very irregularly) but because NUT has properly tagged releases those distros are all waiting. If the current recommendation for users is to just build from git head, shouldn't tagging any random commit as a new release be enough? If that's not also good enough for distros it probably shouldn't be recommended to users either.

jimklimov, 2023-05-03:
Cheers, currently away on a trip so can't look into this. In fact, hoped to
check lingering issues/PRs labeled for 2.8.1 and release that before the
trip, but ultimately did not have time for that.

One issue in particular is about guesstimations (#1279 IIRC) which
misbehaved differently in codebases of 2.7.4, 2.8.0 and current master /
attempted fixes (PR open). Someone intently looking at that to bring a
verdict or solution would help a lot for peace of mind to cut the release :)

Hope this helps,
Jim Klimov

jimklimov, 2023-05-04:
Those are good points, I'll think about that. Currently some gating events are to combine certain different PRs (or solved issues) which together bring a useful feature. In recent code, that could be a new mechanism to send commands to a running driver (reload etc) and making use of that for run-time service tooling. Those are interdependent but different PRs and features, which still only make real sense to appear in a same release.

Somewhat related is making sure NEWS and/or UPDATING documents reflect new improvements (and in a release, do not include scaffolding paragraphs they usually start with), and generally that docs, man pages etc. are updated (if they were not as part of PR... because not everyone can deliver everything and a kitchen sink single-handedly).

Another gating ritual is updating the website (build a "historic" one, bump what is current), which is more automated than it was in the years before, but did not happen often enough in the same manner to discover and automate the rest. Making it roll with master branch updates (and adding the concept of historic sites for as-of-release-X docs) was quite an achievement already :) 

Publishing (and signing... gotta make a new key known) tarballs is yet another pain point, more so for automation.

So thanks for the inputs, hopefully we're getting there somewhere... but please do keep nudging -- there are many rabbits to chase in this field but too few dogs :}