networkupstools/nut #1776: APC CS500 incomplete support

kolbai, 2023-01-06:
Compared to what is [supposed](https://networkupstools.org/ddl/APC/Back-UPS_CS_500.html) to be supported, my CS500 is missing a lot of functionality. Missing variables:
```
battery.temperature
input.sensitivity
input.voltage
input.voltage.nominal
output.voltage
output.voltage.nominal
ups.beeper.status
ups.delay.shutdown
ups.delay.start
ups.load
```

And it only supports two commands:
```
test.panel.start
test.panel.stop
```

What could be the reason? I am particularly interested in `input.voltage` and `ups.load`.

jimklimov, 2023-01-06:
Is your device (or firmware) also circa 2010, or newer? One educated guess is that this APC unit may be from newer generations where they made USB HID support less useful compared to proprietary protocols that NUT does not handle yet (see #139).

Alternately it may be reporting those values, but in different fields than what the existing mapping uses. You can try to follow https://github.com/networkupstools/nut/blob/master/docs/hid-subdrivers.txt#L206 instructions to generate a mapping valid as of your device and compare to existing sources to see if some new lines appeared (then merge them to your `apc-hid.c`, build NUT and test if support appeared, and PR the change please)?

kolbai, 2023-01-06:
My device is manufactured in 2004. Do you think it is the reason why so little functionality is supported?

```
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000	debug level is '4'
   0.000419	upsdrv_initups...
   1.331451	Checking device (1D6B/0003) (004/001)
   1.331644	- VendorID: 1d6b
   1.331668	- ProductID: 0003
   1.331671	- Manufacturer: Linux 5.8.0-41-generic xhci-hcd
   1.331673	- Product: xHCI Host Controller
   1.331676	- Serial Number: 0000:03:00.0
   1.331678	- Bus: 004
   1.331680	- Device release number: 0508
   1.331683	Trying to match device
   1.331689	Device does not match - skipping
   1.331827	Checking device (051D/0002) (003/008)
   1.341507	- VendorID: 051d
   1.341545	- ProductID: 0002
   1.341548	- Manufacturer: American Power Conversion
   1.341551	- Product: Back-UPS 500 FW: 6.5.I USB FW: c1 
   1.341554	- Serial Number: AB0427******
   1.341557	- Bus: 003
   1.341559	- Device release number: 0100
   1.341562	Trying to match device
   1.341589	Device matches
   1.341607	failed to claim USB device: could not claim interface 0: Device or resource busy
   1.342665	detached kernel driver from USB device...
   1.343674	nut_usb_set_altinterface: skipped usb_set_altinterface(udev, 0)
   1.344925	HID descriptor, method 1: (9 bytes) => 09 21 00 01 00 01 22 76 01
   1.344955	HID descriptor length (method 1) 374
   1.344959	i=0, extra[i]=09, extra[i+1]=21
   1.344963	HID descriptor, method 2: (9 bytes) => 09 21 00 01 00 01 22 76 01
   1.344966	HID descriptor length (method 2) 374
   1.344969	HID descriptor length 374
   1.357456	Report Descriptor size = 374
   1.357508	Report Descriptor: (374 bytes) => 05 84 09 04 a1 01 09 24 a1 00 85 01 09 fe
   1.357514	 79 01 75 08 95 01 15 00 26 ff 00 b1 02 85 02 09 ff 79 02 b1 02 85 03 05 85
   1.357519	 09 89 79 04 b1 02 85 04 09 8f 79 03 b1 02 85 05 09 8b 65 00 55 00 b1 02 85
   1.357525	 06 09 44 81 82 09 44 b1 82 09 45 81 82 09 45 b1 82 06 86 ff 09 60 81 82 09
   1.357530	 60 b1 82 85 07 05 85 09 85 75 10 95 01 15 00 27 ff ff 00 00 b1 02 85 08 05
   1.357536	 84 09 40 67 21 d1 f0 00 55 05 15 00 27 fe ff 00 00 b1 02 85 09 09 30 b1 02
   1.357543	 85 0a 09 fd 79 03 75 08 95 01 15 00 26 ff 00 b1 02 85 0b 05 85 09 2c b1 02
   1.357549	 85 0c 09 66 75 18 95 01 67 01 10 10 00 55 00 15 00 27 fe ff ff 00 81 82 09
   1.357554	 66 b1 82 09 68 75 10 95 01 66 01 10 55 00 15 00 27 fe ff 00 00 81 82 09 68
   1.357560	 b1 82 85 0d 09 83 75 18 95 01 67 01 10 10 00 55 00 15 00 27 fe ff ff 00 b1
   1.357588	 02 85 0e 09 67 b1 02 85 0f 09 8c b1 02 85 10 09 8e b1 02 85 11 09 29 b1 82
   1.357594	 85 12 09 8d b1 02 85 13 09 d0 75 08 95 01 65 00 55 00 15 00 25 01 81 82 09
   1.357600	 d0 b1 82 85 14 09 42 81 82 09 42 b1 82 05 84 09 69 81 82 09 69 b1 82 06 86
   1.357605	 ff 85 35 09 72 b1 82 85 1c 09 16 75 18 95 01 15 00 27 ff ff ff 00 b1 82 c0
   1.357611	 05 84 09 12 a1 00 c0 06 86 ff 09 03 a1 00 c0 09 05 a1 00 85 40 09 7c 75 08
   1.357615	 95 01 15 00 25 01 b1 82 c0 c0
   1.357759	Using subdriver: EXPLORE HID 0.1
   1.357790	35 HID objects found
   1.357793	Entering libusb_get_report
   1.358511	Report[get]: (2 bytes) => 01 01
   1.358530	Path: UPS.PowerSummary.iProduct, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   1.358534	Entering libusb_get_report
   1.359072	Report[get]: (2 bytes) => 02 02
   1.359098	Path: UPS.PowerSummary.iSerialNumber, Type: Feature, ReportID: 0x02, Offset: 0, Size: 8, Value: 2
   1.359110	Entering libusb_get_report
   1.359959	Report[get]: (2 bytes) => 03 04
   1.359991	Path: UPS.PowerSummary.iDeviceChemistry, Type: Feature, ReportID: 0x03, Offset: 0, Size: 8, Value: 4
   1.359994	Entering libusb_get_report
   1.361004	Report[get]: (2 bytes) => 04 03
   1.361036	Path: UPS.PowerSummary.iOEMInformation, Type: Feature, ReportID: 0x04, Offset: 0, Size: 8, Value: 3
   1.361040	Entering libusb_get_report
   1.361953	Report[get]: (2 bytes) => 05 01
   1.361984	Path: UPS.PowerSummary.Rechargeable, Type: Feature, ReportID: 0x05, Offset: 0, Size: 8, Value: 1
[... 80 lines cut ...]
   1.973813	Detected a UPS: American Power Conversion/Back-UPS 500 FW: 6.5.I USB FW: c1 
   1.973829	find_nut_info: unknown info type: load.off.delay
   1.973853	find_nut_info: unknown info type: load.on.delay
   1.973867	find_nut_info: unknown info type: load.off.delay
   1.973873	upsdrv_initinfo...
   1.973889	upsdrv_updateinfo...
   2.225874	libusb_get_interrupt: Connection timed out
   2.225916	Got 0 HID objects...
   2.225921	Quick update...
   2.225984	dstate_init: sock /run/nut/usbhid-ups-myups open on fd 5
   2.226047	upsdrv_updateinfo...
   2.478188	libusb_get_interrupt: Connection timed out
   2.478232	Got 0 HID objects...
   2.478237	Quick update...
   4.228456	upsdrv_updateinfo...
   4.479238	libusb_get_interrupt: Connection timed out
   4.479271	Got 0 HID objects...
   4.479276	Quick update...
   6.230755	upsdrv_updateinfo...
   6.230813	libusb_get_interrupt: error submitting URB: Device or resource busy
   6.230817	Got disconnected by another driver: Device or resource busy
   8.233347	upsdrv_updateinfo...
   8.233389	Got to reconnect!

```

jimklimov, 2023-01-06:
Well, a device TOO old might also be a problem. It does not seem to include any of the `UPS.Input.*Voltage` treated as `input.voltage` for example.

Still, one more thing to try is to get NUT 2.8.0 or a custom build of github master branch to see if it works better.
There were some fixes for invalid USB data handling by certain vendors (see https://github.com/networkupstools/nut/labels/USB-HID%20encoding%2FLogMin%2FLogMax tagged issues) which involved APC among others.

kolbai, 2023-01-08:
I have tried building NUT 2.8.0 myself, but ran into strange issues:
```
0.491486	[D2] Trying to match device
0.491489	[D2] match_function_subdriver (non-SHUT mode): matching a device...
0.491494	[D3] match_function_regex: matching a device...
0.491503	[D2] Device matches
0.491505	[D2] Reading first configuration descriptor
0.491512	[D3] libusb_kernel_driver_active() returned 0
0.491516	[D2] failed to claim USB device: Resource busy
0.491542	[D2] Kernel driver already detached
0.491547	[D2] failed to claim USB device: Resource busy
0.491561	[D2] Kernel driver already detached
0.491566	[D2] failed to claim USB device: Resource busy
0.491569	[D2] Kernel driver already detached
0.491572	[D2] failed to claim USB device: Resource busy
0.491576	[D2] Kernel driver already detached
0.491591	Can't claim USB device [051d:0002]@0/0: Entity not found
```
So, I cannot test it.

jimklimov, 2023-01-08:
Did you stop driver services/units delivered by package?

kolbai, 2023-01-10:
Thank you for the tip. I got it working.

I have built NUT with all the latest commits this time and tested it. I saw no fundamental changes. I have used 3rd party program to inspect HID objects ant it found the same 35 objects NUT detects. So, input.voltage, output.voltage etc. is not reported. Does not look like there is much you can do about it.

It would be really helpful if Supported Device page of [APC CS 500](https://networkupstools.org/ddl/APC/Back-UPS_CS_500.html) would be updated with a note that older devices of the same kind have very limited support.

I have contacted manufacturer about this matter. I will let you know if I get any useful information. I am closing this issue for now.

For anyone interested.

_upsc myups_ outputs:
```
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.mfr.date: 2004/07/03
battery.runtime: 2580
battery.type: PbAc
battery.voltage: 13.6
battery.voltage.nominal: 12.0
device.mfr: American Power Conversion
device.model: Back-UPS 500
device.serial: AB0427******
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: auto
driver.parameter.synchronous: auto
driver.state: updateinfo
driver.version: 2.8.0-Windows-355-gbce06545a
driver.version.data: APC HID 0.100
driver.version.internal: 0.49
driver.version.usb: libusb-1.0.23 (API: 0x1000107)
ups.firmware:  6.5.I
ups.firmware.aux:  c1 
ups.mfr: American Power Conversion
ups.mfr.date: 2004/07/03
ups.model: Back-UPS 500
ups.productid: 0002
ups.serial: AB0427******
ups.status: OL
ups.timer.reboot: 0
ups.vendorid: 051d
```

_upscmd -l myups_ outputs:
```
shutdown.reboot - Shut down the load briefly while rebooting the UPS
test.panel.start - Start testing the UPS panel
test.panel.stop - Stop a UPS panel test
```

_usbhid-ups -DDDDD -x explore -a myups -x vendorid=051d_  outputs:
```
Network UPS Tools - Generic HID driver 0.49 (2.8.0-Windows-355-gbce06545a)
USB communication driver (libusb 1.0) 0.43
...
[D2] Checking device 4 of 8 (051D/0002)
[D2] - VendorID: 051d
[D2] - ProductID: 0002
[D2] - Manufacturer: American Power Conversion
[D2] - Product: Back-UPS 500 FW: 6.5.I USB FW: c1 
[D2] - Serial Number: AB0427******
[D2] - Bus: 003
[D2] - Device: 002
[D2] - Device release number: 0100
[D2] Trying to match device
[D2] match_function_subdriver (non-SHUT mode): matching a device...
[D3] match_function_regex: matching a device...
[... 325 lines cut ...]
[D3] Report[get]: (2 bytes) => 40 00
[D5] PhyMax = 0, PhyMin = 0, LogMax = 1, LogMin = 0
[D5] Unit = 00000000, UnitExp = 0
[D5] Exponent = 0
[D5] hid_lookup_path: 00840004 -> UPS
[D5] hid_lookup_path: ff860005 -> not found in lookup table
[D5] hid_lookup_path: ff86007c -> not found in lookup table
[D1] Path: UPS.ff860005.ff86007c, Type: Feature, ReportID: 0x40, Offset: 0, Size: 8, Value: 0
[D5] send_to_all: SETINFO ups.mfr "American Power Conversion"
[D5] send_to_all: SETINFO ups.model "Back-UPS 500 FW: 6.5.I USB FW: c1 "
[D5] send_to_all: SETINFO ups.serial "AB0427******"
[D5] send_to_all: SETINFO ups.vendorid "051d"
[D5] send_to_all: SETINFO ups.productid "0002"
[D2] Report descriptor retrieved (Reportlen = 374)
[D2] Found HID device
[D1] Detected a UPS: American Power Conversion/Back-UPS 500 FW: 6.5.I USB FW: c1 
[D2] find_nut_info: unknown info type: load.off.delay
[D2] find_nut_info: unknown info type: load.on.delay
[D2] find_nut_info: unknown info type: load.off.delay
[D5] send_to_all: SETINFO driver.state "init.quiet"
[D5] send_to_all: SETINFO driver.version "2.8.0-Windows-355-gbce06545a"
[D5] send_to_all: SETINFO driver.version.internal "0.49"
[D5] send_to_all: SETINFO driver.name "usbhid-ups"
[D5] send_to_all: SETINFO driver.state "init.info"
```

jimklimov, 2023-01-10:
Thanks! for now I'll reopen to remember to port data into DDL.

jimklimov, 2023-01-11:
Posted to DDL in https://github.com/networkupstools/nut-ddl/commit/2c84b4310526e04036d342a56d2f7fc6aa63935e

closed 2023-01-11 by jimklimov