networkupstools/nut #1780: usbhid-ups seems to randomly stop without error logs, which leads to stale data, and lost connection

MochiJump, 2023-01-07:
I'm running NUT on Ubuntu 22.04.1 LTS for a Tripp Lite ECO850LCD.

NUT seems to run correctly at first until some point where the driver seems to simply stop without any meaningful logs. If the driver is restarted then everything continues until this happens again. This can happen anywhere from less than thirty minutes to a few hours.

Below are my configs:
nut.conf
```
MODE=standalone
```
ups.conf
```
[myups]
    driver = usbhid-ups
    port = auto
    vendorid = 09ae
    productid = 3024
#    pollfreq = 30
#    pollinterval = 30
    ignorelb
    override.battery.charge.warning = 60
    override.battery.charge.low = 55
```
upsd.conf
```
#  MAXAGE 90
```
upsmon.conf
```
#DEADTIME 90
MONITOR myups@localhost 1 monuser mypass master
```
I've tried running with a higher pollinterval/pollfrequency/maxage/deadtime (shown as commented out in the above config) there does not appear to be a significant difference.

logs when running the usbhid-ups driver at level 2 debug:
```
10269.201822	Quick update...
10269.203626	Path: UPS.OutletSystem.Outlet.DelayBeforeStartup, Type: Feature, ReportID: 0x61, Offset: 0, Size: 16, Value: 65535
10269.206614	Path: UPS.OutletSystem.Outlet.DelayBeforeShutdown, Type: Feature, ReportID: 0x15, Offset: 0, Size: 16, Value: 65535
10269.209702	Path: UPS.OutletSystem.Outlet.DelayBeforeReboot, Type: Feature, ReportID: 0x17, Offset: 0, Size: 16, Value: 65535
10269.212802	Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x32, Offset: 16, Size: 1, Value: 1
10269.212863	Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x32, Offset: 18, Size: 1, Value: 0
10269.212881	Path: UPS.PowerSummary.PresentStatus.FullyCharged, Type: Feature, ReportID: 0x32, Offset: 19, Size: 1, Value: 1
10269.212899	Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x32, Offset: 20, Size: 1, Value: 0
10269.212915	Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x32, Offset: 21, Size: 1, Value: 0
10269.212930	Path: UPS.PowerSummary.PresentStatus.FullyDischarged, Type: Feature, ReportID: 0x32, Offset: 22, Size: 1, Value: 0
10299.031089	upsdrv_updateinfo...
                                                                               
Broadcast message from nut@mochijump (somewhere) (Sat Jan  7 03:36:04 2023):   
                                                                               
Communications with UPS myups@localhost lost   
```
At level 6 debug:
```
661.493058	Path: UPS.PowerConverter.Output.Frequency, Type: Feature, ReportID: 0x1c, Offset: 0, Size: 16, Value: 6010
663.841536	send_to_one: PONG
691.273979	upsdrv_updateinfo...
                                                                              
Broadcast message from nut@mochijump (somewhere) (Sat Jan  7 03:54:29 2023):   
                                                                              
Communications with UPS myups@localhost lost     
```
A sample of the syslog for a failure:
```
Jan  7 03:42:32 mochijump upsd[1737]: SIGHUP: reloading configuration
Jan  7 03:42:38 mochijump upsmon[1740]: Reloading configuration
Jan  7 03:54:27 mochijump upsd[1737]: Data for UPS [myups] is stale - check driver
Jan  7 03:54:29 mochijump upsmon[1740]: Poll UPS [myups@localhost] failed - Data stale
Jan  7 03:54:29 mochijump upsmon[1740]: Communications with UPS myups@localhost lost
Jan  7 03:54:34 mochijump upsmon[1740]: Poll UPS [myups@localhost] failed - Data stale
Jan  7 03:55:04 mochijump upsmon[1740]: message repeated 6 times: [ Poll UPS [myups@localhost] failed - Data stale]
Jan  7 03:55:09 mochijump upsmon[1740]: Poll UPS [myups@localhost] failed - Data stale
Jan  7 03:55:14 mochijump upsmon[1740]: Poll UPS [myups@localhost] failed - Data stale 
```
finally I've ran the driver with gdb to see if I can uncover anything. If exited after a disconnect I get the following logs:
```
194.785167	upsdrv_updateinfo...
                                                                              
Broadcast message from nut@mochijump (somewhere) (Sat Jan  7 04:48:48 2023):   
                                                                              
Communications with UPS myups@localhost lost                                                                                                                                          
                                                                              
^C
Program received signal SIGINT, Interrupt.
__GI___ioctl (fd=4, request=1074287884) at ../sysdeps/unix/sysv/linux/ioctl.c:36
36	../sysdeps/unix/sysv/linux/ioctl.c: No such file or directory.
```
This is consistent across two cases where I've interupted with cntl+c after a disconnect. 
If I do this prior to a disconnect I get a different result:
```
0x00007ffff7e9b74d in __GI___select (nfds=7, readfds=0x7fffffffe270, writefds=0x0, exceptfds=0x0, timeout=0x7fffffffe250) at ../sysdeps/unix/sysv/linux/select.c:69
69	../sysdeps/unix/sysv/linux/select.c: No such file or directory.
```

I'm not sure how to continue to troubleshoot this. I'm considering some kind of job that restarts the driver if it falls over, but that is a less than ideal solution. I'd much prefer to address the issue with the driver seemingly just stopping. Thank you in advance for your assistance

jimklimov, 2023-01-07:
Hi, by log format it seems you are running NUT 2.7.4 or older (probably what the distro packaging offers). Is it an option for you to custom-build and run 2.8.0 or recent master branch from Github?

On one hand, some error-handling, logging etc. changed over the 7 years, including libusb1 support which can fare better. On another, arguably better systemd integration is proposed out of the box (and sd_notify/watchdog support PRed recently, testing).

MochiJump, 2023-01-07:
@jimklimov thanks! I had no idea the distro package had such an old version of NUT. I'm curious about the history here, 2.8.0 has years of development history and ~3x the git commits as 2.7.4, however, it looks like 2.7.4 is the 2nd most recent tag. 

Either way, I will try to follow up with 2.8.0, unfortunately, it might take me a while. Out of respect for your backlog I'm going to assume using the latest will fix my issue and close this ticket for now.

closed 2023-01-07 by MochiJump