networkupstools/nut #1782: upsmon -K is too verbose

gdt, 2023-01-07:
Running upsmon -K results in
```
Network UPS Tools upsmon 2.8.0
kill: No such process
UPS: foo@localhost (primary) (power value 1)
Using power down flag file /etc/killpower
Power down flag is not set
```
which is too verbose in system scripts.   The man page does not say there is output.   I think all of that output, except perhaps a "Power down flag IS set." should be demoted to -D (since we have no --verbose).

gdt, 2023-01-07:
Related to #794 in that not finding a pid file and not being able to signal the previous instance is not an error.

jimklimov, 2023-01-07:
Can you please check if current master (after #1762 fix for #1680 among others) still suffers any variant of this problem? Think I'm off the dev env for the weekend.

gdt, 2023-01-09:
Just tested with git master with another upsmon running:

```
Network UPS Tools upsmon 2.8.0.1
Note: A previous upsmon instance is already running!
Usually it should not be running during OS shutdown,
which is when checking POWERDOWNFLAG makes most sense.
UPS: foo@localhost (primary) (power value 1)
Using power down flag file /etc/killpower
Power down flag is not set
```

and  every single line is too verbose, basically debug output.  (Saying the flag is set is ok when it is.)

When upsmon is not running:
```
Network UPS Tools upsmon 2.8.0.1
kill: No such process
UPS: foo@localhost (primary) (power value 1)
Using power down flag file /etc/killpower
Power down flag is not set
```

jimklimov, 2023-01-09:
I wonder if it is all/mostly stderr - so separable from the "is set" message?..

Also, verdict for scripts is exit code; `>/dev/null 2>&1` is always an option. The rest IIRC grew to help postmortems via logs or console screenshots. "Why my shutdown went how it did?"

jimklimov, 2023-01-09:
Digging a bit more into this on my side:

* What the console session shows for the plain command (note there is a non-zero `debug_min` in `upsmon.conf` so even more text below):
````
:; upsmon -K
Network UPS Tools upsmon 2.8.0-Windows-313-gc7ba2a673
Note: A previous upsmon instance is already running!
Usually it should not be running during OS shutdown,
which is when checking POWERDOWNFLAG makes most sense.
UPS: eco650 (primary) (power value 1)
Using power down flag file /etc/killpower
   0.000000     [D1] debug level is '2'
   0.000013     Power down flag is not set
````

* What of this is STDOUT?
````
:; upsmon -K 2>/dev/null
Network UPS Tools upsmon 2.8.0-Windows-313-gc7ba2a673
Note: A previous upsmon instance is already running!
Usually it should not be running during OS shutdown,
which is when checking POWERDOWNFLAG makes most sense.
````

* What of this is STDERR?
````
:; upsmon -K >/dev/null
UPS: eco650 (primary) (power value 1)
Using power down flag file /etc/killpower
   0.000000     [D1] debug level is '2'
   0.000009     Power down flag is not set
````

So in fact "Power down flag is (not) set" is not even a message on `stdout` that is easy for scripts to grab like `TEXT="$(upsmon -K)"` (not that they should do so or rely on its wording).

Regarding usefullness to postmortems however, I think the texts are there validly and intentionally (maybe except `kill` that now unconditionally reports inability to signal others - practically that means another upsmon is not running at the PID checked):
* Tool/version banner
* Which UPSes will be impacted by poweroff (if it gets requested)
* How we check for the flag (e.g. troubleshoot if the location is mounted at that moment?)
* Verdict

It could be distributed across stderr/strout differently though.

jimklimov, 2023-01-09:
Cross-checked for regression vs. 2.7.4 tag:

* With upsmon daemon running (and it suffered #1680 also):
````
:; ./clients/upsmon -K
Network UPS Tools upsmon 2.7.4
Fatal error: A previous upsmon instance is already running!
Either stop the previous instance first, or use the 'reload' command.
````

* With the service stopped (cleanly, so no PID file to `kill`-probe):
````
:;  ./clients/upsmon -K
Network UPS Tools upsmon 2.7.4
fopen /var/run/upsmon.pid: No such file or directory
UPS: eco650 (master) (power value 1)
Using power down flag file /etc/killpower
Power down flag is not set
````

So at least the current version did not "break expectations" from an older release.

jimklimov, 2023-01-09:
I think for the least surprise, the way to go is to add a "quiet" flag or envvar, which would skip all or most of this messaging, and distributions may be free to use that in init-scripts etc.

Actually, banner is printed before CLI processing loop - so envvar may be better (also for backwards-compatibility where older builds would reject the unknown CLI option).

gdt, 2023-01-09:
I think the presence of this output, absent a -D flag, is simply a bug to be fixed. 

  * version is compiled in and known already (thus not useful)
  * Certainly fopen of a pid file that is not expected to be there is not an error. 
  * Which UPS is already set in config and thus known
  * killpower file location is configured already
  * A warning that another upsmon is running is improper as there is not documentation that says upsmon -K requires this.
  * "power down flag is not set" is the expected case

It seems fine to print to stdout "Power down flag IS SET!" in that case.  That's actually noteworthy.

These comments more or less apply to other programs.   Normal daemons start up silently with no output (and no need to redirect) when there are no error conditions.

My advice is rip off the bandaid, and don't worry about departing from previous incorrect behavior - just fix bugs and don't worry about it.

As for version banner, that should just be omitted entirely unless -D is given.  The version running is the version that is installed, and people with 3 versions installed can be careful (or better yet remove 2, unless they are you doing testing!).

gdt, 2023-01-09:
I didn't follow "which UPSes will be impacted".  upsdrvctl shutdown is documented to shut them all off when called regardless of killpower flag, and the flag is global.  upsmon -K also doesn't actually do that; it just reports if the flag is set or not, via exit status.  The state of the UPSes and how we got there is already in syslog, for any reasonable setup.

jimklimov, 2023-01-09:
Say users left the killpower file in `/var/state/ups`. And then the shutdown routine unmounts `/var`. Flag file no longer "exists" so power is not cut. Some head-scratching with few clues to go on follows...

jimklimov, 2023-01-09:
Also with any reasonably large NUT setup, there are one or more machines running the drivers for directly-connected devices and the data server (and upsmon in "primary/master" mode for those devices), not necessarily powered by those UPSes themselves, and some more upsmon "secondary/slave" clients watching the UPSes which power them (but with no direct connection to control them). So when everybody suddenly scuttles the rack (and maybe somebody gets powered down who should not be, or is not shut-down cleanly), this info helps.

If nothing else, it helps troubleshooting via mailing list or github issues, to glean bits of info users never provide and it takes literally weeks to get that. "I have something called NUT and it misbehaves, here's a few random copied lines from what it said" - what version? what OS?

jimklimov, 2023-01-09:
Well... the further this particular discussion goes, the less I am inclined to radically fix anything about this particular complaint. My sysadmin hat won't let me quiesce the important info, at least not by default :\

If you on behalf of your users think it is not useful, feel free to package the scripts that would redirect it to dev-null :)

gdt, 2023-01-09:
upsmon -K at shutdown not printing it found the file is equivalent to printing it didn't find it.   People need to test and add -D if they need it.

The basic issue is that a system has a ton of daemons and facilities and when deep into each it seems really important.  But if they all shout at the user all the time, it's hard to find the important things with all the noise.  So I suggest nut is not as special as you are saying.

I can see your point about helping people who don't know how to provide good  bug reports, but that seems best addressed with a 'required info for bug reports' wiki page.  The current situation burdens everyone, all the time, with non-useful output that makes useful output from other harder to read.

jimklimov, 2023-01-09:
Of all those other subsystems shutting down and yelling about it, NUT is the only one that can leave the box (and its neighbors) powerless to do anything else (pun intended), so to me that seems fairly special.

jimklimov, 2023-01-09:
I'll delegate the subject to vox populi, I guess :)

gdt, 2023-01-09:
Closed in favor of #1789.