networkupstools/nut #1789: daemons lack a -q (quiet flag)

gdt, 2023-01-09:
Preferences differ, but now there is no approach for people that think daemons should print (stdout/stderr) only things that are noteworthy (critical or an error) or specifically requested.   This is a feature request to add -q to drivers/upsd/upsmon/upslog so that it is quiet absent critical/error/requested.

For a longer discussion, see #1782.

jimklimov, 2023-01-09:
See also #1667 for similar solution precedent to a similar issue.

gdt, 2023-01-09:
I vote for using -v for verbose and making -q default, which avoids the quiet/debug fighting.

As for #1667, I hope ssl was only initialized if the user asked to do something ssl-ish.

gdt, 2023-01-09:
Also, about banner: It is very rare for daemons to print version to stdout/stderr.   The standard approach is to daemoize and log a "foo started" at LOG_INFO.   That might as well have the version.   Certainly if there's a problem that prevents daemonizing, that's fair on stderr

Another common pattern is to have --version or -V to print the version (and do nothing else).  With that, there's no need to print the version if the arguments are invalid, so the whole "when to print the banner" becomes moot.

jimklimov, 2024-07-31:
In PR #2568 among other things I'm trying a balanced envvar-based approach (so easy to stick into wrapper scripts, shell profile, `nut.conf`, etc.) with a new support for `NUT_QUIET_INIT_BANNER=true`. As part of this work, some programs were identified that unilaterally printed the banner as soon as they started, before processing CLI options and possibly failing at that:
* `upsdrvctl` - https://github.com/networkupstools/nut/pull/2568/commits/65250a9ab22ed4439d0d9020a576cf1e75a2da30
* `upslog`, `upsmon`, `upsd` and drivers (`main.c`) - https://github.com/networkupstools/nut/pull/2568/commits/06be3c806dfc45db3f13dba149a0b0920da1766d

Several other programs only print it with the `-V` option, in help/usage, or debug streams.

Troubleshooting-wise, I rule in favor of having it available by default since it takes days of turnaround in issue discussions to pull such details as "which NUT version are you complaining about?" with many users. Some have little way to know that without a lot of spelunking, using embedded builds, HA plugin containers, etc.

But deployments run by power-users who are annoyed by these banners can now be made quieter with a single line in their NUT configs or shell profiles.

@gdt: WDYT, can the issue be deemed resolved by this approach? :)

gdt, 2024-07-31:
You are the BDFL, so it's your call. 

But I think programs should not print these by default, and that's what -V or --version is for (and fine for it to be in the on-error usage output).  The fact that users submit terrible bug reports is not a good reason to burden others with noise or ask them to do something which I view as kludgy.

closed 2024-07-31 by jimklimov