networkupstools/nut #1796: Missing APC HID fields

kolbai, 2023-01-11:
NUT does not recognize some HID fields as seen in #1776. apcupsd have them defined, see [1](https://fossies.org/linux/apcupsd/examples/hid-ups.c) and [2](https://fossies.org/linux/apcupsd/examples/rpt/Back-UPS-500-USB.rpt).

I am not sure if they are correct, but likes like it. Can anyone double check?

```
0xFF860005 = APCGeneralCollection
0xFF86007C = APCForceShutdown
0xFF860016 = APCBattReplacementDate
0xFF860060 = APCStatusFlag
0xFF860072 = APCPanelTest
```

kolbai, 2023-01-11:
Sorry, I was looking at 2.7.4 output, 2.8.0 has it already fixed. The only difference is that apcupsd recognizes 0xFF86007C as `APCForceShutdown` and NUT recognizes it as `APCDelayBeforeReboot`.

jimklimov, 2023-01-11:
I believe `explore` mode is a special fake HID subdriver that does not reject anything to report all it sees. But being a different codepath than `apc-hid.c` it does not know about those custom name mappings.

closed 2023-06-27 by jimklimov