networkupstools/nut #1805: shutdown.reboot does not work for CS500

kolbai, 2023-01-12:
`shutdown.reboot` command does not work for my CS500. In my case the reason is that it sends two byte report 0x40 0x0A, but the 2nd byte should be 0x01. Most likely it has to be fixed here: https://github.com/networkupstools/nut/blob/5b4684c784fc9ec98f6439820221e7934d1bc68f/drivers/apc-hid.c#L455-L456

jimklimov, 2023-01-12:
Just in case: does `upscmd shutdown.return` help here?

Also found what seems a very relevant discussion: https://alioth-lists.debian.net/pipermail/nut-upsuser/2010-November/006303.html (or all the thread represented in https://nut-upsuser.alioth.debian.narkive.com/Fj636tRZ/support-of-shutdown-return-on-a-apc-back-ups-cs-500)

kolbai, 2023-01-12:
I cannot run `upscmd shutdown.return` as it is not supported. 

upscmd -l myups:
```
Instant commands supported on UPS [myups]:

shutdown.reboot - Shut down the load briefly while rebooting the UPS
test.panel.start - Start testing the UPS panel
test.panel.stop - Stop a UPS panel test
```

upscmd myups shutdown.return:
`Unexpected response from upsd: ERR CMD-NOT-SUPPORTED`

I have read those discussions, that's where I found a suggestion to send 0x01 and not 0x0A.
>I haven't been able to get the CS 500 into sleep mode at all, apart from
>by using the old hidups driver that I modified to setvalue(0x86007c, 1)

As explained in the docs, `shutdown.return` command should instruct UPS to turn off and wait until the power is back. In my case that is not the behavior I see, my UPS reboots and briefly shuts down load when `shutdown.reboot` is issued.
https://github.com/networkupstools/nut/blob/8b10928c7fe7898adbcc56b1ca3f34d88d77264c/docs/nut-names.txt#L741-L746