networkupstools/nut #1817: bestfortress shutdown does not respect configured delay - initups not called?

gdt, 2023-01-17:
nut from git, Best Fortress LI660 ("old Fortress"), driver bestfortress

Doing "upsdrvctl shutdown" results in a near-immediate power off, instead of the 10 s one would expect from reading the code.  My copy does (20 instead of 10 in git).
```
        dstate_setinfo("ups.delay.shutdown", "20");
```
in `ups_initinfo()`.   `upsc foo` includes `ups.delay.shutdown: 20`, so this does get set.

Then in `upsdrv_shutdown`:
```
        grace = dstate_getinfo("ups.delay.shutdown");

        if (!grace) {
                upsdebugx(1, "upsdrv_shutdown: ups.delay.shutdown is NULL!");
                grace = "1"; /* apparently, OFF0 does not work */
        }
```
and the debug log said it was sending (in angle brackets)
```
<OFF1
>
```
So the bug is either
  * bestfortress is confused and setting that variable doesn't belong in `ups_initinfo`
  * `upsdrvctl shutdown` doesn't lead to calling `ups_initinfo` but it should

Regardless, I'm going to prepare a PR which is far more defensive, but the larger question is interesting and should probably be addressed.

We can close this when shutdown is sane and open a fresh issue once we can say what's wrong.

jimklimov, 2023-01-17:
There is a spaghetti of similar names, so better double-check the ones posted for reference like this (I know I fail at that sometimes) :)

I suppose you meant `upsdrv_initinfo()` as in https://github.com/networkupstools/nut/blob/408775bb13b711a2b46b6f6ee315db1e72eb902e/drivers/bestfortress.c#L56 vs. `upsdrv_initups()` as in https://github.com/networkupstools/nut/blob/408775bb13b711a2b46b6f6ee315db1e72eb902e/drivers/bestfortress.c#L484

Roles of these driver-dependent methods are detailed in the docs:
https://github.com/networkupstools/nut/blob/408775bb13b711a2b46b6f6ee315db1e72eb902e/docs/new-drivers.txt#L105-L125

When the standard driver programs start, they read `ups.conf` (so e.g. any configured overrides should be seen) when they get the `-a UPSNAME` option (e.g. passed by `upsdrvctl`) at https://github.com/networkupstools/nut/blob/408775bb13b711a2b46b6f6ee315db1e72eb902e/drivers/main.c#L752-L755 and handle low-level hardware-related setup in `upsdrv_initups()`. But indeed then the `forceshutdown` is handled if needed, and only then the `upsdrv_initinfo()` which may mean interactions with device, delays to do so, bail-out, etc.

https://github.com/networkupstools/nut/blob/408775bb13b711a2b46b6f6ee315db1e72eb902e/drivers/main.c#L1026-L1062

Per `git blame`, this order of events is here for eternity, so at least intentional design and not a recent regression : https://github.com/networkupstools/nut/blame/408775bb13b711a2b46b6f6ee315db1e72eb902e/drivers/main.c#L1026-L1062

With all that in mind, probably some of the settings related to shutdown should be in the early `upsdrv_initups()` (and probably optional - if not yet set by driver `-x` arguments or config parameters), so "bestfortress is confused" - taking from your options :) 

Given this took a while to unravel, some follow-ups are due:
* clarify this in docs for the methods
* revise other drivers for similar oversight (no blames here, it looks like a hidden gem)

CC @aquette @clepple for more insights?

gdt, 2023-01-17:
Thanks.   So it's clear from code that upsdrv_initinfo is not called when doing shutdown.   I think the docs shouldn't really suggest that setting one-time variables is good there, then, and should caution that shutdown happens without initinfo.

I'm not having this issue, but it also seems like shutdown, while not supposed to probe, should probably be defensive about serial port state.  So in the bestfortress case, probably should send '\rOFF 20\r' with a CR to start?

PR #1820 (2023-01-27): Improve Best Fortress driver
This PR does two main things:
  - rationalize and improve logging, and add/update comments
  - improve shutdown process, fixing a serious bug where the intended 10s delay was not honored, resulting in a way-too-short 1s delay.  Bump driver version because of this.  Fixes #1817.

I have tested a functionally equivalent earlier version on a Fortress LI660 including `upsmom -c fsd`, and this version with `-D -D -D -D -D`.

Changes are entirely in `drivers/bestfortress.c`; interfaces are not changed.
 
Things that I think can be done later (not in this PR):
  - More su [... cut]
closed 2023-01-27 by jimklimov