networkupstools/nut #1824: PowerWalker VI 3000 RT HID

SirAdams, 2023-01-18:
I have a UPS at my place which I am trying to run both on USB and RS232, but for RS232 I am missing the original cable. 
As I wish to assist in the development of the software I am sending the logs from the USB. 
The UPS has an additional feature that allows you to control the LS1 Output and LS2 Output using WinPower software. There is also the possibility to set the switch-on delay of each input. 

If I can get the original RS232 cable, I will try to make a diagram of the cable. And I will try to attach logs from RS232.

Below the logs, etc.

usbhid-ups:
`root@debian:/usr/src# /lib/nut/usbhid-ups -DD -a ups
Network UPS Tools - Generic HID driver 0.49 (2.8.0.1)
USB communication driver (libusb 1.0) 0.44
   0.000000     [D1] Network UPS Tools version 2.8.0.1 configured with flags: --build=x86_64-linux-gnu --prefix=/usr --includedir=${prefix}/include --mandir=${prefix}/share/man --infodir=${prefix}/share/info --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir=${prefix}/lib/x86_64-linux-gnu --runstatedir=/run --disable-maintainer-mode --disable-dependency-tracking --prefix= --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=${prefix}/lib/x86_64-linux-gnu --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/x86_64-linux-gnu/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --wi
   0.000016     [D1] debug level is '2'
   0.000710     [D1] Succeeded to become_user(nut): now UID=105 GID=111
   0.000727     [D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.44')
   0.000731     [D1] upsdrv_initups (non-SHUT)...
   0.005288     [D2] Checking device 1 of 9 (045B/0210)
   0.101002     [D2] - VendorID: 045b
   0.101011     [D2] - ProductID: 0210
   0.101014     [D2] - Manufacturer: unknown
   0.101017     [D2] - Product: unknown
   0.101020     [D2] - Serial Number: unknown
   0.101023     [D2] - Bus: 002
   0.101026     [D2] - Device: 006
   0.101029     [D2] - Device release number: 0100
   0.101051     [D2] Trying to match device
   0.101063     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.101083     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.101095     [D2] Device does not match - skipping
   0.127803     [D2] Checking device 2 of 9 (0BDA/0306)
   0.131993     [D2] - VendorID: 0bda
   0.132023     [D2] - ProductID: 0306
   0.132041     [D2] - Manufacturer: Realtek
   0.132058     [D2] - Product: USB3.0 Card Reader
   0.132076     [D2] - Serial Number: 201510191731
   0.132093     [D2] - Bus: 002
   0.132110     [D2] - Device: 003
   0.132130     [D2] - Device release number: 0118
   0.132147     [D2] Trying to match device
   0.132167     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.132194     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.132213     [D2] Device does not match - skipping
   0.132247     [D2] Checking device 3 of 9 (045B/0210)
   0.132296     [D2] - VendorID: 045b
   0.132309     [D2] - ProductID: 0210
   0.132325     [D2] - Manufacturer: unknown
   0.132346     [D2] - Product: unknown
   0.132363     [D2] - Serial Number: unknown
   0.132379     [D2] - Bus: 002
   0.132398     [D2] - Device: 005
   0.132417     [D2] - Device release number: 0100
   0.132435     [D2] Trying to match device
   0.132447     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.132462     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.132473     [D2] Device does not match - skipping
   0.132494     [D2] Checking device 4 of 9 (1D6B/0003)
   0.132526     nut_libusb_open: invalid libusb port number 0, falling back to enumeration order counter 3
   0.132746     [D2] - VendorID: 1d6b
   0.132760     [D2] - ProductID: 0003
   0.132771     [D2] - Manufacturer: Linux 6.0.0-6-amd64 xhci-hcd
   0.132786     [D2] - Product: xHCI Host Controller
   0.132805     [D2] - Serial Number: 0000:00:14.0
   0.132823     [D2] - Bus: 002
   0.132845     [D2] - Device: 003
   0.132861     [D2] - Device release number: 0600
   0.132879     [D2] Trying to match device
   0.132898     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.132913     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.132929     [D2] Device does not match - skipping
[... 233 lines cut ...]
   7.821319     [D1] Quick update...
   7.821585     [D2] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   7.821614     [D2] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
   7.821629     [D2] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 16, Size: 8, Value: 1
   7.821644     [D2] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
   9.072321     [D1] upsdrv_updateinfo...
   9.822578     [D2] nut_libusb_get_interrupt: Connection timed out
   9.822607     [D1] Got 0 HID objects...
   9.822619     [D1] Quick update...
   9.822883     [D2] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
   9.822912     [D2] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
   9.822926     [D2] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 16, Size: 8, Value: 1
   9.822938     [D2] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
  11.073604     [D1] upsdrv_updateinfo...
  11.823864     [D2] nut_libusb_get_interrupt: Connection timed out
  11.823894     [D1] Got 0 HID objects...
  11.823907     [D1] Quick update...
  11.824173     [D2] Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
  11.824205     [D2] Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
  11.824219     [D2] Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 16, Size: 8, Value: 1
  11.824246     [D2] Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
^C
`
[image]
[image]

jimklimov, 2023-01-19:
Thanks for this report. Do I understand correctly that at least for some of the data points you want, existing driver already works and data-dump with `/lib/nut/usbhid-ups -d1 -a ups` would print reasonable keys and values?

So e.g. the `nut_libusb_get_interrupt: Connection timed out` are not a problem in the question here - if they are, you can try the `pollonly` mode? (although just in case, can try that anyway to see if "Can't retrieve Report f0: Resource temporarily unavailable" sort of issues go away and more data gets seen)

Regarding the handling of more features, like detailed monitoring/management of "outlet groups" (generally speaking, even if there's one socket soldered to each "group"), I'd try to find info about the vendor-extension entries that remain hexes in the dump above (some seem like arrays so probably are about these outlets). Ideally get a hold of vendor spec, or google into other projects, or sniff the USB comms from vendor software and experiment...

SirAdams, 2023-01-24:
I refreshed the logs. Compilation from today.

SirAdams, 2023-01-24:
Below I paste the logs from UPS for 4 different states. At the beginning of the log is the description. I don't know if this will be useful to you but I have no idea how else to find the outlet status readings for lines 1 and 2. 
If anyone has any suggestions on how else I can investigate this then please share an idea. 

````
root@debian:~# #All Output ON
root@debian:~# /lib/nut/usbhid-ups -DDDD -a ups
Network UPS Tools - Generic HID driver 0.49 (2.8.0.1)
USB communication driver (libusb 1.0) 0.45
   0.000000     [D3] do_global_args: var='maxretry' val='3'
   0.000057     [D3] main_arg: var='driver' val='usbhid-ups'
   0.000062     [D3] main_arg: var='port' val='auto'
   0.000106     [D1] Network UPS Tools version 2.8.0.1 configured with flags: --build=x86_64-linux-gnu --prefix=/usr --includedir=${prefix}/include --mandir=${prefix}/share/man --infodir=${prefix}/share/info --sysconfdir=/etc --localstatedir=/var --disable-option-checking --disable-silent-rules --libdir=${prefix}/lib/x86_64-linux-gnu --runstatedir=/run --disable-maintainer-mode --disable-dependency-tracking --prefix= --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=${prefix}/lib/x86_64-linux-gnu --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/x86_64-linux-gnu/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --wi
   0.000116     [D1] debug level is '4'
   0.000825     [D1] Succeeded to become_user(nut): now UID=105 GID=111
   0.000845     [D2] Initializing an USB-connected UPS with library libusb-1.0.26 (API: 0x1000109) (NUT subdriver name='USB communication driver (libusb 1.0)' ver='0.45')
   0.000849     [D1] upsdrv_initups (non-SHUT)...
   0.005488     [D2] Checking device 1 of 9 (045B/0210)
   0.100628     [D2] - VendorID: 045b
   0.100646     [D2] - ProductID: 0210
   0.100653     [D2] - Manufacturer: unknown
   0.100659     [D2] - Product: unknown
   0.100665     [D2] - Serial Number: unknown
   0.100671     [D2] - Bus: 002
   0.100695     [D2] - Device: 003
   0.100723     [D2] - Device release number: 0100
   0.100734     [D2] Trying to match device
   0.100747     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.100770     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.100782     [D2] Device does not match - skipping
   0.127345     [D2] Checking device 2 of 9 (0BDA/0306)
   0.131487     [D2] - VendorID: 0bda
   0.131514     [D2] - ProductID: 0306
   0.131524     [D2] - Manufacturer: Realtek
   0.131533     [D2] - Product: USB3.0 Card Reader
   0.131543     [D2] - Serial Number: 201510191731
   0.131552     [D2] - Bus: 002
   0.131570     [D2] - Device: 004
   0.131582     [D2] - Device release number: 0118
   0.131593     [D2] Trying to match device
   0.131604     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.131623     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.131637     [D2] Device does not match - skipping
   0.131664     [D2] Checking device 3 of 9 (045B/0210)
   0.131713     [D2] - VendorID: 045b
   0.131725     [D2] - ProductID: 0210
   0.131738     [D2] - Manufacturer: unknown
   0.131749     [D2] - Product: unknown
   0.131761     [D2] - Serial Number: unknown
   0.131772     [D2] - Bus: 002
   0.131782     [D2] - Device: 002
   0.131793     [D2] - Device release number: 0100
   0.131805     [D2] Trying to match device
   0.131817     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.131830     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.131840     [D2] Device does not match - skipping
   0.131857     [D2] Checking device 4 of 9 (1D6B/0003)
   0.132091     [D2] - VendorID: 1d6b
   0.132106     [D2] - ProductID: 0003
   0.132115     [D2] - Manufacturer: Linux 6.0.0-6-amd64 xhci-hcd
   0.132125     [D2] - Product: xHCI Host Controller
   0.132134     [D2] - Serial Number: 0000:00:14.0
   0.132143     [D2] - Bus: 002
   0.132152     [D2] - Device: 001
   0.132161     [D2] - Device release number: 0600
   0.132170     [D2] Trying to match device
   0.132181     [D2] match_function_subdriver (non-SHUT mode): matching a device...
   0.132198     [D2] match_function_subdriver (non-SHUT mode): failed to match a subdriver to vendor and/or product ID
   0.132209     [D2] Device does not match - skipping
   0.132231     [D2] Checking device 5 of 9 (045B/0209)
   0.247994     [D2] - VendorID: 045b
[... 2388 lines cut ...]
   0.324783     [D4] Entering libusb_get_report
   0.325037     [D3] Report[get]: (2 bytes) => c1 01
   0.325057     [D1] Path: UPS.PowerConverter.Input.[1].ffff0094, Type: Feature, ReportID: 0xc1, Offset: 0, Size: 8, Value: 1
   0.325060     [D4] Entering libusb_get_report
   0.325248     [D3] Report[get]: (5 bytes) => 30 01 01 03 03
   0.325257     [D1] Path: UPS.PowerConverter.Input.[1].FlowID, Type: Feature, ReportID: 0x30, Offset: 0, Size: 8, Value: 1
   0.325261     [D4] Entering libusb_get_report
[... cut ...]
   0.325700     [D3] Report[buf]: (3 bytes) => 2d 00 00
   0.325706     [D1] Path: UPS.PowerConverter.Input.[1].PresentStatus.Buck, Type: Input, ReportID: 0x2d, Offset: 8, Size: 8, Value: 0
   0.325709     [D4] Entering libusb_get_report
   0.325930     [D3] Report[get]: (4 bytes) => 32 00 00 00
   0.325943     [D1] Path: UPS.PowerConverter.Input.[1].PresentStatus.FrequencyOutOfRange, Type: Feature, ReportID: 0x32, Offset: 0, Size: 8, Value: 0
   0.325947     [D3] Report[buf]: (4 bytes) => 32 00 00 00
   0.325952     [D1] Path: UPS.PowerConver

SirAdams, 2023-02-06:
I am adding the rs232 log files. Is there any known and supported protocol by NUT that I would be able to use to communicate with UPS?
[console.txt](https://github.com/networkupstools/nut/files/10664876/console.txt)
[3000.txt](https://github.com/networkupstools/nut/files/10664881/3000.txt)
[Read/Write in HTML powerwalker.zip](https://github.com/networkupstools/nut/files/10664887/powerwalker.zip)

jimklimov, 2023-02-08:
Sorry it took a while to get around to this issue :-{

The latter (RS232 dump) looks a lot like Megatec Qx protocol family, so you can try the `nutdrv_qx` with it and pick a subdriver if auto-guess won't work.

For the USB HID part, I now wonder if the problem is that the vendor mapping table just has no entries for fields you see in the USB debug dump (e.g. those grouped by outlet arrays) to map them onto NUT data point names.

https://github.com/networkupstools/nut/blob/master/drivers/liebert-hid.c is rather small.

Please check with https://github.com/networkupstools/nut/blob/master/docs/hid-subdrivers.txt (and possibly other docs) about details how to eventually call https://github.com/networkupstools/nut/blob/master/scripts/subdriver/gen-usbhid-subdriver.sh and have it generate a mapping table for your device. Then use `meld`, `winmerge` or some similar tool to port the entries from the file it creates into existing `liebert-hid.c` subdriver, try to re-build with it and see if that works (at least should be a decent starting point - not sure e.g. if it would instantly support commands or some fiddling with mapping flags would be needed, etc.).