networkupstools/nut #1827: grep issues

neilromig, 2023-01-20:
Running NUT 2.8.0 on Arch Linux I get warnings relating to egrep usage which come from nut-driver-enumerator.sh:

`nut-driver@myups[478]: egrep: warning: egrep is obsolescent; using grep -E`

I also see this (which I have been unable to track down as yet:

`nut-driver@myups[426]: grep: warning: stray \ before #`

jimklimov, 2023-01-20:
Thanks. The `egrep` suggestions were fixed some time ago (maybe after 2.8.0 release). One legacy-build target complained back, but had an XPG alternative.

Not sure I saw the other one, but there were some syntax-fixes it could be part of.

Do you have a chance to try the variant from master branch?

neilromig, 2023-01-20:
I have compiled from master branch (via Arch AUR repository) and comparing files I have worked out the offending grep is the last line of function upslist_normalizeFile_filter() in nut-driver-enumerator.sh:

`grep -E '([\[\=]|^[^ '"$TABCHAR"']*$|^[^ '"$TABCHAR"']*[ '"$TABCHAR"']*\#.*$)'`

So, I have removed the "stray \" at the end of the expression as it seems surplus to requirements, the hash having no special meaning as far as I can determine from reading the grep man page.

Expression is now:

`grep -E '([\[\=]|^[^ '"$TABCHAR"']*$|^[^ '"$TABCHAR"']*[ '"$TABCHAR"']*#.*$)'`

jimklimov, 2023-01-20:
Thanks! Sort of odd of them to complain about a valid regex (all chars *may* be escaped, for a bit of expensive overkill), but I guess also nice of them to raise noise and cause optimization :)

jimklimov, 2023-01-20:
Care to post a PR in your name? :)

PR #1828 (2023-01-21): Update nut-driver-enumerator.sh.in
Remove backslash from grep expression at line 836 to avoid grep warning of stray backslash.

Closes: #1827
closed 2023-01-21 by jimklimov