networkupstools/nut #1831: make -j2 fails

gdt, 2023-01-21:
NetBSD 9, system make, recent git (actually bestfortress PR).   Building with -j should enable parallel builds, but apparently there is a missing dependency.
```
--- upscode2 ---
libtool: link: gcc -I../../include -I/usr/pkg/include/libusb-1.0 -I/usr/pkg/include/neon -isystem /usr/pkg/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu99 -Wno-system-headers -Wall -Wextra -Wsign-compare -pedantic -Wno-error -Wl,-R/usr/pkg/lib -o upscode2 upscode2.o  -L/usr/pkg/lib ./.libs/libdummy_serial.a ./.libs/libdummy.a ../common/.libs/libcommon.a ../common/.libs/libparseconf.a -lm -lpthread
--- victronups ---
libtool: link: gcc -I../../include -I/usr/pkg/include/libusb-1.0 -I/usr/pkg/include/neon -isystem /usr/pkg/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu99 -Wno-system-headers -Wall -Wextra -Wsign-compare -pedantic -Wno-error -Wl,-R/usr/pkg/lib -o victronups victronups.o  -L/usr/pkg/lib ./.libs/libdummy_serial.a ./.libs/libdummy.a ../common/.libs/libcommon.a ../common/.libs/libparseconf.a -lpthread
Making all in tools
--- all-recursive ---
make[1]: make[1]: don't know how to make ../scripts/hotplug/libhid.usermap. Stop

make[1]: stopped in /home/n0/gdt/SOFTWARE/IOT/nut/BUILD/tools
*** [all-recursive] Error code 1
```

The problem is likely in tools/Makefile.am.

The build seems ok with GNU make, but that's not documented to be necessary (and isn't without -j) and it's  a bug to need GNU make anyway.

I'll see if I can figure this out.

jimklimov, 2023-01-21:
Yes, inability to parallel-make is a bug nowadays. It is regularly done on NUT CI farm with GNU, (Free/Open)BSD and Sun makes. Note many of them require a space between `-j` and number (two CLI tokens).

I'd wager other systems might not decide to build hotplug support so this bit is the path less walked?

jimklimov, 2023-01-21:
To be clearer, it was not always the case, and did take some refactoring (dumbing down) of the makefiles a couple of years ago. So maybe something was missed.

gdt, 2023-01-22:
I was surprised to see this biult, as I thought it was Linux-specific.  I'm just running configure with few args, well
```
    CPPFLAGS=-I/usr/pkg/include \
             LDFLAGS="-L/usr/pkg/lib -Wl,-R/usr/pkg/lib" \
             time ../configure --prefix=/usr/pkg \
             --prefix=/usr/pkg \
             --with-drvpath=/usr/pkg/libexec/nut \
             --with-pidpath=/var/db/nut \
             --with-pidpidpath=/var/db/nut \
             --with-user=nut --with-group=nut \
             --with-doc=man \
             --without-python2 \
             --with-python=${PYTHON}
```
which is really not much.  tools/Makefile.am has a clue:
```
# Anyway, for the time being, we force build in ./ before nut-scanner,                                                                                                          
# to have nutscan-{usb,snmp}.h built before going into the nut-scanner                                                                                                          
# sub-directory. For good measure we also call this from nut-scanner's                                                                                                          
# make, to handle developer workflow (editing the *.c sources this uses).                                                                                                       
```

jimklimov, 2023-01-22:
Note: no "pidpifile", were you after "altpidfile" and/or "statepath"?

gdt, 2023-01-22:
altpidfile....  But I do this builds to make dist and then build the distfile in pkgsrc context.  But thanks for catching that and I fixed my build script.

gdt, 2023-01-22:
The perl script that generates `libhid.usermap` is run via a rule that says:

$(GENERATED_USB_FILES):

but libhid.usermap is in `GENERATED_USB_OS_FILES`.  It looks like the rule needs to fire on those 4 targets also.

gdt, 2023-01-22:
```
diff --git a/tools/Makefile.am b/tools/Makefile.am
index 49af8d915..b4e14a8a8 100644
--- a/tools/Makefile.am
+++ b/tools/Makefile.am
@@ -60,7 +60,7 @@ $(GENERATED_SNMP_FILES): $(top_srcdir)/drivers/*-mib.c
                echo "----------------------------------------------------------------------"; \
        fi
 
-$(GENERATED_USB_FILES): $(top_srcdir)/drivers/*-hid.c $(top_srcdir)/drivers/*usb*.c $(top_srcdir)/drivers/nutdrv_qx.c
+$(GENERATED_USB_OS_FILES) $(GENERATED_USB_FILES): $(top_srcdir)/drivers/*-hid.c $(top_srcdir)/drivers/*usb*.c $(top_srcdir)/drivers/nutdrv_qx.c
        @if perl -e 1; then \
                echo "Regenerating the USB helper files in SRC dir."; \
                TOP_SRCDIR="$(top_srcdir)" ; export TOP_SRCDIR; \
```

With that,  I can build -j6 on a 4-core machine (and -j2).  It feels like not the right fix, but maybe it is -- that perl script makes a lot of things.

gdt, 2023-01-22:
Also, I'd say either there should be make rules to build these and not in autogen/EXTRA_DIST, or there shouldn't be make rules with the autogen/EXTRA_DIST.   Having both is confusing and error prone.

jimklimov, 2023-01-22:
Well, as said - at `configure`/`make` time the build platform may have no perl or python and sources used (from tarball or git - after `autogen.sh` for the latter) should be fine with that. Still, it is valid to want to regenerate those files based on developer changes without configuring the whole project just to accomodate for a new `VID:PID` pair to test against.

From this PoV, probably your PR to add the `make` dependency does make sense (but may cause "useless" regenerations of those files due to just timestamp changes as files are unpacked, so not sure if worth it for 99% of builds).

gdt, 2023-01-22:
Timestamps should be restored on untar, so that should avoid unnecessary regen.

I can see the rules for regen during development.

So I guess you will merge both PRs?

jimklimov, 2023-01-22:
Probably yes if no builders complain :) I'll look for ways to ensure a single run of the script if it decides to rebuild any of these files, so they don't waste CPU time and/or have potential to collide in parallel builds - if I do that early on, it could make your PR moot...

PR #1834 (2023-01-24): Explicitly distribute all files generated by `autogen.sh` into dist tarballs
Could be behind the problems in issue #1831 although still strange (probably toolkits differ in what rules they generate and/or how they are handled).

CC @gdt : can you please test this branch without the recipe change you proposed per #1832? Does this fix alone address the problem in practice?

UPDATE:
Closes: #1832
Closes: #1831
closed 2023-01-24 by jimklimov