networkupstools/nut #1833: Support sleep/hibernate integration (generally and with systemd)

jimklimov, 2023-01-22:
There were several discussions in the past about NUT daemons emitting messages like "Data stale" when a system wakes up after prolonged sleep. One particular use-case is putting a system into hibernation when an UPS goes depleted, or into sleep when on-battery (hoping the lone RAM's power draw can be kept up for hours). This is not a platform-specific issue, everybody might do it, but it may have platform-specific optimal solutions (and/or general ones).

The messages make sense technically (lots of time passed since last poll) but seem confusing to some users and pollute the console cluttered by various subsystems after wake-up.

This issue is about exploring ways to integrate with service management frameworks like systemd to "know" a sleep happened, e.g. deal with dependencies on/against https://www.freedesktop.org/software/systemd/man/systemd-suspend.service.html family of units, or deliver hooks for them (example in https://blog.christophersmart.com/2016/05/11/running-scripts-before-and-after-suspend-with-systemd/ but official doc warns not to - deemed local hacks), or explore its "Inhibitor interface" per https://blog.christophersmart.com/2016/05/11/running-scripts-before-and-after-suspend-with-systemd/ and possibly something is possible by listening to systemd dbus...

With such integration, NUT daemons would know to silently reestablish connections when the system wakes up, and maybe chirp a message about that specifically. One solution could be to use the pre/post-sleep hooks to completely stop and start NUT services and so avoid "staleness" (also possibly account for IP address changes after wake-up).

A general-purpose alternative could be to somehow watch for system clock jumps (e.g. we have a 5 or 30 second pollfreq, and suddenly last loop took tens of minutes or more), and/or possibly track "local" time (as opposed to monotonous kernel clock) to see and react to NTP/RTC time jumps - which along with loops taking too long can be due to sleep or winter/summer time changes. Again, the purpose would be to tailor the reaction (and noise made) to this situation.

Enthusiasts welcome to take on this, especially those who can test it well (have systems that do go to sleep while monitoring)!

braiam, 2024-06-21:
Yeah, I've had issues with the system turning off after a sleep, when I know that the power will be out for 10-15 minutes, which makes me willing to deactivate the nut-monitor service before suspension to avoid such things. The log demonstrating the event:

```
2024-06-21T13:06:11+00:00 nut-monitor[1449]: UPS nutdev1@localhost on battery
2024-06-21T13:06:11+00:00 nut-monitor[143430]: Network UPS Tools upsmon 2.8.1
2024-06-21T13:16:56+00:00 nut-server[1424]: Data for UPS [nutdev1] is stale - check driver
2024-06-21T13:16:56+00:00 nut-server[1424]: UPS [nutdev1] data is no longer stale
2024-06-21T13:16:56+00:00 nut-monitor[1449]: Poll UPS [nutdev1@localhost] failed - Write error: Broken pipe
2024-06-21T13:16:56+00:00 nut-monitor[1449]: Communications with UPS nutdev1@localhost lost
2024-06-21T13:16:56+00:00 nut-monitor[1449]: UPS [nutdev1@localhost] was last known to be not fully online and currently is not communicating, assuming dead
2024-06-21T13:16:56+00:00 nut-monitor[1449]: FSD set on UPS nutdev1@localhost failed: Driver not connected
2024-06-21T13:16:56+00:00 nut-monitor[1449]: Executing automatic power-fail shutdown
2024-06-21T13:16:56+00:00 nut-monitor[1449]: Auto logout and shutdown proceeding
2024-06-21T13:16:55+00:00 upsd[1424]: Data for UPS [nutdev1] is stale - check driver
2024-06-21T13:16:56+00:00 nut-monitor[143737]: Network UPS Tools upsmon 2.8.1
2024-06-21T13:16:55+00:00 upsd[1424]: UPS [nutdev1] data is no longer stale
2024-06-21T13:16:56+00:00 nut-monitor[143742]: Network UPS Tools upsmon 2.8.1
2024-06-21T13:17:00+00:00 nut-monitor[1449]: Network UPS Tools upsmon 2.8.1
2024-06-21T13:17:02+00:00 shutdown[144102]: Failed to set wall message, ignoring: Message recipient disconnected from message bus without replying
2024-06-21T13:17:02+00:00 shutdown[144102]: Shutdown scheduled for Fri 2024-06-21 09:17:00 AST, use 'shutdown -c' to cancel.
2024-06-21T13:17:02+00:00 nut-monitor[1438]: Network UPS Tools upsmon 2.8.1
2024-06-21T13:17:02+00:00 systemd[1]: nut-monitor.service: Deactivated successfully.
```

Ropid, 2024-08-15:
After experimenting a bit, I ended up with this service file here to work around the sleep issue, it stops and restarts `nut.target` before and after sleep:

```
# /etc/systemd/system/nut-sleep.service
[Unit]
Description=NUT sleep hook
Before=sleep.target
StopWhenUnneeded=yes

[Service]
Type=oneshot
RemainAfterExit=yes
ExecStart=/usr/bin/systemctl stop nut.target
ExecStop=/usr/bin/systemctl start --no-block nut.target

[Install]
WantedBy=sleep.target
```

jimklimov, 2024-08-15:
Looks quite neat, in fact!

PR #2597 (2024-08-23): Implement (currently systemd) OS sleeping support in `upsmon` client
Main goal: detect going to sleep and subsequently waking up, in which case we neuter collected UPS status bits and last-read timestamps to avoid shutdowns because of stale data in a last-known-critical situation, and so rely only on up-to-date (re-)connections and data reads in this case. This problem is more likely if `SHUTDOWNCMD="systemctl suspend/hibernate/..."` or equivalent is used in `upsmon.conf` as the reaction to depleting UPS batteries.

Design is currently centered around libsystemd concepts, but is hopefully exte [... cut]
closed 2024-08-23 by jimklimov