networkupstools/nut #1837: Amazon Basic ABST600 supported by usbhid-ups

BigGeorgeTx, 2023-01-23:
I was able to get this to work with the usbhid-ups driver, as suggested by someone in the Amazon questions section for the device

"Q: Will this ups work with the linux nut (network ups tools) software in place of the windows/mac software provided?
A: Yes. At least on my 600VA unit...

Here's my /etc/ups/ups.conf:

[ABST600]
driver = usbhid-ups
port = auto
desc = "The Amazon UPS"

And my /etc/ups/upsd.users:

[admin]
password = mypass
actions = SET
instcmds = ALL

[testuser]
password = pass
instcmds = test.battery.start
instcmds = test.battery.stop

I plugged in the USB cable, started the driver and daemon:

sudo upsdrvctl start
sudo upsd

Then checked the beeper status:
sudo upsc ABST600 ups.beeper.status

Then disabled it...
sudo upscmd ABST600 beeper.disable [see less](https://www.amazon.com/AmazonBasics-Standby-UPS-600VA-Outlets/dp/B073Q48YGF/ref=sr_1_1_sspa?crid=15DR9WJ1JXZPL&keywords=ups%2Bbattery%2Bbackup&qid=1674513612&sprefix=ups%2Caps%2C159&sr=8-1-spons&spLa=ZW5jcnlwdGVkUXVhbGlmaWVyPUExVFMzMjNBQUlQSEFTJmVuY3J5cHRlZElkPUEwNzExMzEyMjBENlY0VU9VVlNFRCZlbmNyeXB0ZWRBZElkPUEwNDAyNDgwS1FCR1MwSDBFQVk3JndpZGdldE5hbWU9c3BfYXRmJmFjdGlvbj1jbGlja1JlZGlyZWN0JmRvTm90TG9nQ2xpY2s9dHJ1ZQ&th=1#)"

The model number on the device and the FCC number is the same as a Cyberpower device that suggests a different drive, but that didn't work and this driver does.

jimklimov, 2023-01-24:
Thanks! Is there a device-data dump available? (The NUT keys:values)?

jimklimov, 2023-01-24:
Yes, thanks!