networkupstools/nut #1843: vULTECH UPS1050VA-PRO HOMEASSISTANT

Achy7, 2023-02-01:
Hi all,
on raspberry hassio i can try work nut server . i cant config i try with : 

- name: apcups
  driver: nutdrv_qx
  port: auto
  config:
    - vendorid = 0925
    - productid = 1234
    - subdriver = armac

and i have this error : 

[18:00:44] INFO: Configuring Device named apcups...
[18:00:45] INFO: Starting the UPS drivers...
Subdriver 'armac' not found!
Network UPS Tools - Generic Q* USB/Serial driver 0.28 (2.7.4)
USB communication driver 0.33
Network UPS Tools - UPS driver controller 2.7.4
Driver failed to start (exit status=1)
cont-init: info: /etc/cont-init.d/nut.sh exited 1
cont-init: info: running /etc/cont-init.d/nutclient.sh
cont-init: info: /etc/cont-init.d/nutclient.sh exited 0
cont-init: warning: some scripts exited nonzero
s6-rc: warning: unable to start service legacy-cont-init: command exited 1
/run/s6/basedir/scripts/rc.init: warning: s6-rc failed to properly bring all the services up! Check your logs (in /run/uncaught-logs/current if you have in-container logging) for more information.
/run/s6/basedir/scripts/rc.init: fatal: stopping the container.
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

WHAT u suggest me ?

jimklimov, 2023-02-01:
It could help to read the errors you've posted, e.g. `Subdriver 'armac' not found!` and the project release change log.

According to https://github.com/networkupstools/nut/blob/ee1aa59ece0feb1762f8967470931f0533523f10/NEWS#L360-L362 and history at https://github.com/networkupstools/nut/blame/24e56f02bcada781e4dff32aa763da49e74c905f/drivers/nutdrv_qx.c#L1655 it was added in Dec 2021, so became part of NUT v2.8.0 release.

The NUT v2.7.4 referenced above was released Mar 9, 2016 so quite a bit before.

Also, why is the device misleadingly called "apcups"? Not a problem per se, but confusing ;)

Looking at USB IDs, an older `riello_usb` might handle the device in the release you have.

1achy, 2023-02-01:
I follow suggestion for Riello driver:

[image]

Now it start with some error i check

[image]
[image]
[image]
[image]

jimklimov, 2023-02-01:
Hm, try without the `productid`, support to tune that in riello drivers must have appeared more recently than 7 years ago :\

jimklimov, 2023-02-01:
FWIW, you can also try to follow an instruction posted by a community member about building a newer HA plugin container with NUT, to try your chances with a recent GitHub snapshot: https://github.com/networkupstools/nut/wiki/Building-NUT-integration-for-Home-Assistant (otherwise, can't help about HA integration per se, never used it myself)

1achy, 2023-02-01:
> Hm, try without the `productid`, support to tune that in riello drivers must have appeared more recently than 7 years ago :\

Without productid It Is dont found and server go in stop

[image]
[image]

1achy, 2023-02-01:
[image]

jimklimov, 2023-02-02:
Oh, it slipped my mind that in your earlier screenshot you ONLY had a productid, but not a vendorid. I believe both are required as part of one identifier.

1achy, 2023-02-02:
> Oh, it slipped my mind that in your earlier screenshot you ONLY had a productid, but not a vendorid. I believe both are required as part of one identifier.

thanks but i try with all combine but no work another try with 👍 

config: 

- name: vultechups
  driver: richcomm_usb
  port: auto
  config: []

now i see some information:

[image]

log:
 0.034603	Shutting down any UPSes in MASTER mode...
 258.314971	Client upsmonmaster@127.0.0.1 set FSD on UPS [vultechups]
   0.040419	This system is a master... waiting for slave logout...
   0.047163	Executing automatic power-fail shutdown
{"message":"Event nut.ups_event fired."}Network UPS Tools upsmon 2.7.4
{"message":"Event nut.ups_event fired."}Network UPS Tools upsmon 2.7.4
{"message":"Event nut.ups_event fired."}Network UPS Tools upsmon 2.7.4
s6-svscanctl: fatal: unable to control /var/run/s6/services: No such file or directory
   5.107765	Unable to call shutdown command: s6-svscanctl -t /var/run/s6/services
Network UPS Tools upsmon 2.7.4
[18:50:15] INFO: upsmon stopped, restarting...
[18:50:16] INFO: Starting the UPS monitor and shutdown controller...
kill: No such process
   0.000000	Using power down flag file /etc/killpower
   0.000831	UPS: vultechups@localhost (master) (power value 1)
   0.004319	debug level is '1'
   0.004459	Warning: running as one big root process by request (upsmon -p)
   0.007019	Init SSL without certificate database
   0.022088	Trying to connect to UPS [vultechups@localhost]
 264.478028	User upsmonmaster@127.0.0.1 logged into UPS [vultechups]
   0.029760	Logged into UPS vultechups@localhost
   0.036578	Critical UPS: vultechups@localhost
   0.036882	Shutting down any UPSes in MASTER mode...
 264.490343	Client upsmonmaster@127.0.0.1 set FSD on UPS [vultechups]
   0.041616	This system is a master... waiting for slave logout...
   0.042702	Executing automatic power-fail shutdown
{"message":"Event nut.ups_event fired."}{"message":"Event nut.ups_event fired."}Network UPS Tools upsmon 2.7.4
Network UPS Tools upsmon 2.7.4
{"message":"Event nut.ups_event fired."}Network UPS Tools upsmon 2.7.4
s6-svscanctl: fatal: unable to control /var/run/s6/services: No such file or directory
   5.082934	Unable to call shutdown command: s6-svscanctl -t /var/run/s6/services
Network UPS Tools upsmon 2.7.4
[18:50:21] INFO: upsmon stopped, restarting...
[18:50:22] INFO: Starting the UPS monitor and shutdown controller...
kill: No such process
   0.000000	Using power down flag file /etc/killpower
   0.001775	UPS: vultechups@localhost (master) (power value 1)
   0.005093	debug level is '1'
   0.005235	Warning: running as one big root process by request (upsmon -p)
   0.008056	Init SSL without certificate database
   0.021263	Trying to connect to UPS [vultechups@localhost]
 270.574178	User upsmonmaster@127.0.0.1 logged into UPS [vultechups]
   0.030041	Logged into UPS vultechups@localhost
   0.035464	Critical UPS: vultechups@localhost
   0.037535	Shutting down any UPSes in MASTER mode...
 270.585892	Client upsmonmaster@127.0.0.1 set FSD on UPS [vultechups]
   0.040299	This system is a master... waiting for slave logout...
   0.041848	Executing automatic power-fail shutdown
{"message":"Event nut.ups_event fired."}{"message":"Event nut.ups_event fired."}Network UPS Tools upsmon 2.7.4
Network UPS Tools upsmon 2.7.4
{"message":"Event nut.ups_event fired."}Network UPS Tools upsmon 2.7.4
s6-svscanctl: fatal: unable to control /var/run/s6/services: No such file or directory
   5.087024	Unable to call shutdown command: s6-svscanctl -t /var/run/s6/services
Network UPS Tools upsmon 2.7.4
[... 21 lines cut ...]
Network UPS Tools upsmon 2.7.4
   5.086112	Unable to call shutdown command: s6-svscanctl -t /var/run/s6/services
[18:50:33] INFO: upsmon stopped, restarting...
[18:50:34] INFO: Starting the UPS monitor and shutdown controller...
kill: No such process
   0.000000	Using power down flag file /etc/killpower
   0.001434	UPS: vultechups@localhost (master) (power value 1)
   0.002913	debug level is '1'
   0.003047	Warning: running as one big root process by request (upsmon -p)
   0.005525	Init SSL without certificate database
   0.018042	Trying to connect to UPS [vultechups@localhost]
 283.068677	User upsmonmaster@127.0.0.1 logged into UPS [vultechups]
   0.024361	Logged into UPS vultechups@localhost
   0.028641	Critical UPS: vultechups@localhost
   0.028911	Shutting down any UPSes in MASTER mode...
 283.074794	Client upsmonmaster@127.0.0.1 set FSD on UPS [vultechups]
   0.030576	This system is a master... waiting for slave logout...
   0.031684	Executing automatic power-fail shutdown
{"message":"Event nut.ups_event fired."}Network UPS Tools upsmon 2.7.4
{"message":"Event nut.ups_event fired."}{"message":"Event nut.ups_event fired."}Network UPS Tools upsmon 2.7.4
Network UPS Tools upsmon 2.7.4
s6-svscanctl: fatal: unable to control /var/run/s6/services: No such file or directory
   5.066415	Unable to call shutdown command: s6-svscanctl -t /var/run/s6/services
Network UPS Tools upsmon 2.7.4
[18:50:40] INFO: upsmon stopped, restarting...

jimklimov, 2023-02-03:
And is it really on-battery and depleted at that moment? So is the shutdown correct or bogus?..

1achy, 2023-02-05:
> And is it really on-battery and depleted at that moment? So is the shutdown correct or bogus?..

Sorry for delay but system in other house. Check and is bug . when change status no change on HA.

jimklimov, 2023-02-05:
I wonder if it is then due to this device talking some other protocol dialect than those supported in NUT 2.7.4 release. Quite a few Qx subdrivers were added by 2.8.0 and maybe later too.

jimklimov, 2023-02-05:
In that case you may have to investigate building a HA plugin with NUT...

Is the OS there a general-purpose Linux running HA as an app/service/container? Can you try building current NUT on the raspberry directly (not necessarily installing it), to check if it sees your device data well? You'd need to run the driver in build workplace picking arguments for subdrivers and producing a data dump, until one hopefully makes sense. Less moving parts and all that... (to rule out HA/containers being the problem).

1achy, 2023-10-22:
[image]

HELP

jimklimov, 2023-10-22:
Which NUT version are you running with at the moment? An number of fixes for the "armac" sub-driver got merged recently (in master branch), are you using that?

* #2005
* #2003
* #1978

Other than that, it seems like the device disconnects on its own (may be similar to https://github.com/networkupstools/nut/wiki/Troubleshooting-eventual-disconnections-(Data-stale) and might be alleviated by a more frequent polling, unless it is some issue with the controller in the UPS so it reboots and gets re-discovered). Re-enumeration as a different device number seems a bit odd, unless you've re-pluged stuff (not necessarily the UPS itself, but some kbd/mouse/flash...) - however it does say that one device could not be enabled => check the cable... did you try another?

cash83, 2024-04-24:
Hi, has anyone managed to make it work?  I sound new and I'm learning.  If you can give me a hand, thanks
[image]

ilpol984, 2024-09-20:
Hello, 
I've menaged to work VULTECH UPS700VA-XS with nut 2.8.2.

It doesn't work out of the box thow. I've to comment out some lines of the nutdrv_qx as the ups report the state FSD even if it's simply on-line (probably the FSD state is readed in a wrong position but I cannot find the place where the driver reads it to correct it.

Here you can find my modifications (nutdrv_qx.c):
   4179         //if (ups_status & STATUS(FSD)) {
   4180         //              alarm_set("Shutdown imminent!");
   4181         //}

   4222         if (ups_status & STATUS(FSD)) {
   4223 //              status_set("FSD");              /* Forced shutdown */
   4224         }

Compiled with:

./configure --prefix=/usr --with-user=nut --with-group=nut --sysconfdir=/etc/nut
make
make install

And this is my ups.conf:
[nutdev1]
   driver = "nutdrv_qx"
   port = "auto"
   vendorid = 0925
   productid = 1234
   subdriver = armac
   bus = "001"

Does anybody knows where the drivers read the FSD state, so I can try a better fix?

Thank you, regards,