networkupstools/nut #1844: Services will not start with OS when installing from source

RickoT, 2023-02-02:
Greetings!

I had to use source to build NUT because the Debian packages available on the Raspberry Pi do not include the latest MIBS for my UPS.  I followed the instructions to set everything up and it all works brilliantly EXCEPT the services do not start on boot.  Below are all the steps I followed to set up the Pi post installation

```
raspi-config
	System -> Hostname
	System -> Network at Boot
	Advanced -> Expand File System

echo 'interface eth0' >> /etc/dhcpcd.conf
echo 'static ip_address=<<IP>>' >> /etc/dhcpcd.conf
echo 'static routers=<<Gateway>>' >> /etc/dhcpcd.conf
echo 'static domain_name_servers=<<DNS>>' >> /etc/dhcpcd.conf

echo 'net.ipv6.conf.all.disable_ipv6 = 1' >> /etc/sysctl.conf 
echo 'net.ipv6.conf.default.disable_ipv6 = 1' >> /etc/sysctl.conf 
echo 'net.ipv6.conf.lo.disable_ipv6 = 1' >> /etc/sysctl.conf 

echo 'dtoverlay=disable-wifi' >> /boot/config.txt
echo 'dtoverlay=disable-bt' >> /boot/config.txt
reboot

apt update && apt upgrade -y
apt install git autoconf automake libtool libgd-dev libperl-dev lighttpd libsnmp-dev libltdl-dev libsystemd-dev -y

useradd ups
groupadd nut	
usermod -aG nut ups
mkdir -p /var/state/ups
chmod 0770 /var/state/ups
chown root:nut /var/state/ups

reboot

git clone https://github.com/networkupstools/nut.git
cd nut

./autogen.sh
./configure \
	--with-user=ups \
	--with-group=nut  \
	--with-cgi \
	--with-snmp \
	--with-libsystemd \
	--with-wrap \
	--with-python3 \
	--without-usb \
	--without-serial \
	--without-neon \
	--without-macosx_ups \
	--without-linux_i2c \
	--without-ipmi \
	--without-powerman \
	--without-modbus
	
make
make install

echo "PATH=$PATH:/usr/local/ups/bin:/usr/local/ups/sbin" >> /etc/environment
systemctl enable nut-monitor
systemctl enable nut-server
systemctl enable nut-driver-enumerator

lighttpd-enable-mod cgi
echo 'server.modules += ( "mod_cgi" )' > /etc/lighttpd/conf-enabled/10-cgi.conf
echo '' >> /etc/lighttpd/conf-enabled/10-cgi.conf
echo '$HTTP["url"] =~ "^/cgi-bin/" {' >> /etc/lighttpd/conf-enabled/10-cgi.conf
echo '  cgi.assign = ( "" => "" )' >> /etc/lighttpd/conf-enabled/10-cgi.conf
echo '  alias.url += ( "/nut/cgi-bin" => "/usr/local/ups/cgi-bin" )' >> /etc/lighttpd/conf-enabled/10-cgi.conf
echo '  alias.url += ( "/cgi-bin/nut" => "/usr/local/ups/cgi-bin" )' >> /etc/lighttpd/conf-enabled/10-cgi.conf
echo '  alias.url += ( "/nut" => "/usr/local/ups/etc" )' >> /etc/lighttpd/conf-enabled/10-cgi.conf
echo '}' >> /etc/lighttpd/conf-enabled/10-cgi.conf

reboot
```

Here is the result of systemctl nut-monitor (same for nut-server, nut-driver-enumerator) after rebooting
```
● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; vendor preset: enabled)
     Active: inactive (dead)
```

jimklimov, 2023-02-04:
Most of the NUT units are `WantedBy=nut.target` (which is in turn `WantedBy=multi-user.target` so bound to system startup), and I don't see it enabled in your detalization above.

FYI, recently published https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

RickoT, 2023-02-04:
Perfect!  running systemctl enable nut.target resolved my issue.  I did not see this step in the official installation documentation, so that may need to be added.

Here is the documentation i was following to base my steps on : https://networkupstools.org/docs/user-manual.chunked/ar01s05.html

Thanks!
~Rick

RickoT, 2023-02-04:
In the future, will running systemctl enable nut.target also enable the dependencies or do I need to run enable on the nut-monitor, nut-server as well?

jimklimov, 2023-02-05:
Thanks, will check!  

You should enable all those you need - and it allows to not start daemons you do not need (e.g. upsd on systems that are only `upsmon secondary`)

closed 2023-02-08 by jimklimov