networkupstools/nut #1846: huawei-ups2000: Data Stale with Endless "Register has a read failure. Retrying..."

biergaizi, 2023-02-08:
On my server, I've recently discovered a failure mode in `huawei-ups2000` driver in NUT 2.8.0. After running NUT with huawei-ups2000 driver for a while, perhaps for days or weeks, the driver would suddenly fail at a point, then `huawei-ups2000` would retry the read, unfortunately, none of the retries would be successfully, and the system log would eventually be spammed by endless retry attempts, such as:

If the physical link is broken (e.g. due to a USB-to-serial-converter disconnect and reconnect), this is the expected behavior since the underlying serial device has disappeared and there's nothing that NUT can do, and it has already been documented in the man page. However, inspection of the system log showed that no disconnection occurred, and the error was purely logical in nature.

```
Jan 18 15:00:00 freebsd newsyslog[20173]: logfile turned over due to size>1000K
Jan 18 15:00:01 freebsd huawei-ups2000[81806]: Register 12000 has a read failure. Retrying...
Jan 18 15:00:03 freebsd syslogd: last message repeated 2 times
Jan 18 15:00:04 freebsd huawei-ups2000[81806]: Register 12000 has a fatal read failure.
Jan 18 15:00:04 freebsd huawei-ups2000[81806]: register 1024 has invalid value 0008,
Jan 18 15:00:04 freebsd huawei-ups2000[81806]: Register 41180 has a read failure. Retrying...
Jan 18 15:00:06 freebsd syslogd: last message repeated 2 times
Jan 18 15:00:07 freebsd huawei-ups2000[81806]: Register 41180 has a fatal read failure.
Jan 18 15:00:07 freebsd huawei-ups2000[81806]: Register 19009 has a read failure. Retrying...
Jan 18 15:00:08 freebsd syslogd: last message repeated 1 times
Jan 18 15:00:09 freebsd huawei-ups2000[81806]: register 1024 has invalid value 000a,
Jan 18 15:00:09 freebsd huawei-ups2000[81806]: Register 41180 has a read failure. Retrying...
Jan 18 15:00:11 freebsd syslogd: last message repeated 2 times
Jan 18 15:00:12 freebsd huawei-ups2000[81806]: Register 41180 has a fatal read failure.
Jan 18 15:00:12 freebsd huawei-ups2000[81806]: Register 11043 has a read failure. Retrying...
Jan 18 15:00:13 freebsd syslogd: last message repeated 1 times
Jan 18 15:00:15 freebsd huawei-ups2000[81806]: Register 12002 has a read failure. Retrying...
Jan 18 15:00:16 freebsd syslogd: last message repeated 1 times
Jan 18 15:00:17 freebsd huawei-ups2000[81806]: Register 12002 has a fatal read failure.
Jan 18 15:00:17 freebsd huawei-ups2000[81806]: Register 41180 has a read failure. Retrying...
Jan 18 15:00:18 freebsd huawei-ups2000[81806]: Register 19009 has a read failure. Retrying...
Jan 18 15:00:19 freebsd syslogd: last message repeated 1 times
Jan 18 15:00:20 freebsd huawei-ups2000[81806]: Register 11043 has a read failure. Retrying...
Jan 18 15:00:22 freebsd syslogd: last message repeated 2 times
Jan 18 15:00:23 freebsd huawei-ups2000[81806]: Register 11043 has a fatal read failure.
Jan 18 15:00:26 freebsd huawei-ups2000[81806]: register 1024 has invalid value 000a,
Jan 18 15:00:28 freebsd huawei-ups2000[81806]: Register 41180 has a read failure. Retrying...
Jan 18 15:00:29 freebsd syslogd: last message repeated 1 times
Jan 18 15:00:30 freebsd huawei-ups2000[81806]: Register 11000 has a read failure. Retrying...
Jan 18 15:00:31 freebsd syslogd: last message repeated 1 times
Jan 18 15:00:32 freebsd huawei-ups2000[81806]: Register 12000 has a read failure. Retrying...
Jan 18 15:00:34 freebsd syslogd: last message repeated 2 times
Jan 18 15:00:35 freebsd huawei-ups2000[81806]: Register 12000 has a fatal read failure.
```

Unfortunately, the error code or message is not logged and the nature of the issue could not be investigated easily. To investigate the issue further, I submitted the patch https://github.com/networkupstools/nut/commit/da131f52d29c969d1acd7da36294cd54dc3fc6fc to the upstream. And I was able to get some diagnostics after the error occurred again:

```
Feb  2 11:01:47 freebsd huawei-ups2000[66317]: modbus_read_registers() failed (-1, errno 60): Operation timed out
Feb  2 11:01:47 freebsd huawei-ups2000[66317]: Register 11000 has a read failure. Retrying...
Feb  2 11:01:48 freebsd huawei-ups2000[66317]: modbus_read_registers() failed (-1, errno 112345690): Invalid CRC
Feb  2 11:01:48 freebsd huawei-ups2000[66317]: Register 11000 has a read failure. Retrying...
Feb  2 11:01:49 freebsd huawei-ups2000[66317]: modbus_read_registers() failed (-1, errno 112345695): Response not from requested slave
Feb  2 11:01:49 freebsd huawei-ups2000[66317]: Register 11000 has a read failure. Retrying...
Feb  2 11:01:50 freebsd huawei-ups2000[66317]: modbus_read_registers() failed (-1, errno 112345695): Response not from requested slave
Feb  2 11:01:50 freebsd huawei-ups2000[66317]: Register 11000 has a fatal read failure.
Feb  2 11:01:51 freebsd upsd[67279]: Data for UPS [huawei] is stale - check driver
Feb  2 11:01:52 freebsd upsd[67279]: UPS [huawei] data is no longer stale
Feb  2 12:52:49 freebsd huawei-ups2000[66317]: modbus_read_registers() failed (-1, errno 60): Operation timed out
Feb  2 12:52:49 freebsd huawei-ups2000[66317]: Register 12000 has a read failure. Retrying...
Feb  2 12:52:50 freebsd huawei-ups2000[66317]: modbus_read_registers() failed (-1, errno 112345690): Invalid CRC
Feb  2 12:52:50 freebsd huawei-ups2000[66317]: Register 12000 has a read failure. Retrying...
Feb  2 12:52:51 freebsd huawei-ups2000[66317]: modbus_read_registers() failed (-1, errno 112345690): Invalid CRC
Feb  2 12:52:51 freebsd huawei-ups2000[66317]: Register 12000 has a read failure. Retrying...
Feb  2 12:52:52 freebsd huawei-ups2000[66317]: modbus_read_registers() failed (-1, errno 112345690): Invalid CRC
Feb  2 12:52:52 freebsd huawei-ups2000[66317]: Register 12000 has a fatal read failure.
Feb  2 12:52:52 freebsd upsd[67279]: Data for UPS [huawei] is stale - check driver
```

Thus, I'm pretty confident about how the chain of events:

1. Currently, the read timeout value is set to 1 second. Unfortunately, sometimes the UPS takes longer to respond. Thus, `modbus_read_registers()` times out and raises an error.
2. When the driver attempts to reread the register again, instead of getting the result of the current request, the partial stale data from the previous read is read from the serial buffer, thus, it causes this read to also fail with an "Invalid CRC" error.
3. Step 2 would reoccur repeatedly and none of the retry would be successful.

Now I'm testing a patch to flush the buffer via `modbus_flush()` first before calling `modbus_read_registers()`. The timeout value should also be bumped to perhaps 2 seconds.

I'll send a pull request if my test shows that the patch is functional.

Please tag this issue for NUT 2.8.1 so it can be fixed before a new release is made.

jimklimov, 2023-02-08:
Thanks for the investigation and report! :)

Is the timeout there a maximum value, or the timeframe for request (will it block for e.g. 2 seconds, or can return faster if all is ok)? If it does block for specified time regardless of speed of response, and a slow reply is a rare occurrence (runs ok for weeks and months), maybe a back-off mechanism makes sense (retry with longer timeout only if a short one failed)?..

> If the physical link is broken (e.g. due to a USB-to-serial-converter disconnect and reconnect), this is the expected behavior since the underlying serial device has disappeared and there's nothing that NUT can do, and it has already been documented in the man page.

Actually some drivers do offer detection of disconnection (or assumption of that, e.g. based on last successful interaction being too many cycles ago; possibly USB drivers also already track a permissions-denied after a successful init - probably meaning reconnect and kernel/udev/... grabbing the device node), and try to reconnect. Would this approach help here?

biergaizi, 2023-02-08:
> Is the timeout there a maximum value, or the timeframe for request (will it block for e.g. 2 seconds, or can return faster if all is ok)?

This is a global serial data read timeout in libmodbus, it's the time interval between sending a request and receiving a valid Modbus message. If the device responses earlier, the request  `modbus_read_registers()` returns immediately. If no valid message has been received, `modbus_read_registers()` returns with an error message. Apparently sometimes the device takes longer to execute a query message. 

> Actually some drivers do offer detection of disconnection (or assumption of that, e.g. based on last successful interaction being too many cycles ago; possibly USB drivers also already track a [...]

Yes, in principle it should be possible to close the device descriptor and reopen it to recover. But with caveat - the kernel must register the USB-to-serial converter as a device file with exactly the same name, this cannot be guaranteed without explicitly configuring it in udev. So this should be carefully documented if we're going to implement it. Also, if it's to be implemented, it would be a good idea to present the same behavior in all device drivers that use a serial port. Perhaps as a requirement in the coding standard.

biergaizi, 2023-02-11:
Good news, the timeout errors reoccurred on my server, proving that the fix was effective as expected. I'll submit a patch soon.

```
Feb 10 03:40:28 freebsd huawei-ups2000[75057]: modbus_read_registers() failed (-1, errno 60): Operation timed out
Feb 10 03:40:28 freebsd huawei-ups2000[75057]: Register 12000 has a read failure. Retrying...
Feb 10 07:52:05 freebsd huawei-ups2000[75057]: modbus_read_registers() failed (-1, errno 60): Operation timed out
Feb 10 07:52:05 freebsd huawei-ups2000[75057]: Register 12000 has a read failure. Retrying...
Feb 10 08:32:35 freebsd huawei-ups2000[75057]: modbus_read_registers() failed (-1, errno 60): Operation timed out
Feb 10 08:32:35 freebsd huawei-ups2000[75057]: Register 12000 has a read failure. Retrying...
Feb 10 10:25:26 freebsd huawei-ups2000[75057]: modbus_read_registers() failed (-1, errno 60): Operation timed out
Feb 10 10:25:26 freebsd huawei-ups2000[75057]: Register 41180 has a read failure. Retrying...
Feb 10 10:59:04 freebsd huawei-ups2000[75057]: modbus_read_registers() failed (-1, errno 60): Operation timed out
Feb 10 10:59:04 freebsd huawei-ups2000[75057]: Register 12000 has a read failure. Retrying...
Feb 10 16:26:59 freebsd huawei-ups2000[75057]: modbus_read_registers() failed (-1, errno 60): Operation timed out
Feb 10 16:26:59 freebsd huawei-ups2000[75057]: Register 11000 has a read failure. Retrying...
```

biergaizi, 2023-02-11:
I've just submitted Pull Request #1847 to fix the issue.

closed 2023-02-12 by jimklimov