networkupstools/nut #1849: support Minuteman device PRO1500RT2U (a0a0)

JonGilmore, 2023-02-12:
I'm testing NUT on debian with a Minuteman PRO1500RT2U and I'm unable to get it working. Upon looking at https://github.com/networkupstools/nut/issues/555, it appears that device `a0a0` isn't currently supported. PR to add this support: https://github.com/networkupstools/nut/pull/1848

Please let me know if we need more debug info than I've provided below

```
lsusb
Bus 001 Device 002: ID 8087:8001 Intel Corp. Integrated Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 003: ID 8087:0a2a Intel Corp. Bluetooth wireless interface
Bus 002 Device 002: ID 05dd:a0a0 Delta Electronics, Inc. MINUTEMAN UPS
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
/lib/nut/usbhid-ups -DDDDD -a minuteman
... snipped ...
   0.489349	Checking device (05DD/A0A0) (002/002)
   0.499736	- VendorID: 05dd
   0.499773	- ProductID: a0a0
   0.499790	- Manufacturer: Minuteman
   0.499810	- Product: MINUTEMAN UPS
   0.499828	- Serial Number: unknown
   0.499846	- Bus: 002
   0.499865	- Device release number: 0002
   0.499887	Trying to match device
   0.499913	Device does not match - skipping
```

jimklimov, 2023-02-12:
Is this "not match" with a build of your PR branch?

JonGilmore, 2023-02-12:
> Is this "not match" with a build of your PR branch?

Sorry, can you clarify your question? When I was running `usbhid-ups` to get a debug output, I just wanted to demonstrate that the actual device that is shown with `lsusb` is not matching anything that `nut` recognizes. Hopefully that clears things up.

jimklimov, 2023-02-12:
Ok, thanks!

closed 2023-03-02 by jimklimov