networkupstools/nut #1851: Network UPS Tools (NUT) "Can't connect to UPS" - Tripp Lite SMART500RT1U

Jsalas424, 2023-02-14:
I'm running a bare bones NUT configuration to monitor a single UPS that is [stated to be compatible on the website][1] - I can't get it to work.

This is a Tripp Lite SMART500RT1U connected via USB cable.

My syslog says it's there:
```
Feb 14 16:54:36 Server kernel: usb 2-6: new low-speed USB device number 4 using xhci_hcd
Feb 14 16:54:36 Server kernel: usb 2-6: New USB device found, idVendor=09ae, idProduct=0001, bcdDevice= 0.0a
Feb 14 16:54:36 Server kernel: usb 2-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Feb 14 16:54:36 Server kernel: usb 2-6: Product: TRIPP LITE SMART500RT1U 
Feb 14 16:54:36 Server kernel: usb 2-6: Manufacturer: Tripp Lite 
Feb 14 16:54:36 Server kernel: hid-generic 0003:09AE:0001.0003: hiddev0,hidraw2: USB HID v1.10 Device [Tripp Lite  TRIPP LITE SMART500RT1U ] on usb-0000:00:14.0-6/input0

root@Server:~# lsusb
...
Bus 002 Device 004: ID 09ae:0001 Tripp Lite TRIPP LITE SMART500RT1U 
```

Here's my nut config:

/etc/nut/ups.conf
```
[UPS]
        driver = tripplite_usb
        port = auto
        vendorid = 09ae
        productid = 0001
```

/etc/nut/nut.conf
```
MODE=standalone
```

:(
```
root@Server:~# systemctl restart nut-server
root@Server:~# systemctl status nut-server
● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-02-14 17:11:38 EST; 3s ago
    Process: 1417620 ExecStart=/sbin/upsd (code=exited, status=0/SUCCESS)
   Main PID: 1417621 (upsd)
      Tasks: 1 (limit: 38378)
     Memory: 680.0K
        CPU: 4ms
     CGroup: /system.slice/nut-server.service
             └─1417621 /lib/nut/upsd

Feb 14 17:11:38 Server systemd[1]: Starting Network UPS Tools - power devices information server...
Feb 14 17:11:38 Server upsd[1417620]: fopen /run/nut/upsd.pid: No such file or directory
Feb 14 17:11:38 Server upsd[1417620]: listening on 127.0.0.1 port 3493
Feb 14 17:11:38 Server upsd[1417620]: listening on ::1 port 3493
Feb 14 17:11:38 Server upsd[1417620]: listening on 127.0.0.1 port 3493
Feb 14 17:11:38 Server upsd[1417620]: listening on ::1 port 3493
Feb 14 17:11:38 Server upsd[1417620]: Can't connect to UPS [UPS] (tripplite_usb-UPS): No such file >
Feb 14 17:11:38 Server upsd[1417620]: Can't connect to UPS [UPS] (tripplite_usb-UPS): No such file >
Feb 14 17:11:38 Server upsd[1417621]: Startup successful
Feb 14 17:11:38 Server systemd[1]: Started Network UPS Tools - power devices information server.
```

NUT v2.7.4-13

  [1]: https://networkupstools.org/ddl/Tripp_Lite/SMART500RT1U.html

jimklimov, 2023-02-14:
Can't see from the posted logs - did you start the `nut-driver` for the device? did that fail?

FWIW might also try `usbhid-ups` in case the device (also?) talks the standard protocol...

Jsalas424, 2023-02-15:
@jimklimov  Well this seems like a stupid question, but how do I start the driver? I thought that using systemctl to start nut-server was all I needed to do. It's been a while since I configured NUT.

jimklimov, 2023-02-16:
Depends on distribution you use and if e.g. "mode" is a concept there. Also on NUT base version.

2.7.4 and some before it had a common `systemctl start nut-driver` for all devices as one unit.

2.8.0/master have a `nut-driver-enumerator` script and units to generate a `nut-driver@<NAME>.service` instances for each device separately.

Generally the 3-layer architecture is in play, so there's also the `nut-monitor` to run in most cases.

Jsalas424, 2023-02-16:
This is NUT v2.7.4-13 and I'm on Proxmox (fancy debian). I have NUT running on with other UPS' on these machines as well. 

Failed with the usbhid-ups driver
```
root@Server:~# journalctl -xe
Feb 16 07:20:07 Server upsdrvctl[3770664]: Network UPS Tools - UPS driver controller 2.7.4
Feb 16 07:20:07 Server upsdrvctl[3770666]: Can't open /run/nut/usbhid-ups-auto.pid: No such file or>
Feb 16 07:20:07 Server upsdrvctl[3770666]: Network UPS Tools - UPS driver controller 2.7.4
Feb 16 07:20:07 Server systemd[1]: nut-driver.service: Control process exited, code=exited, status=>
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStop= process belonging to unit nut-driver.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Feb 16 07:20:07 Server systemd[1]: nut-driver.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit nut-driver.service has entered the 'failed' state with result 'exit-code'.
Feb 16 07:20:07 Server systemd[1]: Failed to start Network UPS Tools - power device driver controll>
░░ Subject: A start job for unit nut-driver.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit nut-driver.service has finished with a failure.
░░ 
░░ The job identifier is 822887 and the job result is failed.
```

And with the tripplite_usb driver:

```
Feb 16 07:22:18 Server upsdrvctl[3772973]: Network UPS Tools - UPS driver controller 2.7.4
Feb 16 07:22:18 Server upsdrvctl[3772975]: Can't open /run/nut/tripplite_usb-auto.pid: No such file>
Feb 16 07:22:18 Server upsdrvctl[3772975]: Network UPS Tools - UPS driver controller 2.7.4
Feb 16 07:22:18 Server systemd[1]: nut-driver.service: Control process exited, code=exited, status=>
░░ Subject: Unit process exited
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ An ExecStop= process belonging to unit nut-driver.service has exited.
░░ 
░░ The process' exit code is 'exited' and its exit status is 1.
Feb 16 07:22:18 Server systemd[1]: nut-driver.service: Failed with result 'exit-code'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ The unit nut-driver.service has entered the 'failed' state with result 'exit-code'.
Feb 16 07:22:18 Server systemd[1]: Failed to start Network UPS Tools - power device driver controll>
░░ Subject: A start job for unit nut-driver.service has failed
░░ Defined-By: systemd
░░ Support: https://www.debian.org/support
░░ 
░░ A start job for unit nut-driver.service has finished with a failure.
░░ 
░░ The job identifier is 822948 and the job result is failed.
```

jimklimov, 2023-02-16:
Can you try to manually start it in command-line with boosted debug level? e.g. try a data dump with

`/lib/nut/usbhid-ups -a UPS -u root -DDDDDD -d1`

to see what it complains about when trying to match the device etc.

Maybe also cross-check with a custom build from master branch (see wiki for "inplace" mode as a quick shot)

Jsalas424, 2023-02-16:
@jimklimov thanks for that, here's the output:

Adapted your code to use the tripplite_usb driver
```
root@TracheNodeB:~# /lib/nut/tripplite_usb -a UPS -u root -DDDDDD -d1
Network UPS Tools - Tripp Lite OMNIVS / SMARTPRO driver 0.29 (2.7.4)
Warning: This is an experimental driver.
Some features may not function correctly.

/lib/nut/tripplite_usb: invalid option -- 'd'
   0.000000     Error: unknown option -?. Try -h for help.
```

Next, I attempted with the usbhid-ups driver and got a Fatal error:
```
root@TracheNodeB:~# /lib/nut/usbhid-ups -a UPS -u root -DDDDDD -d1
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33

Fatal error: 'maxretry' is not a valid variable name for this driver.

Look in the man page or call this driver with -h for a list of
valid variable names and flags.
```

Okay, so just remove that variable and try again:
```
root@TracheNodeB:~# /lib/nut/usbhid-ups -a UPS -u root -DDDDDD -d1
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
/lib/nut/usbhid-ups: invalid option -- 'd'
   0.000000     Error: unknown option -?. Try -h for help.
```
Well that's the same error for both drivers. Not sure what that "invalid option -- 'd'" thing means.

Jsalas424, 2023-02-16:
Reattempted with the other driver:

```
root@Server:~# /lib/nut/usbhid-ups -a UPS -u root -DDDDDD

....

  0.261067     Trying to match device
   0.261073     This Tripp Lite device (09ae/0001) is not supported by usbhid-ups.
Please use the tripplite_usb driver instead.

   0.261080     Device does not match - skipping
   0.261089     Checking device (09EB/0131) (002/003)
   0.262981     - VendorID: 09eb
   0.262985     - ProductID: 0131
   0.262988     - Manufacturer: Generic
   0.262992     - Product: USB
   0.262996     - Serial Number: unknown
   0.263001     - Bus: 002
   0.263006     - Device release number: 0110

```

jimklimov, 2023-02-17:
The `-d1` option is for dump mode (collect data in one cycle, print it and exit the driver) which is less sensitive to permissions (no PID file needed) etc. Seems your packaged NUT version predates that :\

jimklimov, 2023-02-17:
The `tripplite_usb` apparently sees the device with meaningful data, so the problem boils down to getting the driver started in normal conditions. That would likely be about permissions - e.g. would it still see the data if started without `-u root` option?

If not, check if the NUT package provided `udev.rules` file includes the `VID:PID` pair for this device so its devfs node would be accessible to `nut` runtime user? Alternately, the `user = root` can be specified in `ups.conf` section (but not recommended if easy to avoid).

rknichols, 2023-02-19:
I'm having the same problem in Rocky Linux 8 and also in Fedora 37. Those distros do provide 62-nut-usbups.rules file containing a matching idVendor and idProduct pair for each of my UPSs:
```
ATTR{idVendor}=="0764", ATTR{idProduct}=="0501", MODE="664", GROUP="dialout"
ATTR{idVendor}=="0764", ATTR{idProduct}=="0601", MODE="664", GROUP="dialout"
```
and the appropriate devices are created in /dev:
```
crw-rw-r--. 1 root dialout 189, 2 2023-02-19 14:11:41 /dev/bus/usb/001/003
crw-rw-r--. 1 root dialout 189, 2 2023-02-19 14:11:41 /dev/bus/usb/001/004
```
Group "dialout" looks strange, but user "nut" is indeed a member of the "dialout" group.

If I manually run usbdrvctl with either "-u root" or "-u nut", it will find the devices successfully. It's not apparent to me how to get that option passed in the course of a normal systemd startup.

noloader, 2023-02-19:
> Group "dialout" looks strange, but user "nut" is indeed a member of the "dialout" group.

dialout is for modems. I suppose dialout is Ok if the program and device are communicating with AT commands, but it would not be my first choice.

I thought a different group was used for UPS, like user ups and group ups. How was the package configured? Did it use something like `--with-user=ups --with-group=ups`?

jimklimov, 2023-02-20:
My guess is that with NUT's history with serial-port UPSes (same media as modems), many distros went with allowing NUT processes to look at all serial ports in search of a device. Then it persisted with udev and similar tools to dynamically assign permissions to filesystem nodes of certain connections.

A packaged NUT should run as `nut` user by default. There are also `RUN_AS_USER` keywords for some of the config files involved, see docs.

rknichols, 2023-02-20:
I'm not the builder, but I downloaded the .src.rpm file, and the SPEC file contains

     --with-group=dialout

     --with-user=%(name)

Looks like "dialout" was to fix <https://bugzilla.redhat.com/show_bug.cgi?id=494020>,

     "nut do not provide any mechanism to grant access to /dev/ttyS0"

On 2/19/23 17:01, Jeffrey Walton wrote:
>
>     Group "dialout" looks strange, but user "nut" is indeed a member of the "dialout" group.
>
> dialout is for modems. I suppose dialout is Ok if the program and device are communicating with AT commands, but it would not be my first choice.
>
> I thought a different group was used for UPS, like user ups and group ups. How was the package configured? Did it use something like |--with-user=ups --with-group=ups|?
>
> Message ID: ***@***.***>
>

rknichols, 2023-02-20:
The RUN_AS_USER setting is in upsmon.conf, and appears to affect only the monitor program, not the upsdrvctl program, which is the one having a problem.

jimklimov, 2023-02-20:
For drivers there's `ups.conf` with its optional `user=nut` (or root) that you can add in a device section.

rknichols, 2023-02-20:
I tried with both "user=nut" and "user=root", both in the global part of ups.conf and in the individual sections for my devices. There was no change.Message ID: ***@***.***>

rknichols, 2023-02-20:
No SELinux AVCs were reported, and I've also tried this in permissive mode. Again, no difference, and note that the error message does not mention permissions:

Message ID: ***@***.***>

> upsd[138823]: Can't connect to UPS [Cyber-2] (usbhid-ups-Cyber-2): No such file or direcory
> upsd[138823]: Can't connect to UPS [Cyber-1] (usbhid-ups-Cyber-1): No such file or direcory

Those "usbhid-ups-Cyber-[12]" devices are supposed appear as sockets in /run/nut

The only time I see SELinux AVCs is after I run "usbdrvctl start" manually (as root). That finds the UPSs successfully, but the socket that gets created in /run/nut inherits the SELinux type "unconfined_t" from the calling root shell, and a process running as nut_upsd_t would not be allowed to connect to that in enforcing mode.

jimklimov, 2023-02-20:
Wait, those latter messages from data server `upsd` say it can't talk to the driver(s). Where are messages about failing to start the drivers, e.g. `nut-driver` unit?

rknichols, 2023-02-20:
I put a wrapper around every nut executable in /usr/sbin to record its arguments and privileges. Here is the output after "systemctl start nut.target" (running in Fedora 37 with just one UPS device configured, permissive mode, "user=root" in ups.conf):
```
08:26:31 /usr/sbin/upsd -F
    Permissive uid=0(root) gid=0(root) groups=0(root) context=system_u:system_r:nut_upsd_t:s0
08:26:31 /usr/sbin/upsmon -F
    Permissive uid=0(root) gid=0(root) groups=0(root) context=system_u:system_r:nut_upsmon_t:s0
```
I presume there is something missing???

rknichols, 2023-02-20:
It doesn't appear to me that anything tries to start nut-driver.service. Here is the unit file for nut.target:Unit]
```
Description=Network UPS Tools - target for power device drivers, data server and monitoring client (if enabled) on this system
After=local-fs.target nut-driver.target nut-server.service nut-monitor.service
Wants=local-fs.target nut-driver.target nut-server.service nut-monitor.service
# network.target

[Install]
WantedBy=multi-user.target
```
and the unit file for nut-driver.target:
```
[Unit]
Description=Network UPS Tools - target for power device drivers on this system
After=local-fs.target
# network.target
PartOf=nut.target

[Install]
WantedBy=nut.target
```
I'm a "noobe" with systemd, but that second unit file doesn't appear to do anything.

jimklimov, 2023-02-21:
Targets are umbrellas for service mgmt. What does their nut-driver.service look like (or nut-driver@.service if NUT 2.8.0+)?

rknichols, 2023-02-21:
It seems I am stuck with the broken 2.8.0 packages, 2.8.0-3 in Rocky 8, and 2.8.0-8 in Fedora 37.

nut-driver.target seems to be an umbrella with nothing under it. For example, nut.target includes
`Wants=local-fs.target nut-driver.target nut-server.service nut-monitor.service`
but nut-driver.target doesn't seem to "Want" anything.

I've attached the nut-driver@.service file (with a .txt extension added so that bugzilla will accept it). It looks fine, but I can't find anything that instantiates it.
[nut-driver@.service.txt](https://github.com/networkupstools/nut/files/10797019/nut-driver%40.service.txt)

jimklimov, 2023-02-21:
For most settings describing relations between units, systemd (and SMF) allow either of them to speak up.

In case of NUT 2.8.0+, an unspecified amount of `nut-driver@UPSNAME.service` instances each say they are `WantedBy=nut-driver.target` which is more viable than listing them in a pre-packaged file :) Such instances should be generated by `nut-driver-enumerator` (script that you may run, and service calling it).

The breakage in Fedora (not sure if fixed now) involved typos in systemd-tmpfiles config, IIRC, so working directories were not made/owned well for NUT daemons.

rknichols, 2023-02-21:
Finally, it's working.

Now, If you would kindly indicate which document I should have read to tell me that I specifically need to enable "nut-driver-enumerator.service" in addition to "nut.target", I would appreciate it. The supplied "nut.target" does "Want" nut-driver.target, but as noted above, there is no further "Want" and nothing happens. The nut.target unit doesn't "Want" or "Exec" anything at all.

If I **do** run "systemctl enable nut-driver-enumerator.service", then a link in /etc/systemd/nut.target.wants is created, and the enumerator starts as expected whenever nut.target is started. Without that manual step, the enumerator simply does not get run.

None of the documents in /usr/share/doc/nut/docs suggest that anything needs to be done to make the enumerator run.

jimklimov, 2023-02-21:
Thanks for the report, probably that's a flaw on Fedora packaging side, for their bug tracker (at least the practical part, PR to improve the docs visible to user after installation to at least mention this well is welcome here) :\

Just one note, "target" units by systemd design do not Exec anything, don't have a syntactic section for that. They just glue run-time mass-enablement requests (or completed start-up milestones) together.

rknichols, 2023-02-22:
> Just one note, "target" units by systemd design do not Exec anything, don't have a syntactic section for that. They just glue run-time mass-enablement requests (or completed start-up milestones) together.

That being true, they do have "Wants", either in the unit file itself or dropped into a /etc/systemd/system/{name}.target.wants/ directory. In a new installation, nut-driver.target has neither.

I'll have to take a look at the source RPM and see if I can figure out where the problem might have originated. Thanks for your time, though.

jimklimov, 2023-02-22:
FYI: the https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests also touches on this, and was recently added into `INSTALL.nut` document.

rknichols, 2023-02-24:
https://bugzilla.redhat.com/show_bug.cgi?id=2173121 has been submitted.

jimklimov, 2023-02-24:
Actually the solution to propose was rather about docs update or making sure NDE service runs on start (enable it via package postinstall). 

I don't think world-wide packaging defaulta should enable monitoring of a particular UPS ;) More so, it is not a "netserver" by default since many hosts with NUT only run upsmon, and talk to their neighbor in control of the UPS itself (who has the drivers and data server, and may have upsmon if fed by the same UPS or watches another).

The "wants" links appear when both target/service units involved are being enabled (or disabled/enabled if a definition was updated on-disk).

rknichols, 2023-02-24:
That .patch file was just a sample setup to demonstrate the problem. Without that, the first response would be, "You must have a mistake in your configuration. Show your configuration files."

Yes, the "Wants" links are created when/if the enumerator is run, accent on the "if".

Jsalas424, 2023-04-02:
> The `tripplite_usb` apparently sees the device with meaningful data, so the problem boils down to getting the driver started in normal conditions. That would likely be about permissions - e.g. would [...]

I was actually on version 2.7.4-13 on my Proxmox OS. Adding user = root to ups.conf fixed it. Is there a "better" way to do this without giving blanket root permissions?