networkupstools/nut #1853: Data stale Powerwalker Powerwalker VI 2200 SHL on Proxmox

MarvinMynx, 2023-02-21:
PowerWalker VI 2200 SHL
OS: Proxmox - Debian GNU/Linux 11 (bullseye)

upsd -v
```
Network UPS Tools upsd 2.7.4
```

uname -a
```
Linux pve 5.15.83-1-pve #1 SMP PVE 5.15.83-1 (2022-12-15T00:00Z) x86_64 GNU/Linux
```

I have a problem with my UPS. After some time, it sort of Disconnects and as a result the Driver times out or something. 
I don't quite understand the Problem.
It's my first time debugging/configuring a USB Device / UPS.
I have many Config Files and Log Messages that hopefully describe the Problem better than I can.
And I have made some Screenshots from the Last time this happened Today. (This happens Multiple Times a Day)

First Image: DATA STALE
[image]
[image]

Second Image: Reconnect to Temporarily get Rid of the Error
[image]

lsusb | grep -i ups
```
Bus 001 Device 006: ID 06da:ffff Phoenixtec Power Co., Ltd Offline UPS
```

upsc powerwalker
```
Init SSL without certificate database
battery. Charge: 100
battery.runtime: 3290
battery.type: PbAc
device.mfr: PPC
device.model: Offline UPS HID UPS
device.serial: 000000000
device.type: ups
driver.name: usbhid-ups
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 1
driver.parameter.port: auto
driver.parameter.productid: FFFF
driver.parameter.synchronous: no
driver.parameter.vendorid: 06DA
driver.version: 2.7.4
driver.version.data: MGE HID 1.40
driver.version.internal: 0.41
input.voltage: 235.0
outlet.1.status: on
output.frequency: 49.6
output.frequency.nominal: 50
output.voltage: 235.0
output.voltage.nominal: 230
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.delay.start: 30
ups.load: 1
ups.mfr: PPC
ups.model: Offline UPS HID UPS
ups.power.nominal: 9
ups.productid: ffff
ups.serial: 000000000
ups.status: OL
ups.test.result: No test initiated
ups.timer.shutdown: 0
ups.timer.start: 0
[... 265 lines cut ...]
          /usr/sbin/upsmon -c fsd
          ;;
       upsgone)
          logger -t upssched-cmd "UPS has been gone too long, can't reach"
          ;;
       *)
          logger -t upssched-cmd "Unrecognized command: $1"
          ;;
 esac
```

I needed this udev Rule because otherwise the Server would not Boot up and dmesg would be full of USB Reconnects every 10 Seconds.

cat /etc/udev/rules.d/99-nut-ups.rule
```
SUBSYSTEM!="usb", GOTO="nut-usbups_rules_end"

# Powerwalker
#  e.g. Powerwalker VI 2200 SHL UPS - usbhid-ups
ACTION=="add|change", SUBSYSTEM=="usb|usb_device", SUBSYSTEMS=="usb|usb_device", ATTR{idVendor}=="06da", ATTR{idProduct}=="ffff", MODE="664", GROUP="nut", RUN+="/sbin/upsdrvctl stop; /sbin/upsdrvctl start"

LABEL="nut-usbups_rules_end"
```

I hope someone can help me diagnose this problem so I can use my UPS reliably.

jimklimov, 2023-02-21:
Do you have the driver normally running as a systemd unit? Possibly the "restart" via upsdrvctl causes the unit to also restart and complain due to conflict trying to use the USB port.

FWIW, NUT 2.8.0+ should have a more diligent retry logic. If you can get it built (e.g. current git master) it should hopefully behave better. See wiki for "in-place" build support.

Also *maybe* changing to "pollonly" mode with frequent ~15sec polls might help, if this device tends to go asleep and disconnect when bored, like some CPS were reported to do...

MarvinMynx, 2023-02-21:
Also, also, I pretty much left the Default Settings intact as far as possible. 
So, if someone thinks increasing the Poll Rate/Freq Retry Times or something like this would help, I'm willing to try it.
But Personally I *Think* that the problem is something else here.

MarvinMynx, 2023-02-21:
I got another Error i thin because of the systemd unit nut-driver: 
```
● nut-driver.service - Network UPS Tools - power device driver controller
     Loaded: loaded (/lib/systemd/system/nut-driver.service; static)
     Active: active (running) since Tue 2023-02-21 15:00:22 CET; 3h 32min ago
    Process: 1650101 ExecStart=/sbin/upsdrvctl start (code=exited, status=0/SUCCESS)
   Main PID: 1650241 (usbhid-ups)
      Tasks: 1 (limit: 38268)
     Memory: 496.0K
        CPU: 5.944s
     CGroup: /system.slice/nut-driver.service
             └─1650241 /lib/nut/usbhid-ups -a powerwalker

Feb 21 15:03:25 pve usbhid-ups[1650241]: libusb_get_interrupt: error submitting URB: Device or resource busy
Feb 21 15:03:25 pve usbhid-ups[1650241]: Got disconnected by another driver: Device or resource busy
Feb 21 15:03:28 pve usbhid-ups[1650241]: libusb_get_interrupt: error submitting URB: Device or resource busy
Feb 21 15:03:28 pve usbhid-ups[1650241]: Got disconnected by another driver: Device or resource busy
Feb 21 15:03:31 pve usbhid-ups[1650241]: libusb_get_interrupt: error submitting URB: Device or resource busy
Feb 21 15:03:31 pve usbhid-ups[1650241]: Got disconnected by another driver: Device or resource busy
Feb 21 15:03:34 pve usbhid-ups[1650241]: libusb_get_interrupt: error submitting URB: Device or resource busy
Feb 21 15:03:34 pve usbhid-ups[1650241]: Got disconnected by another driver: Device or resource busy
Feb 21 15:03:37 pve usbhid-ups[1650241]: libusb_get_interrupt: error submitting URB: Device or resource busy
Feb 21 15:03:37 pve usbhid-ups[1650241]: Got disconnected by another driver: Device or resource busy
```
```
● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-02-21 14:59:50 CET; 3h 32min ago
    Process: 1649116 ExecStart=/sbin/upsd (code=exited, status=0/SUCCESS)
   Main PID: 1649117 (upsd)
      Tasks: 1 (limit: 38268)
     Memory: 896.0K
        CPU: 318ms
     CGroup: /system.slice/nut-server.service
             └─1649117 /lib/nut/upsd

Feb 21 17:46:14 pve upsd[1649117]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): Connection refused
Feb 21 17:51:14 pve upsd[1649117]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): Connection refused
Feb 21 17:56:14 pve upsd[1649117]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): Connection refused
Feb 21 18:01:14 pve upsd[1649117]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): Connection refused
Feb 21 18:06:14 pve upsd[1649117]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): Connection refused
Feb 21 18:11:14 pve upsd[1649117]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): Connection refused
Feb 21 18:16:14 pve upsd[1649117]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): Connection refused
Feb 21 18:21:14 pve upsd[1649117]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): Connection refused
Feb 21 18:26:14 pve upsd[1649117]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): Connection refused
Feb 21 18:31:14 pve upsd[1649117]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): Connection refused
```
```
● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-02-21 14:59:50 CET; 3h 32min ago
    Process: 1649132 ExecStart=/sbin/upsmon (code=exited, status=0/SUCCESS)
   Main PID: 1649134 (upsmon)
      Tasks: 2 (limit: 38268)
     Memory: 916.0K
        CPU: 366ms
     CGroup: /system.slice/nut-monitor.service
             ├─1649133 /lib/nut/upsmon
             └─1649134 /lib/nut/upsmon

Feb 21 18:31:55 pve upsmon[1649134]: Poll UPS [powerwalker@localhost] failed - Driver not connected
Feb 21 18:32:00 pve upsmon[1649134]: Poll UPS [powerwalker@localhost] failed - Driver not connected
Feb 21 18:32:05 pve upsmon[1649134]: Poll UPS [powerwalker@localhost] failed - Driver not connected
Feb 21 18:32:10 pve upsmon[1649134]: Poll UPS [powerwalker@localhost] failed - Driver not connected
Feb 21 18:32:15 pve upsmon[1649134]: Poll UPS [powerwalker@localhost] failed - Driver not connected
Feb 21 18:32:20 pve upsmon[1649134]: Poll UPS [powerwalker@localhost] failed - Driver not connected
Feb 21 18:32:25 pve upsmon[1649134]: Poll UPS [powerwalker@localhost] failed - Driver not connected
Feb 21 18:32:30 pve upsmon[1649134]: Poll UPS [powerwalker@localhost] failed - Driver not connected
Feb 21 18:32:35 pve upsmon[1649134]: Poll UPS [powerwalker@localhost] failed - Driver not connected
Feb 21 18:32:40 pve upsmon[1649134]: Poll UPS [powerwalker@localhost] failed - Driver not connected
[... 56 lines cut ...]
```
● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-02-21 18:36:26 CET; 22s ago
    Process: 2054882 ExecStart=/sbin/upsmon (code=exited, status=0/SUCCESS)
   Main PID: 2054884 (upsmon)
      Tasks: 3 (limit: 38268)
     Memory: 1.1M
        CPU: 11ms
     CGroup: /system. Slice/nut-monitor.service
             ├─2054883 /lib/nut/upsmon
             ├─2054884 /lib/nut/upsmon
[... cut ...]
Feb 21 18:36:26 pve systemd[1]: Started Network UPS Tools - power device monitor and shutdown controller.
Feb 21 18:36:31 pve upsmon[2054884]: Poll UPS [powerwalker@localhost] failed - Driver not connected
Feb 21 18:36:31 pve upsmon[2054884]: Communications with UPS powerwalker@localhost lost
Feb 21 18:36:31 pve upssched[2055021]: Timer daemon started
Feb 21 18:36:31 pve upssched[2055021]: New timer: commbad (30 seconds)
Feb 21 18:36:36 pve upsmon[2054884]: Communications with UPS powerwalker@localhost established
Feb 21 18:36:36 pve upssched[2055021]: Cancelling timer: commbad
```

MarvinMynx, 2023-02-21:
For better Debugging Purposes i now first Stopped the Driver with
```
upsdrvctl stop
```
And then Started the Driver like this:
```
/lib/nut/usbhid-ups -DD -a powerwalker
```
And i got this:
```
Network UPS Tools - Generic HID driver 0.41 (2.7.4)
USB communication driver 0.33
   0.000000     debug level is '2'
   0.000546     upsdrv_initups...
   0.165144     Checking device (1D6B/0002) (003/001)
   0.165270     - VendorID: 1d6b
   0.165277     - ProductID: 0002
   0.165280     - Manufacturer: unknown
   0.165284     - Product: unknown
   0.165295     - Serial Number: unknown
   0.165300     - Bus: 003
   0.165306     - Device release number: 0515
   0.165311     Trying to match device
   0.165319     Device does not match - skipping
   0.180750     Checking device (0BC2/AB34) (002/002)
   0.180773     - VendorID: 0bc2
   0.180778     - ProductID: ab34
   0.180784     - Manufacturer: unknown
   0.180789     - Product: unknown
   0.180794     - Serial Number: unknown
   0.180799     - Bus: 002
   0.180806     - Device release number: 0100
   0.180810     Trying to match device
   0.180817     Device does not match - skipping
   0.180822     Checking device (1D6B/0003) (002/001)
   0.180836     - VendorID: 1d6b
   0.180839     - ProductID: 0003
   0.180844     - Manufacturer: unknown
   0.180849     - Product: unknown
   0.180852     - Serial Number: unknown
   0.180856     - Bus: 002
   0.180859     - Device release number: 0515
   0.180863     Trying to match device
   0.180867     Device does not match - skipping
   0.180873     Checking device (06DA/FFFF) (001/006)
   0.185787     - VendorID: 06da
   0.185793     - ProductID: ffff
   0.185796     - Manufacturer: PPC
   0.185798     - Product: Offline UPS
   0.185801     - Serial Number: 000000000
   0.185803     - Bus: 001
   0.185806     - Device release number: 0003
   0.185808     Trying to match device
   0.185827     Device matches
   0.186575     HID descriptor length 495
   0.204729     Report Descriptor size = 495
   0.204805     Using subdriver: MGE HID 1.40
   0.204808     45 HID objects found
   0.205333     Path: UPS.BatterySystem.Battery.Test, Type: Feature, ReportID: 0x24, Offset: 0, Size: 8, Value: 6
   0.205874     Path: UPS.Flow.[4].ConfigApparentPower, Type: Feature, ReportID: 0x74, Offset: 0, Size: 16, Value: 9
   0.206403     Path: UPS.Flow.[4].ConfigFrequency, Type: Feature, ReportID: 0x36, Offset: 0, Size: 8, Value: 50
   0.206924     Path: UPS.Flow.[4].ConfigVoltage, Type: Feature, ReportID: 0x72, Offset: 0, Size: 8, Value: 230
   0.207452     Path: UPS.PowerConverter.ConverterType, Type: Feature, ReportID: 0x4a, Offset: 0, Size: 8, Value: 1
   0.208093     Path: UPS.PowerConverter.Input.[1].Voltage, Type: Feature, ReportID: 0x31, Offset: 0, Size: 16, Value: 231
   0.208630     Path: UPS.PowerConverter.Input.[1].PresentStatus.Boost, Type: Feature, ReportID: 0x2d, Offset: 0, Size: 8, Value: 0
   0.208639     Path: UPS.PowerConverter.Input.[1].PresentStatus.Boost, Type: Input, ReportID: 0x2d, Offset: 0, Size: 8, Value: 0
   0.208644     Path: UPS.PowerConverter.Input.[1].PresentStatus.Buck, Type: Feature, ReportID: 0x2d, Offset: 8, Size: 8, Value: 0
   0.208649     Path: UPS.PowerConverter.Input.[1].PresentStatus.Buck, Type: Input, ReportID: 0x2d, Offset: 8, Size: 8, Value: 0
   0.209104     Path: UPS.PowerConverter.Output.Frequency, Type: Feature, ReportID: 0x42, Offset: 0, Size: 16, Value: 49.6
   0.209111     Path: UPS.PowerConverter.Output.Voltage, Type: Feature, ReportID: 0x42, Offset: 16, Size: 16, Value: 231
[... 338 lines cut ...]
  15.570825     Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
  15.570834     Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 16, Size: 8, Value: 1
  15.570841     Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
  16.319942     upsdrv_updateinfo...
  16.571011     libusb_get_interrupt: Connection timed out
  16.571024     Got 0 HID objects...
  16.571028     Quick update...
  16.571518     Path: UPS.PowerSummary.DelayBeforeStartup, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 32, Value: 0
  16.572119     Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x09, Offset: 0, Size: 32, Value: 0
  16.572592     Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
  16.572603     Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
  16.572619     Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 16, Size: 8, Value: 1
  16.572626     Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
  17.320729     upsdrv_updateinfo...
  17.571187     libusb_get_interrupt: Connection timed out
  17.571200     Got 0 HID objects...
  17.571204     Quick update...
  17.571746     Path: UPS.PowerSummary.DelayBeforeStartup, Type: Feature, ReportID: 0x0a, Offset: 0, Size: 32, Value: 0
  17.572221     Path: UPS.PowerSummary.DelayBeforeShutdown, Type: Feature, ReportID: 0x09, Offset: 0, Size: 32, Value: 0
  17.572732     Path: UPS.PowerSummary.PresentStatus.ACPresent, Type: Feature, ReportID: 0x01, Offset: 0, Size: 8, Value: 1
  17.572741     Path: UPS.PowerSummary.PresentStatus.Discharging, Type: Feature, ReportID: 0x01, Offset: 24, Size: 8, Value: 0
  17.572746     Path: UPS.PowerSummary.PresentStatus.Charging, Type: Feature, ReportID: 0x01, Offset: 16, Size: 8, Value: 1
  17.572750     Path: UPS.PowerSummary.PresentStatus.BelowRemainingCapacityLimit, Type: Feature, ReportID: 0x01, Offset: 8, Size: 8, Value: 0
^C
```

MarvinMynx, 2023-02-21:
And now I have DATA STALE again, it's that often a Day...
systemctl status nut-driver.service; systemctl status nut-server.service; systemctl status nut-client.service
```
● nut-driver.service - Network UPS Tools - power device driver controller
     Loaded: loaded (/lib/systemd/system/nut-driver.service; static)
     Active: failed (Result: exit-code) since Tue 2023-02-21 19:15:12 CET; 1h 5min ago
    Process: 2124171 ExecStart=/sbin/upsdrvctl start (code=exited, status=0/SUCCESS)
    Process: 2124197 ExecStop=/sbin/upsdrvctl stop (code=exited, status=1/FAILURE)
   Main PID: 2124175 (code=exited, status=0/SUCCESS)
        CPU: 9ms

Feb 21 19:15:12 pve upsdrvctl[2124173]: Network UPS Tools - Generic HID driver 0.41 (2.7.4)
Feb 21 19:15:12 pve upsdrvctl[2124173]: USB communication driver 0.33
Feb 21 19:15:12 pve upsdrvctl[2124171]: Network UPS Tools - UPS driver controller 2.7.4
Feb 21 19:15:12 pve usbhid-ups[2124175]: Startup successful
Feb 21 19:15:12 pve systemd[1]: Started Network UPS Tools - power device driver controller.
Feb 21 19:15:12 pve usbhid-ups[2124175]: Signal 15: exiting
Feb 21 19:15:12 pve upsdrvctl[2124197]: Can't open /run/nut/usbhid-ups-auto.pid: No such file or directory
Feb 21 19:15:12 pve upsdrvctl[2124197]: Network UPS Tools - UPS driver controller 2.7.4
Feb 21 19:15:12 pve systemd[1]: nut-driver.service: Control process exited, code=exited, status=1/FAILURE
Feb 21 19:15:12 pve systemd[1]: nut-driver.service: Failed with result 'exit-code'.
● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-02-21 19:15:12 CET; 1h 5min ago
    Process: 2124176 ExecStart=/sbin/upsd (code=exited, status=0/SUCCESS)
   Main PID: 2124177 (upsd)
      Tasks: 1 (limit: 38268)
     Memory: 700.0K
        CPU: 79ms
     CGroup: /system.slice/nut-server.service
             └─2124177 /lib/nut/upsd

Feb 21 19:15:12 pve upsd[2124176]: fopen /run/nut/upsd.pid: No such file or directory
Feb 21 19:15:12 pve upsd[2124176]: listening on 0.0.0.0 port 3493
Feb 21 19:15:12 pve upsd[2124176]: listening on 0.0.0.0 port 3493
Feb 21 19:15:12 pve upsd[2124176]: Connected to UPS [powerwalker]: usbhid-ups-powerwalker
Feb 21 19:15:12 pve upsd[2124176]: Connected to UPS [powerwalker]: usbhid-ups-powerwalker
Feb 21 19:15:12 pve upsd[2124177]: Startup successful
Feb 21 19:15:12 pve systemd[1]: Started Network UPS Tools - power devices information server.
Feb 21 19:15:12 pve upsd[2124177]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): No such file or directory
Feb 21 19:15:19 pve upsd[2124177]: Connected to UPS [powerwalker]: usbhid-ups-powerwalker
Feb 21 20:18:12 pve upsd[2124177]: Data for UPS [powerwalker] is stale - check driver
● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-02-21 19:15:12 CET; 1h 5min ago
    Process: 2124189 ExecStart=/sbin/upsmon (code=exited, status=0/SUCCESS)
   Main PID: 2124191 (upsmon)
      Tasks: 2 (limit: 38268)
     Memory: 912.0K
        CPU: 72ms
     CGroup: /system.slice/nut-monitor.service
             ├─2124190 /lib/nut/upsmon
             └─2124191 /lib/nut/upsmon

Feb 21 20:19:32 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:19:37 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:19:42 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:19:47 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:19:52 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:19:57 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:20:02 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:20:07 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:20:12 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:20:17 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
```

upscmd -l powerwalker     
```                                                                                                                                                                                
Error: Data stale
```
upsc powerwalker                                                                                                                                                                                          
```
Init SSL without certificate database
Error: Data stale
```

MarvinMynx, 2023-02-21:
I left the SSH Session open this time and got further confused with this Messages:
```
Broadcast message from root@pve (somewhere) (Tue Feb 21 20:28:17 2023):

UPS powerwalker@localhost is unavailable

Broadcast message from root@pve (somewhere) (Tue Feb 21 20:33:22 2023):

UPS powerwalker@localhost is unavailable

Broadcast message from root@pve (somewhere) (Tue Feb 21 20:38:27 2023):

UPS powerwalker@localhost is unavailable

Broadcast message from root@pve (somewhere) (Tue Feb 21 20:43:32 2023):

UPS powerwalker@localhost is unavailable

Broadcast message from root@pve (somewhere) (Tue Feb 21 20:48:37 2023):

UPS powerwalker@localhost is unavailable

Broadcast message from root@pve (somewhere) (Tue Feb 21 20:48:57 2023):

Communications with UPS powerwalker@localhost established

Broadcast message from root@pve (somewhere) (Tue Feb 21 20:49:12 2023):

Communications with UPS powerwalker@localhost lost
```
systemctl status nut-driver.service; systemctl status nut-server.service; systemctl status nut-client.service
```
● nut-driver.service - Network UPS Tools - power device driver controller
     Loaded: loaded (/lib/systemd/system/nut-driver.service; static)
     Active: failed (Result: exit-code) since Tue 2023-02-21 19:15:12 CET; 1h 34min ago
    Process: 2124171 ExecStart=/sbin/upsdrvctl start (code=exited, status=0/SUCCESS)
    Process: 2124197 ExecStop=/sbin/upsdrvctl stop (code=exited, status=1/FAILURE)
   Main PID: 2124175 (code=exited, status=0/SUCCESS)
        CPU: 9ms

Feb 21 19:15:12 pve upsdrvctl[2124173]: Network UPS Tools - Generic HID driver 0.41 (2.7.4)
Feb 21 19:15:12 pve upsdrvctl[2124173]: USB communication driver 0.33
Feb 21 19:15:12 pve upsdrvctl[2124171]: Network UPS Tools - UPS driver controller 2.7.4
Feb 21 19:15:12 pve usbhid-ups[2124175]: Startup successful
Feb 21 19:15:12 pve systemd[1]: Started Network UPS Tools - power device driver controller.
Feb 21 19:15:12 pve usbhid-ups[2124175]: Signal 15: exiting
Feb 21 19:15:12 pve upsdrvctl[2124197]: Can't open /run/nut/usbhid-ups-auto.pid: No such file or directory
Feb 21 19:15:12 pve upsdrvctl[2124197]: Network UPS Tools - UPS driver controller 2.7.4
Feb 21 19:15:12 pve systemd[1]: nut-driver.service: Control process exited, code=exited, status=1/FAILURE
Feb 21 19:15:12 pve systemd[1]: nut-driver.service: Failed with result 'exit-code'.
● nut-server.service - Network UPS Tools - power devices information server
     Loaded: loaded (/lib/systemd/system/nut-server.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-02-21 19:15:12 CET; 1h 34min ago
    Process: 2124176 ExecStart=/sbin/upsd (code=exited, status=0/SUCCESS)
   Main PID: 2124177 (upsd)
      Tasks: 1 (limit: 38268)
     Memory: 904.0K
        CPU: 106ms
     CGroup: /system.slice/nut-server.service
             └─2124177 /lib/nut/upsd

Feb 21 19:15:12 pve upsd[2124176]: Connected to UPS [powerwalker]: usbhid-ups-powerwalker
Feb 21 19:15:12 pve upsd[2124176]: Connected to UPS [powerwalker]: usbhid-ups-powerwalker
Feb 21 19:15:12 pve upsd[2124177]: Startup successful
Feb 21 19:15:12 pve systemd[1]: Started Network UPS Tools - power devices information server.
Feb 21 19:15:12 pve upsd[2124177]: Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): No such file or directory
Feb 21 19:15:19 pve upsd[2124177]: Connected to UPS [powerwalker]: usbhid-ups-powerwalker
Feb 21 20:18:12 pve upsd[2124177]: Data for UPS [powerwalker] is stale - check driver
Feb 21 20:48:52 pve upsd[2124177]: Send ping to UPS [powerwalker] failed: Resource temporarily unavailable
Feb 21 20:48:54 pve upsd[2124177]: Connected to UPS [powerwalker]: usbhid-ups-powerwalker
Feb 21 20:49:11 pve upsd[2124177]: Data for UPS [powerwalker] is stale - check driver
● nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; vendor preset: enabled)
     Active: active (running) since Tue 2023-02-21 19:15:12 CET; 1h 34min ago
    Process: 2124189 ExecStart=/sbin/upsmon (code=exited, status=0/SUCCESS)
   Main PID: 2124191 (upsmon)
      Tasks: 3 (limit: 38268)
     Memory: 1.1M
        CPU: 143ms
     CGroup: /system.slice/nut-monitor.service
             ├─2124190 /lib/nut/upsmon
             ├─2124191 /lib/nut/upsmon
             └─2293927 /usr/sbin/upssched Communications with UPS powerwalker@localhost lost

Feb 21 20:49:13 pve upssched[2293927]: New timer: commbad (30 seconds)
Feb 21 20:49:17 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:49:22 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:49:27 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:49:32 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:49:37 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:49:42 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:49:43 pve upssched[2293927]: Event: commbad
Feb 21 20:49:47 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
Feb 21 20:49:52 pve upsmon[2124191]: Poll UPS [powerwalker@localhost] failed - Data stale
```

jimklimov, 2023-02-21:
> can't build

That's odd - Proxmox being based on Debian. In fact got one running as NUT testbed and part of CI farm. You may have to install tools and other prerequisites, see `docs/config-prereqs.txt`

jimklimov, 2023-02-21:
The driver log literally says `Got disconnected by another driver: Device or resource busy` - do you need an even more smoking gun? ;)

If your drivers are managed by systemd, do not intrude with other methods, same as for any other services.

jimklimov, 2023-02-21:
Note: NUT 2.7.4 had all drivers under a single `nut-driver.service` (so all drvs were restarted by the unit if you had several devices to watch). 2.8.0+ has it granular, with each `ups.conf` section wrapped by a `nut-driver@UPSNAME.service` instance (courtesy of `nut-driver-enumerator`script and service).

MarvinMynx, 2023-02-21:
> The driver log literally says `Got disconnected by another driver: Device or resource busy` - do you need an even more smoking gun? ;)
> 
> If your drivers are managed by systemd, do not intrude with other methods, same as for any other services.

Yeah but which other Driver and why? I don't get it? Because it does this even if I don't intrude at all.

> Note: NUT 2.7.4 had all drivers under a single `nut-driver.service` (so all drvs were restarted by the unit if you had several devices to watch). 2.8.0+ has it granular, with each `ups.conf` section [...]

Okay, but i have only one UPS Device connected.

> > can't build
> 
> That's odd - Proxmox being based on Debian. In fact got one running as NUT testbed and part of CI farm. You may have to install tools and other prerequisites, see `docs/config-prereqs.txt`

I have read the Document to the Point of:
```
mkdir -p nut && cd nut && \
    git clone https://github.com/networkupstools/nut/ -b master .
./autogen.sh && \
    ./configure --with-doc=all --with-all --with-cgi && \
    make all && make check && make spellcheck
```

Can you please help me with how to do this? 
I never build something at all for Linux. 
Which Packages do I need to Install for example.

One the Above Command I get the following Error Message:
```
./autogen.sh                                                                                                    
Regenerating Augeas ups.conf lens with 'python3'...
Regenerating the USB helper files...
Calling autoreconf...
./autogen.sh: 127: autoreconf: not found
FAILED: did not generate an executable configure script!

# Note: on some systems "autoreconf", "automake" et al are dispatcher
# scripts, and need you to explicitly say which version you want, e.g.
#    export AUTOCONF_VERSION=2.65 AUTOMAKE_VERSION=1.13
# If you get issues with AC_DISABLE_STATIC make sure you have libtool.
```
What exactly does export AUTOCONF_VERSION=2.65 do, and why do I need to do this?

MarvinMynx, 2023-02-21:
> The driver log literally says `Got disconnected by another driver: Device or resource busy` - do you need an even more smoking gun? ;)
> 
> If your drivers are managed by systemd, do not intrude with other methods, same as for any other services.

Yeah but which other Driver and why? I don't get it? Because it does this even if I don't intrude at all.

> Note: NUT 2.7.4 had all drivers under a single `nut-driver.service` (so all drvs were restarted by the unit if you had several devices to watch). 2.8.0+ has it granular, with each `ups.conf` section [...]

Okay, but i have only one UPS Device connected.

> > can't build
> 
> That's odd - Proxmox being based on Debian. In fact got one running as NUT testbed and part of CI farm. You may have to install tools and other prerequisites, see `docs/config-prereqs.txt`

I have read the Document to the Point of:
```
mkdir -p nut && cd nut && \
    git clone https://github.com/networkupstools/nut/ -b master .
./autogen.sh && \
    ./configure --with-doc=all --with-all --with-cgi && \
    make all && make check && make spellcheck
```

Can you please help me with how to do this? 
I never build something at all for Linux. 
Which Packages do I need to Install for example.

One the Above Command I get the following Error Message:
```
./autogen.sh                                                                                                    
Regenerating Augeas ups.conf lens with 'python3'...
Regenerating the USB helper files...
Calling autoreconf...
./autogen.sh: 127: autoreconf: not found
FAILED: did not generate an executable configure script!

# Note: on some systems "autoreconf", "automake" et al are dispatcher
# scripts, and need you to explicitly say which version you want, e.g.
#    export AUTOCONF_VERSION=2.65 AUTOMAKE_VERSION=1.13
# If you get issues with AC_DISABLE_STATIC make sure you have libtool.
```
What exactly does export AUTOCONF_VERSION=2.65 do, and why do i need to do this?

jimklimov, 2023-02-21:
Try reading more, and also the layman text around command blocks :) That document goes to greater detail listing packages to install for different OSes (notes from CI farm which looks at a lot of those).

> Which other driver

As I wrote earlier, you seem to have a conflict of driver daemon started by systemd (per nut-driver.service, monolithic earlier or instantiated later), and separately one you launch or kill by direct calls to upsdrvctl. These are two separate daemons trying to get exclusive acccess to same USB port. When you do `upsdrvctl stop`, systemd restarts the deceased service because it must. Then you do `upsdrvctl start`, and the two driver processes elbow each other away.

Try to `systemctl restart nut-driver` in your udev rules, if that helps easier with your older NUT installation.

jimklimov, 2023-02-21:
> What exactly does export AUTOCONF_VERSION=2.65 do, and why do I need to do this?

It sets an environment variable in your shell, exported to be inherited by its later children (such as the build programs and scripts).

On some systems `autoSOMETHING` filenames are dispatching wrappers that call a specific version, as specified by such envvar.

MarvinMynx, 2023-02-21:
> Try reading more, and also the layman text around command blocks :) That document goes to greater detail listing packages to install for different OSes (notes from CI farm which looks at a lot of [...]

But as far as i can see that's not what is happening. 
This it is the First NUT Installation. And I only started to use systemctl start/stop/ect. nut-driver after I discovered it while I was trying to Debug. Before I noticed that I had a Problem, I only ever used this:
```
sudo service nut-server restart
sudo service nut-client restart
sudo systemctl restart nut-monitor
sudo upsdrvctl stop
sudo upsdrvctl start
```
To Start or Restart NUT. That's why I stopped and disabled the nut-driver systemd unit again and reverted to using upsdrvctl stop/start exclusively again today.
But I don't know which way is better or intended and why one would be better than the other.

> > What exactly does export AUTOCONF_VERSION=2.65 do, and why do I need to do this?
> 
> It sets an environment variable in your shell, exported to be inherited by its later children (such as the build programs and scripts).
> 
> On some systems (e.g. OpenBSD) `autoSOMETHING` filenames are dispatching wrappers that call a specific version, as specified by such envvar. Haven't seen that on mainstream Linux, but there are many distros out there...
> 
> Somehow people are expected to be aware if not fluent in systems they use :) No fault at that, never late to start :)

Well yeah ^^" I'm honest up until now I never needed to build something on Linux. I got along pretty well at work and at home without building something ^^" And it always seemed so complex.
Don't I need to install the make Package for example? and other build tools? Don't I need src Repos to build something?
Or will mindlessly exporting the aforementioned Variable work out?

MarvinMynx, 2023-02-21:
I've read the Doc file further along, ignoring that i can't Reproduce the aforementioned Command, and got to this:
```
apt-get install \
    ccache time \
    git python perl curl \
    make autoconf automake libltdl-dev libtool-bin libtool \
    valgrind \
    cppcheck \
    pkg-config \
    gcc g++ clang
```
This will Install A LOT OF STUFF. Is this actually needed / Recommended? 
Should this be installed on a Production Like system of my own?

jimklimov, 2023-02-21:
Not too much of addition really (deps for CGI and docs would have a big footprint though, so listed separately; you might also choose just one of clang or gcc/g++), and should not be a problem for a production system unless you're in some regulated environment or with a very small rootfs. 

With proxmox, technically you can easily make a container with same Debian version and all the tools, pick the `configure` parameters from distro source package (debian/rules file), build NUT there and `make install DESTDIR=/tmp/nut`, tarball the result from `/tmp/nut` and unpack it on your main OS.

Not sure I want to recommend that with your current proclaimed level of Linux proficiency, but as far as "doable" - yes, it certainly is :) And sorry, but I can't tap a crash course on systems administration and packaging here in finite time ;}

jimklimov, 2023-02-21:
As for service start/stop, I'd expe the original distro's NUT packages that you installed to have registered, enabled and started the systemd services. But maybe I'm thinking too well of them.

MarvinMynx, 2023-02-21:
> Not too much of addition really (deps for CGI and docs would have a big footprint though, so listed separately; you might also choose just one of clang or gcc/g++), and should not be a problem for a [...]

No Problem I very much thank you for all the Replays so far. And I will get an RHEL Course later this Year from Work so that hopefully will help a lot too :)

I will Install these Packages on the main System without the LXC Container because yeah ^^" as you mentioned, it is easier ^^"
But I will do so, tomorrow Because it near midnight here in Germany ^^"

> As for service start/stop, I'd expect the original distro's NUT packages that you installed to have registered, enabled and started the systemd services. But maybe I'm thinking too well of them.

I think you are because nothing was started or enabled after just installing the Packages. 
I think the systemd Services were available though. I'm not sure if I checked ^^"
It's also possible that I just used the wrong Command to start up NUT the whole time ^^"
if I'm honest I must admit that I just followed a guide on how to set up NUT. 
That Guide is 1 Year old though maybe this has changed in that time and that's why I'm so confused.

Again thank you very much for all of your replays today :)

jimklimov, 2023-02-22:
FYI: the https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests also touches on this, and was recently added into `INSTALL.nut` document.

MarvinMynx, 2023-02-22:
> As for service start/stop, I'd expect the original distro's NUT packages that you installed to have registered, enabled and started the systemd services. But maybe I'm thinking too well of them.

Okay I now switched from:
```
sudo service nut-server restart
sudo service nut-client restart
sudo systemctl restart nut-monitor
sudo upsdrvctl stop
sudo upsdrvctl start
```
to:
```
systemctl restart nut-server; systemctl restart nut-client; systemctl restart nut-monitor; systemctl restart nut-driver.service
```
And changed the udev Rule accordingly to systemctl restart nut-driver.service.

But it doesn't Change anythin. After an Hour or so, i still get DATA STALE Again.

so, I don't know what's up with systemd and stuff, I still don't get it. 

Will try to Build but after what I've seen p until now, I think this won't resolve my issues. Kind of losing my mind with this.

MarvinMynx, 2023-02-22:
Would I need to Reboot to Apply changes to the udev Rule properly? that would maybe explain my Last comment.

jimklimov, 2023-02-22:
In the years leading up to 2.8.0 release (and after it too) there was a lot of effort into improving debug-ability like this, including a `DEBUG_MIN` setting in various daemons' config files to avoid messing with init scripts or units to trace how they go about their life in production mode (avoiding "works for me" effect with a root shell vs. service context, etc.)

Understandably, the 2.7.4 once released (and packaged by distros) is set in stone and there's so much we can do to easily debug it. That said, it is possible to e.g. add `-D` flags for daemons into respective `*.service` files for higher debug verbosity -- which would make the daemons foregrounded so you may have to fiddle with their service `Type` (simple vs. forking) as well, if that would help discover more hints.

The "ERR ACCESS-DENIED" suggests some problem logging in from upsmon client to upsd server - check `MONITOR` lines with passwords in `upsmon.conf`, and user definitions in `upsd.users`, and that these files are only(!) readable by NUT run-time user/group. I think daemons may refuse to use these files if their protection is too lax.

For udev, a reboot should help, but it may suffice to restart its service (may have to track down the name in your distro, see suggestions in the wiki article linked above).

MarvinMynx, 2023-02-22:
> The "ERR ACCESS-DENIED" suggests some problem logging in from upsmon client to upsd server - check `MONITOR` lines with passwords in `upsmon.conf`, and user definitions in `upsd.users`, and that [...]

Okay, The Permissions are correct per Default and the Password in Both Files does definitely Match. 
I checked this before a thousand times :(
```
ll /etc/nut/upsmon.conf
.rw-r----- root nut 1.1 KB Tue Feb 21 14:56:38 2023  /etc/nut/upsmon.conf

ll /etc/nut/upsd.users
.rw-r----- root nut 48 B Tue Feb 21 14:53:54 2023  /etc/nut/upsd.users

cat /etc/nut/upsmon.conf | grep "Password"
MONITOR powerwalker@localhost 1 admin Password master

cat /etc/nut/upsd.users | grep "Password"
  password = Password
```

MarvinMynx, 2023-02-22:
> mkdir -p nut && cd nut && \
>     git clone https://github.com/networkupstools/nut/ -b master .
> ./autogen.sh && \
>     ./configure --with-doc=all --with-all --with-cgi && \
>     make all && make check && make spellcheck

Since nothing else is working or does make sense to me, I now installed all the Dependencies according to the Docs:
```
apt-get install ccache time git python perl curl make autoconf automake libltdl-dev libtool-bin libtool valgrind cppcheck pkg-config gcc g++ clang python3 python3.9 aspell-en asciidoc source-highlight python3-pygments dblatex libgd-dev libsystemd-dev libcppunit-dev libssl-dev libnss3-dev augeas-tools libaugeas-dev augeas-lenses libusb-dev libusb-1.0-0-dev libi2c-dev libmodbus-dev libsn
```
And then ran:
````
./autogen.sh
```
And:
````
./configure --with-doc=all --with-all --with-cgi
```
Got an Error on the Last command, STIL Missing Dependencies?
```
configure: error: neon libraries not found, required for neon based XML/HTTP driver
```

MarvinMynx, 2023-02-22:
Installed Additional Packages, where I thought I didn't need them while reading the DOC.
```
apt-get install lua5.1-dev bash dash ksh busybox libneon27-dev
```
And then did ./autogen.sh and ./configure --with-doc=all --with-all --with-cgi again.
It seems like it ran through this time.

MarvinMynx, 2023-02-22:
I now ran
```
make all && make check && make spellcheck
```
and it seems like it has been completed successfully.
I should now have a/Multiple local .deb file(s) somewhere that I need to install via apt, correct?
Where would I find that .deb file(s)?

MarvinMynx, 2023-02-22:
Where the configure and make commands even correct?
Or would I need one of the following?
Or is both Possible? (Pros / Cons / Whats Recommanded?)

```
:; cd /tmp
:; git clone https://github.com/networkupstools/nut
:; cd nut
:; ./autogen.sh
:; ./configure --enable-inplace-runtime # --maybe-some-other-options
:; make -j 4 all check && sudo make install
```
```
...
:; make -j 4 all check && \
   { sudo systemctl stop nut-monitor nut-server || true ; } && \
   { sudo systemctl stop nut-driver.service || true ; } && \
   { sudo systemctl stop nut-driver.target || true ; } && \
   { sudo systemctl stop nut.target || true ; } && \
   sudo make install && \
   sudo systemctl daemon-reload && \
   sudo systemd-tmpfiles --create && \
   sudo systemctl disable nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   sudo systemctl enable  nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   { sudo systemctl restart udev || true ; } && \
   sudo systemctl restart nut-driver-enumerator.service nut-monitor nut-server
```

jimklimov, 2023-02-22:
Sort of both. The first block (except last line with `make install`) is the `...` in the second block :)

The first part is about building and installing (in-place overwrite, no packaging involved) a new NUT version on more or less any OS.
The second part focuses on Linux with systemd. I believe Wiki article text and section titles imply that ;)

jimklimov, 2023-02-22:
NUT does not currently include packaging recipes (or very few - e.g. a reference one for Solaris/illumos) since it is distros' concern about where files are laid out and how users are named, etc. Can't make one recipe here that would satisfy everyone in the field. 

There is a long-standing low-priority issue to trawl deb, spec, ... recipes in popular distros and place them into NUT sources as examples/references usable for a `make package` action, sort of "better than nothing", but it was not much pursued yet.

MarvinMynx, 2023-02-22:
> Sort of both. The first block (except last line with `make install`) is the `...` in the second block :)
>
> The first part is about building and installing (in-place overwrite, no packaging involved) [...]

So as soon as systemctl is working/present on a system and can start some systemd services, I have i Linux system with systemd, isn't that correct?

Okay, I was unclear earlier. Which of the 3 Should I Use? 
```
mkdir -p nut && cd nut && \
    git clone https://github.com/networkupstools/nut/ -b master .
./autogen.sh && \
    ./configure --with-doc=all --with-all --with-cgi && \
    make all && make check && make spellcheck
```
```
:; cd /tmp
:; git clone https://github.com/networkupstools/nut
:; cd nut
:; ./autogen.sh
:; ./configure --enable-inplace-runtime # --maybe-some-other-options
:; make -j 4 all check && sudo make install
```
```
:; cd /tmp
:; git clone https://github.com/networkupstools/nut
:; cd nut
:; ./autogen.sh
:; ./configure --enable-inplace-runtime # --maybe-some-other-options
:; make -j 4 all check && \
   { sudo systemctl stop nut-monitor nut-server || true ; } && \
   { sudo systemctl stop nut-driver.service || true ; } && \
   { sudo systemctl stop nut-driver.target || true ; } && \
   { sudo systemctl stop nut.target || true ; } && \
   sudo make install && \
   sudo systemctl daemon-reload && \
   sudo systemd-tmpfiles --create && \
   sudo systemctl disable nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   sudo systemctl enable  nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   { sudo systemctl restart udev || true ; } && \
   sudo systemctl restart nut-driver-enumerator.service nut-monitor nut-server
```

was the First one correct?
Or would I need one of the following?
After your last Replays, I would tend to use the last one, just because systemd is available on Proxmox.

MarvinMynx, 2023-02-22:
Okay, I restarted the udev Service, I didn't do this before.
Udev rule is still systemctrl, and Rebooted to be extra sure.
It seems like the problem is resolved, but I still get wired errors on some Services 
But everything is working probably.
These lines still really confuse me: (all on nut-monitor.service)
```
nut-monitor.service: Can't open PID file /run/nut/upsmon.pid (yet?) after start: Operation not permitted
nut-monitor.service: Supervising process 73186 which is not our child. We'll most likely not notice when it exists.
Login on UPS [powerwalker@localhost] failed - got [ERR ACCESS-DENIED]
ll /run/nut/upsmon.pid
.rw-r--r-- root root 6 B Wed Feb 22 20:46:59 2023  /run/nut/upsmon.pid
```
nut-driver.service and nut-server.service are looking good at the moment.
restarting just that nut-monitor-service will change nothing and do nothing despite changing the PID of upsmon AND the according Log Line "Supervising process PID" (same PID as through nut-monitor.service restarted upsmon).
And the thing is, I never used something else than systemct to control nut-monitor.service.

MarvinMynx, 2023-02-22:
You already told me that I should check the Password in the following two Files / Lines:
```
grep -Ri "abcd1234" /etc/nut/*.conf /etc/nut/*.users
/etc/nut/upsmon.conf:MONITOR powerwalker@localhost 1 admin abcd1234 master
/etc/nut/upsd.users:  password = abcd1234
```
But it is matching. Are there maybe limitations on which character can be used or how long the Password can be?
Mine is 8 Chars/Digits Long and consists only of normal upper- or lower-case letters and Numbers. Nothing else.
So I would be extremely surprised if this were too much.

MarvinMynx, 2023-02-22:
Okay, I have DATA STALE Again... So restarting udev Service and/or rebooting after switching to systemd Services didn't work out either...
So back to Plan B on how to properly build the current Version. I'm still unsure on how to proceed. with configure and make.
(And I kind of still think that just upgrading won't resolve my issue anyways :/ )

Sorry really getting Headaches and Depressed here. Constantly thinking "Comm on it can't be that difficult" ...

MarvinMynx, 2023-02-23:
Okay, so I still don't know which Doc Part actually is needed to run... Since both are Part of the In-Place Upgrade Document and the Descriptions don't differ that much to me.
https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests#replacing-a-nut-deployment
https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests#replacing-a-systemd-deployment
I'm not continuing until I know for certain that I'll build the correct thing, and not destroy my system or create dependency hell or something like this.

MarvinMynx, 2023-02-23:
I now ran the following:
```
git clone https://github.com/networkupstools/nut
cd nut
./autogen.sh
./configure --enable-inplace-runtime
make -j 4 all check
```
Configure had the following output:
````
NUT Configuration summary:
==========================

* configured version:   2.8.0.1 (v2.8.0-Windows-568-gf813c1404)
* build serial drivers: yes
* build USB drivers:    yes (libusb-1.0)
* build neon based XML driver:  yes
* enable Avahi support: yes
* build Powerman PDU client driver:     yes
* build Modbus driver:  yes
* build IPMI driver:    yes (FreeIPMI)
* build Mac OS X meta-driver:   no
* build i2c based drivers:      yes
* enable SSL support:   yes (OpenSSL)
* enable libwrap (tcp-wrappers) support:        yes
* enable libltdl (Libtool dlopen abstraction) support:  yes
* build nut-scanner:    yes
* build CGI programs:   no
* install NUT-Monitor desktop application:      no
* install PyNUT binding module: yes
* use default  Python  interpreter:     /usr/bin/python
* use specific Python2 interpreter:     /usr/bin/python2.7
* use specific Python3 interpreter:     /usr/bin/python3
* build and install documentation:      man=auto
* build specific documentation format(s):       yes  man
* build and install the development files:      no
* consider basic SMF support:   no
* consider basic systemd support:       yes
* build with tighter systemd support:   yes
* build C++11 codebase (client library, etc.):  yes
* build C++ tests with CPPUNIT: yes
* build SNMP drivers:   yes
* build SNMP drivers with statically linked lib(net)snmp:       no
* User to run as:       nobody
* Group of user to run as:      nogroup

NUT Paths:
----------

* Default installation prefix path:     /usr
* State file path:      /var/state/ups
* Unprivileged PID file path:   /var/state/ups
* Privileged PID file path:     /run
* Driver program path:  /usr/bin
* CGI program path:     /usr/cgi-bin
* HTML file path:       /usr/html
* Config file path:     /usr/etc
* Data file path:       /usr/share
* Tool program path:    /usr/bin
* System program path:  /usr/sbin
* System library path:  /usr/lib
* System exec-library path:     /usr/libexec

NUT Paths for third-party integrations:
---------------------------------------

* Default  Python  interpreter site-packages:   /usr/local/lib/python2.7/dist-packages
* Specific Python2 interpreter site-packages:   /usr/local/lib/python2.7/dist-packages
* Specific Python3 interpreter site-packages:   /usr/local/lib/python3.9/dist-packages
* pkg-config *.pc directory:    ${libdir}/pkgconfig => /usr/lib/pkgconfig
* Service units for systemd:    /lib/systemd/system
* Shutdown hooks for systemd:   /lib/systemd/system-shutdown
* Systemd-tmpfiles configs:     /usr/lib/tmpfiles.d
* Augeas lenses directory:      /usr/share/augeas/lenses/dist
* Udev rules directory: /lib/udev

NUT Build/Target system info:
-----------------------------

* Compact version of C compiler:        gcc (Debian 10.2.1-6) 10.2.1 20210110
* Compact version of C++ compiler:      g++ (Debian 10.2.1-6) 10.2.1 20210110
* Compact version of C preprocessor:    gcc (Debian 10.2.1-6) 10.2.1 20210110
* host env spec we run on:      x86_64-pc-linux-gnu
* host env spec we built on:    x86_64-pc-linux-gnu
* host env spec we built for:   x86_64-pc-linux-gnu
* host OS short spec we run on: x86_64-linux-gnu
* host OS short spec we built on:       x86_64-linux-gnu
* host OS short spec we built for:      x86_64-linux-gnu

NUT Compiler settings:
----------------------

* CC            : gcc
* CFLAGS        : -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu99 -Wno-system-headers -Wall -Wextra -Wsign-compare -pedantic -Wno-error
* CXX           : g++
* CXXFLAGS      : -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu++11 -Wno-system-headers -Wall -Wextra -Wno-error
* CPP           : gcc -E
* CPPFLAGS      :
* CONFIG_FLAGS  : --prefix=/usr --sbindir=/usr/sbin --bindir=/usr/bin
```
But now I need to know if the following command is the right one to Safely Replace my Running Version correctly:
```
{ sudo systemctl stop nut-monitor nut-server || true ; } && \
   { sudo systemctl stop nut-driver.service || true ; } && \
   { sudo systemctl stop nut-driver.target || true ; } && \
   { sudo systemctl stop nut.target || true ; } && \
   sudo make install && \
   sudo systemctl daemon-reload && \
   sudo systemd-tmpfiles --create && \
   sudo systemctl disable nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   sudo systemctl enable  nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   { sudo systemctl restart udev || true ; } && \
   sudo systemctl restart nut-driver-enumerator.service nut-monitor nut-server
```
I already have RUN_AS_USER in ups.conf, this should stay during the in-playce upgrade I strongly suggest.
And I would have to delete the udev rule that I created manually.

But since I installed the deb Package via apt the Wiki says I would have to track down my build parameters and paths and stuff and not even a hint on how to do it or how to actually use these if I would already know how to track down these things.
That's why I don't know which one is correct.

MarvinMynx, 2023-02-23:
It would also help if I could get clarification on my last post regarding the configured output.

> NUT Paths:
> ----------
>
> * Default installation prefix path:     /usr
> * State file path:      /var/state/ups
> * Unprivileged PID file path:   /var/state/ups
> * Privileged PID file path:     /run
> * [...]

Are these the Paths the Doc is Talking about that I would need to hunt down for my specific Packaging?
If I would check on my system if these Paths exist and if the config path for example contains my current config, would it be the correct command to replace my installation?
But I already can see that the Config Path differs from mine (/etc/nut/), so I'm again more confused than before because up until now, and according to your replays, I thought the systemd version would be the correct one.
But I still need confirmation if my thoughts are correct. Like I'm running on the spot here. Without further input from anyone really, I can't proceed.

So please someone help...

MarvinMynx, 2023-02-23:
Or are these two not on the same level like
1:
2:
But
1:
1.1?
Because the Configure is only described in the first one, and that describes I need to hunt down these paths and stuff, so if later is the case, I would need to hunt this down in either case, is that correct? 

> https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests#replacing-a-nut-deployment
> https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests#replacing-a-systemd-deployment

MarvinMynx, 2023-02-23:
Okay, I've noticed that the two Versions really don't differ from one another at all. 
The only difference is the Reloading/enabling of the services...
That totally confused me... I would just post one command, and for the systemd part I would make the header a little smaller and then just post the commands to first stop and then start and reload everything instead of duplicating it to a second make command one-liner. That confused me more than anything else in the last 36 Hours.

So, in fact, I would need to hunt down my paths and stuff in either case because its essentials are the same case. 
And I never read anything about how to actually do this or where to start or...

So less confused now, but still Running against a Brick wall not knowing how to proceed 
(How to get all of the appropriate pieces of information to correctly configure it)

Like I already downloaded a .rpm from RHEL8 Repos and looked inside, to get back some default files or track down default paths, but first, I think that that is not exactly what to do here, and second, there would be no indication (including the Doc Files again) on which info's exactly I need out of that File. (Like all the paths that are in the Package, or something else entirely?)
And still, even if I had these pieces of information, I still don't see anything regarding how to use these in the configure command despite: "# --maybe-some-other-options".

The config Path mentioned by me a few moments ago for example, how would I provide the correct path to configure correctly? Nothing in the Docs as far as I can see.

jimklimov, 2023-02-23:
Please, I must finally say this is not the right venue for learning about Linux, OSes or packaging. Just hard for me to pack years of learning into a dozen sentences, tapping on a phone during commute. You ask how, I link the docs which say how (maybe needs a dictionary to decipher though) and literally link to the RPM and DEB recipe files for common distros and you come back giving the impression you barely skimmed it after days of reading and trying. Am I putting more effort into solving your puzzle than you are?

That said, it is hard to guess the unclear parts of the docs precisely because of the experience, in all projects, so constructive feedback from "newcomers" and PRs to improve are welcome.

Back to the topic, RHEL recipes are of moderate use to you on a Debian system like Proxmox. The linked https://salsa.debian.org/debian/nut/-/blob/debian/debian/rules file lists the options they use.

Technically for "in-place" replacements, it suffices to match the config paths and run-time users, and replace the ways system starts NUT so it would run the new build. Overwriting all old binaries in the same places is a space-saving bonus but not a requirement.

jimklimov, 2023-02-23:
As for Data Stale, did you try a shorter polling frequency and/or `pollonly` instead of default work by interrupts (assuming the UPS or port on USB hub become bored when there's no traffic for a while, and begins power-saving - so losing the link)?

When in last experiments it did become stale, did the systemd-managed driver restart work well (e.g. no longer got two drivers pulling the rug from each other)?

Note that with 2.8.0 installed you should comment away the part with restarts (in fact there would be an udev.hwdb instead of udev.rules IIRC) since the goal/hope is that the newer driver reconnects better on its own.

jimklimov, 2023-02-23:
Also one of the threads recently asked about systemd complaints about upsmon PID file - these are expected (search the issue tracker for details) due to multi-process architecture of it.

jimklimov, 2023-02-24:
> * User to run as:       nobody
> * Group of user to run as:      nogroup

I wonder if that causes issues (e.g. some files/dirs get owned by that instead of `nut` inherited from packaging). Perhaps an explicit `--with-user=...` and `--with-group=...` would fare better to build-in same account names.

MarvinMynx, 2023-02-24:
> As for Data Stale, did you try a shorter polling frequency and/or `pollonly` instead of default work by interrupts (assuming the UPS or port on USB hub become bored when there's no traffic for a while, and begins power-saving - so losing the link)?

Without changing any settings, it seems like I don't get Data Stale anymore. I think also since I've completely switched to systemd services. Instead, I get Connection failure: Connection timed out on the Monitor Service and upsc isn't working, same Error after a while.

I'll still try faster Poll Frequency
According to man upsmon.conf the used Parameters for that are POLLFREQ and DEADTIME. 
Default Values Are:
```
POLLFREQ 5
POLLFREQALERT 5
DEADTIME 15 (The Bigger one of POLLFREQ or POLLFREQALERT times 3)
```
These are exactly my Configured Values.
I'll try the Following now:
```
POLLFREQ 2
POLLFREQALERT 2
DEADTIME 6 (The Bigger one of POLLFREQ or POLLFREQALERT times 3)
```
According to "pollonly" I didn't find any hint on the man Page or the Docs. 
I don't know what this is supposed to mean or how I would set this.
I'll let you know if these changes worked out or not in a few hours or so.

> lso one of the threads recently asked about systemd complaints about upsmon PID file - these are expected (search the issue tracker for details) due to multi-process architecture of it.

Is this Message from systemctl nut-server.service also expected?
```
Can't connect to UPS [powerwalker] (usbhid-ups-powerwalker): No such file or directory
```

MarvinMynx, 2023-02-24:
> As for Data Stale, did you try a shorter polling frequency and/or `pollonly` instead of default work by interrupts (assuming the UPS or port on USB hub become bored when there's no traffic for a while, and begins power-saving - so losing the link)?

Didn't Help I have data stale again.

MarvinMynx, 2023-02-25:
Tried to run:
```
git clone https://github.com/networkupstools/nut
cd nut
./autogen.sh
./configure --enable-inplace-runtime --sysconfdir=/etc/nut --includedir=/usr/include --mandir=/usr/share/man --libdir=\$${prefix}/lib/$(DEB_HOST_MULTIARCH) --libexecdir=/usr/libexec --with-ssl --with-nss --with-cgi --with-dev --enable-static --with-statepath=/run/nut --with-altpidpath=/run/nut --with-drvpath=/lib/nut --with-cgipath=/usr/lib/cgi-bin/nut --with-htmlpath=/usr/share/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib/$(DEB_HOST_MULTIARCH)/pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d
make -j 4 all check && \                                                                                                                                                                                  
   { sudo systemctl stop nut-monitor nut-server || true ; } && \
   { sudo systemctl stop nut-driver.service || true ; } && \
   { sudo systemctl stop nut-driver.target || true ; } && \
   { sudo systemctl stop nut.target || true ; } && \
   sudo make install && \
   sudo systemctl daemon-reload && \
   sudo systemd-tmpfiles --create && \
   sudo systemctl disable nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   sudo systemctl enable  nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   { sudo systemctl restart udev || true ; } && \
   sudo systemctl restart nut-driver-enumerator.service nut-monitor nut-server
```
And itr Failed with this Erorr Message:
```
/bin/bash ../libtool  --tag=CC   --mode=link gcc -I../include -I/usr/include/nss -I/usr/include/nspr  -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu99 -Wno-system-headers -Wall -Wextra -Wsign-compare -pedantic -Wno-error -version-info 6:1:0 -export-symbols-regex ^upscli_   -o libupsclient.la -rpath lib upsclient.lo ../common/libcommonclient.la  -lnss3 -lnssutil3 -lsmime3 -lssl3 -lplds4 -lplc4 -lnspr4 -lpthread
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../include -DHAVE_NUTCOMMON=1 -I../include -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu++11 -Wno-system-headers -Wall -Wextra -Wno-error -MT nutclientmem.lo -MD -MP -MF .deps/nutclientmem.Tpo -c nutclientmem.cpp  -fPIC -DPIC -o .libs/nutclientmem.o
libtool:   error: only absolute run-paths are allowed
make[1]: *** [Makefile:919: libupsclient.la] Error 1
make[1]: *** Waiting for unfinished jobs....
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../include -DHAVE_NUTCOMMON=1 -I../include -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu++11 -Wno-system-headers -Wall -Wextra -Wno-error -MT nutclientmem.lo -MD -MP -MF .deps/nutclientmem.Tpo -c nutclientmem.cpp -o nutclientmem.o >/dev/null 2>&1
libtool: compile:  g++ -DHAVE_CONFIG_H -I. -I../include -DHAVE_NUTCOMMON=1 -I../include -isystem /usr/local/include -g -O2 -Wno-reserved-identifier -Wno-unknown-warning-option -std=gnu++11 -Wno-system-headers -Wall -Wextra -Wno-error -MT nutclient.lo -MD -MP -MF .deps/nutclient.Tpo -c nutclient.cpp -o nutclient.o >/dev/null 2>&1
make[1]: Leaving directory '/root/nut/clients'
make: *** [Makefile:755: all-recursive] Error 1
```
[image]

Configure Parameters Where:
```
NUT Configuration summary:
==========================

* configured version:   2.8.0.1 (v2.8.0-Windows-568-gf813c1404)
* build serial drivers: yes
* build USB drivers:    yes (libusb-1.0)
* build neon based XML driver:  yes
* enable Avahi support: yes
* build Powerman PDU client driver:     yes
* build Modbus driver:  yes
* build IPMI driver:    yes (FreeIPMI)
* build Mac OS X meta-driver:   no
* build i2c based drivers:      yes
* enable SSL support:   yes (Mozilla NSS)
* enable libwrap (tcp-wrappers) support:        yes
* enable libltdl (Libtool dlopen abstraction) support:  yes
* build nut-scanner:    yes
* build CGI programs:   yes
* install NUT-Monitor desktop application:      no
* install PyNUT binding module: yes
* use default  Python  interpreter:     /usr/bin/python
* use specific Python2 interpreter:     /usr/bin/python2.7
* use specific Python3 interpreter:     /usr/bin/python3
* build and install documentation:      man=auto
* build specific documentation format(s):       yes  man
* build and install the development files:      yes
* consider basic SMF support:   no
* consider basic systemd support:       yes
* build with tighter systemd support:   yes
* build C++11 codebase (client library, etc.):  yes
* build C++ tests with CPPUNIT: yes
* build SNMP drivers:   yes
* build SNMP drivers with statically linked lib(net)snmp:       no
* User to run as:       nut
* Group of user to run as:      nut
[... 33 lines cut ...]

* Compact version of C compiler:        gcc (Debian 10.2.1-6) 10.2.1 20210110
* Compact version of C++ compiler:      g++ (Debian 10.2.1-6) 10.2.1 20210110
* Compact version of C preprocessor:    gcc (Debian 10.2.1-6) 10.2.1 20210110
* host env spec we run on:      x86_64-pc-linux-gnu
* host env spec we built on:    x86_64-pc-linux-gnu
* host env spec we built for:   x86_64-pc-linux-gnu
* host OS short spec we run on: x86_64-linux-gnu
[... cut ...] are/nut/www --with-pidpath=/run/nut --datadir=/usr/share/nut --with-pkgconfig-dir=/usr/lib//pkgconfig --with-user=nut --with-group=nut --with-udev-dir=/lib/udev --with-systemdsystemunitdir=/lib/systemd/system --with-systemdshutdowndir=/lib/systemd/system-shutdown --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d
```

Nothing left to try again. Building isn't working, despite making sure I'm using the correct commands in this Thread like a thousand times...
Without any further Help, I can't continue.

MarvinMynx, 2023-02-25:
Okay, I compared the Parameters one by one and tried different things. Here is what I've tried, after comparing Every Directory if it exists or not etc.:
```
--- Section 1 ------------------------------------------------------------------------
git clone https://github.com/networkupstools/nut
cd nut
./autogen.sh
--- Section 2 ------------------------------------------------------------------------
./configure --enable-inplace-runtime --prefix= \
	                     --sysconfdir=/etc/nut \
	                     --includedir=/usr/include \
			     --mandir=/usr/share/man \
			     --libexecdir=/usr/libexec \
	                     --with-statepath=/run/nut \
	                     --with-altpidpath=/run/nut \
	                     --with-drvpath=/lib/nut \
	                     --with-cgipath=/usr/lib/cgi-bin/nut \
	                     --with-htmlpath=/usr/share/nut/www \
	                     --with-pidpath=/run/nut \
	                     --datadir=/usr/share/nut \
	                     --with-user=nut --with-group=nut \
	                     --with-udev-dir=/lib/udev \
			     --with-systemdsystemunitdir=/lib/systemd/system \
	                     --with-systemdshutdowndir=/lib/systemd/system-shutdown \
			     --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d \
	                     --with-ssl \
	                     --with-cgi \
--- Section 3 ------------------------------------------------------------------------
	                     --with-pkgconfig-dir=/usr/lib/x86_64-linux-gnu/pkgconfig \
	                     --libdir=/lib/x86_64-linux-gnu
--- Section 4 ------------------------------------------------------------------------
	                     --with-pkgconfig-dir=/usr/lib/$(DEB_HOST_MULTIARCH)/pkgconfig \
	                     --libdir=\$${prefix}/lib/$(DEB_HOST_MULTIARCH) \
--- Section 5 ------------------------------------------------------------------------
						 --with-nss \
	                     --with-dev \
	                     --enable-static \
--- Section 6 ------------------------------------------------------------------------
make -j 4 all check
--- Section 7 ------------------------------------------------------------------------
{ sudo systemctl stop nut-monitor nut-server || true ; } && \
   { sudo systemctl stop nut-driver.service || true ; } && \
   { sudo systemctl stop nut-driver.target || true ; } && \
   { sudo systemctl stop nut.target || true ; } && \
   sudo make install && \
   sudo systemctl daemon-reload && \
   sudo systemd-tmpfiles --create && \
   sudo systemctl disable nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   sudo systemctl enable  nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   { sudo systemctl restart udev || true ; } && \
   sudo systemctl restart nut-driver-enumerator.service nut-monitor nut-server
```
First I Tried to Run Sections 1-3 and then Section 6. The Output:
```
NUT Configuration summary:
==========================

* configured version:   2.8.0.1 (v2.8.0-Windows-568-gf813c1404)
* build serial drivers: yes
* build USB drivers:    yes (libusb-1.0)
* build neon based XML driver:  yes
* enable Avahi support: yes
* build Powerman PDU client driver:     yes
* build Modbus driver:  yes
* build IPMI driver:    yes (FreeIPMI)
* build Mac OS X meta-driver:   no
* build i2c based drivers:      yes
* enable SSL support:   yes (OpenSSL)
* enable libwrap (tcp-wrappers) support:        yes
* enable libltdl (Libtool dlopen abstraction) support:  yes
* build nut-scanner:    yes
[... 203 lines cut ...]
	                     --datadir=/usr/share/nut \
	                     --with-user=nut --with-group=nut \
	                     --with-udev-dir=/lib/udev \
			     --with-systemdsystemunitdir=/lib/systemd/system \
	                     --with-systemdshutdowndir=/lib/systemd/system-shutdown \
			     --with-systemdtmpfilesdir=/usr/lib/tmpfiles.d \
	                     --with-ssl \
	                     --with-cgi \
make -j 4 all check
```
Does this make sense? If my assumptions are correct it would now finally be safe to proceed with the following command:
```
{ sudo systemctl stop nut-monitor nut-server || true ; } && \
   { sudo systemctl stop nut-driver.service || true ; } && \
   { sudo systemctl stop nut-driver.target || true ; } && \
   { sudo systemctl stop nut.target || true ; } && \
   sudo make install && \
   sudo systemctl daemon-reload && \
   sudo systemd-tmpfiles --create && \
   sudo systemctl disable nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   sudo systemctl enable  nut.target nut-driver.target nut-monitor nut-server nut-driver-enumerator.path nut-driver-enumerator.service && \
   { sudo systemctl restart udev || true ; } && \
   sudo systemctl restart nut-driver-enumerator.service nut-monitor nut-server
```
Is that correct?

jimklimov, 2023-02-26:
With a weekend and laptop here, looking at this thread from start again to hopefully make more sense of it. Seems there were at least 3 issues similar to this discussed in parallel, so comments for one could leak into another. Being similar, they still were probably relevant too :)

TL;DR: There was quite a bit of breadcrumb tracking across the logs you posted (thanks, being drowned in debug data may be annoying but is often helpful). Some of the original issues were discussed before. Some not discussed may be about the contents of systemd units delivered years ago by NUT 2.7.4, such as lack of driver restarts if they fail, and/or timing issues during start-up (upsmon logging in to watch a not yet known UPS).

It seems your system may have exposed a new issue with `configure --enable-inplace-runtime` discovery of setup details, possibly due to permissions. Some ideas to try are posted below; I'll also try to reproduce the situation and try to address it if I can.

So just collecting a few bullet-points from a braindump as I go through posts and screenshots; some action items should be marked with a checkbox:

* NUT 2.7.4 is an almost 7-year-old release, unfortunately many OSes still ship it. NUT 2.8.0 is now almost a year old, some OSes (including Debian Experimental repos, I think) began to provide it. Among improvements of 2.8.0 release, and more later (on master branch), there were bugs fixed and enhancements proposed in areas of debug-ability, recovery from connection loss, and systemd service management. So trying a new version was suggested. If your distro does not offer a newer NUT (more so not the master branch snapshots), building one privately is always an option. Which brought up the matter of "how to do it?"
    * Beside the ability to "build and run" NUT, there is a matter of general OS integration - user accounts, filesystem and device node permissions, service start-up and shut-down, configuration file locations, etc. It is perfectly possible to forfeit what OS packagers prepared for you, uninstall their packages (or never have them in the first place), build and "make install" NUT even with defaults (placing all its files in a tree under /usr/local/ups, and third-party systems files like systemd units where appropriate) and run like that. However for the sake of not losing later an ability to upgrade back into what the distro provides when they come around to releasing a new version of the package, people generally want to use the same basic settings (accounts, paths...) - and distros are very opinionated about what goes where (so it is needed to know that out of band).
    * There are literally hundreds of distributions across dozens of operating systems and packaging efforts shipping NUT in some form. NUT as an upstream project is not even aware of most of these. While there is documentation on building it from scratch, it indeed does expect some experience (or eagerness and ability to experiment), and like with a lot of venerable FOSS projects it originally comes from days before packaging as a concept was wide-spread (as well as BYOD approach to systems, so nobody even thought of writing it differently).
    * Nowadays hinting at "how to build something like what your distro ships" is the best approach, especially in terms of keeping this up to date. Alternative would be to link or even copy hundreds of recipes for all the OSes and their releases (and NUT releases), with parameters changing daily as all parts of the equations evolve - unsurmountable.
    * Given that Fedora and Debian have a lot of spin-offs that practically cover a majority of deployed systems, linking to repositories with their packaging recipes was deemed a useful exception - at least to demonstrate what sort of parameters are involved. Still, it is up to end-users to know (or discover) which packaging repositories their deployments actually use, and what the actual (or nearest likely) recipes are involved, if they go along this path. Finding Git/SVN/CVS/Tarball/... sources of their packaging is one approach, finding "source packages" (grouping recipes and a tarball of e.g. NUT) where available is another, reverse-engineering from paths etc. installed by their end-user packages is yet another.
    * Point taken about clarifying in the Wiki article that "DEB rules" and "RPM spec files" mean "recipes" for respective packaging technologies. Somehow readers are expected to know their A-B-C's, and `packaging recipes used by your distribution (e.g. [RPM spec](https://src.fedoraproject.org/rpms/nut/blob/rawhide/f/nut.spec) or [DEB rules](https://salsa.debian.org/debian/nut/-/blob/debian/debian/rules) files, etc.)` seemed unambiguous, but... gotta think about it, re-wording suggestions are welcome.
    * It is hopefully understood that NUT docs are not a proper place for being a textbook on systems administration, programming and packaging, generic and detailed for all OSes and platforms NUT can run on :)
* Higher debug-verbosity settings can help in troubleshooting, in practice up to level 6 `-DDDDDD` can be useful, especially if logged into e.g. a systemd journal, and then as you know there was a disconnection/data stale situation, you dig around that time stamp to see what the different daemons did, particularly if NUT daemons and services tried to reconnect or restart (and if extra restarts from udev rules broke into the logical chain helpfully or toxically).
[... cut ...]
But I think these two were almost addressed by https://github.com/networkupstools/nut/issues/1853#issuecomment-1445172919 : the "Section 3" is better than the literal text in "Section 4", but to match existing packaging perfectly, the prefix should be `/usr` so `--libdir=/usr/lib/x86_64-linux-gnu`

Regarding "Section 5", the `--with-nss` tells it to use Mozilla NSS for crypto; otherwise it would auto-detect if NSS or OpenSSL are available. The other two options are less consequential for run-time builds.

MarvinMynx, 2023-02-26:
Still Reading But I can let you know this so far:

> * One idea addressed recently (and probably your codebase does have that fix) was about inability of an arbitrary user to see into the configuration directory (e.g. `/etc/nut` or `/etc/ups` on most [...]

I can say that I did run everything discussed in this thread at any times with the user root logged in via "sudo -i"

MarvinMynx, 2023-02-26:
huh, i can't replicate my earlier successful Build I described. That's odd...

MarvinMynx, 2023-02-26:
Okay it seems like I were able to reproduce the successful run (getting 'Testsuite summary...') once on the laptop again, and then run the same immediately again and I didn't get the same output. (Not getting 'Testsuite summary...').

I don't know how this is possible but this is definitely the case and I'm starting to question my sanity...

jimklimov, 2023-02-26:
Odd, just realized that `debian/rules` does specify `--prefix=` (empty) and so `--libdir=\$${prefix}/lib/$(DEB_HOST_MULTIARCH)` would resolve to `/lib/...` indeed.

jimklimov, 2023-02-27:
Now, trying to install NUT packages (I am not sure if you pulled the `nut` umbrella only, or additional packages, or just some of the core ones) to configure an "in-place" build later. Output of their packaging looks quirky out of the box:
````
:; apt-get install nut
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  freeipmi-common libavahi-core7 libdbus-1-dev libfreeipmi17 libipmimonitoring6 libpowerman0
Use 'apt autoremove' to remove them.
The following additional packages will be installed:
  libnutscan1 libupsclient4 nut-client nut-server
Suggested packages:
  nut-monitor nut-cgi nut-ipmi nut-snmp nut-xml
The following NEW packages will be installed:
  libnutscan1 libupsclient4 nut nut-client nut-server
0 upgraded, 5 newly installed, 0 to remove and 23 not upgraded.
Need to get 1,759 kB of archives.
After this operation, 6,803 kB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://ftp.cz.debian.org/debian bullseye/main amd64 libupsclient4 amd64 2.7.4-13 [186 kB]
Get:2 http://ftp.cz.debian.org/debian bullseye/main amd64 nut-client amd64 2.7.4-13 [252 kB]
Get:3 http://ftp.cz.debian.org/debian bullseye/main amd64 libnutscan1 amd64 2.7.4-13 [192 kB]
Get:4 http://ftp.cz.debian.org/debian bullseye/main amd64 nut-server amd64 2.7.4-13 [881 kB]
Get:5 http://ftp.cz.debian.org/debian bullseye/main amd64 nut all 2.7.4-13 [247 kB]
Fetched 1,759 kB in 6s (283 kB/s)
Selecting previously unselected package libupsclient4:amd64.
(Reading database ... 121121 files and directories currently installed.)
Preparing to unpack .../libupsclient4_2.7.4-13_amd64.deb ...
Unpacking libupsclient4:amd64 (2.7.4-13) ...
Selecting previously unselected package nut-client.
Preparing to unpack .../nut-client_2.7.4-13_amd64.deb ...
Unpacking nut-client (2.7.4-13) ...
Selecting previously unselected package libnutscan1:amd64.
Preparing to unpack .../libnutscan1_2.7.4-13_amd64.deb ...
Unpacking libnutscan1:amd64 (2.7.4-13) ...
Selecting previously unselected package nut-server.
Preparing to unpack .../nut-server_2.7.4-13_amd64.deb ...
Unpacking nut-server (2.7.4-13) ...
Selecting previously unselected package nut.
Preparing to unpack .../archives/nut_2.7.4-13_all.deb ...
Unpacking nut (2.7.4-13) ...
Setting up libnutscan1:amd64 (2.7.4-13) ...
Setting up libupsclient4:amd64 (2.7.4-13) ...
Setting up nut-client (2.7.4-13) ...
Created symlink /etc/systemd/system/multi-user.target.wants/nut-monitor.service → /lib/systemd/system/nut-monitor.service.
Job for nut-monitor.service failed because the service did not take the steps required by its unit configuration.
See "systemctl status nut-monitor.service" and "journalctl -xe" for details.
Setting up nut-server (2.7.4-13) ...
Created symlink /etc/systemd/system/multi-user.target.wants/nut-server.service → /lib/systemd/system/nut-server.service.
nut-driver.service is a disabled or a static unit, not starting it.
Setting up nut (2.7.4-13) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for libc-bin (2.31-13+deb11u5) ...
````

They suggest a `nut-monitor` package which does not exist: the `nut-client` takes its place to deliver the `nut-monitor` service for `upsmon`, which failed due to lack of configuration (kinda expected out of the box), so lots of scary letters validly:
````
...
Setting up nut-client (2.7.4-13) ...
Created symlink /etc/systemd/system/multi-user.target.wants/nut-monitor.service → /lib/systemd/system/nut-monitor.service.
Job for nut-monitor.service failed because the service did not take the steps required by its unit configuration.
See "systemctl status nut-monitor.service" and "journalctl -xe" for details.
...

:; systemctl status nut-monitor.service
* nut-monitor.service - Network UPS Tools - power device monitor and shutdown controller
     Loaded: loaded (/lib/systemd/system/nut-monitor.service; enabled; vendor preset: enabled)
     Active: failed (Result: protocol) since Mon 2023-02-27 01:02:46 CET; 2min 30s ago
        CPU: 849us

Feb 27 01:02:46 pve systemd[1]: Starting Network UPS Tools - power device monitor and shutdown controller...
Feb 27 01:02:46 pve upsmon[160966]: upsmon disabled, please adjust the configuration to your needs
Feb 27 01:02:46 pve upsmon[160966]: Then set MODE to a suitable value in /etc/nut/nut.conf to enable it
Feb 27 01:02:46 pve systemd[1]: nut-monitor.service: Can't open PID file /run/nut/upsmon.pid (yet?) after start: Operation not permitted
Feb 27 01:02:46 pve systemd[1]: nut-monitor.service: Failed with result 'protocol'.
Feb 27 01:02:46 pve systemd[1]: Failed to start Network UPS Tools - power device monitor and shutdown controller.
````

Results of `make clean ; ./configure --enable-inplace-runtime` on this system were indeed weird, and different for root and unprivileged users, so indeed I have homework to tend to.

MarvinMynx, 2023-02-27:
> (I am not sure if you pulled the `nut` umbrella only, or additional packages, or just some of the core ones)

I've installed nut with the Following two Lines:
```
apt install nut nut-client nut-server
apt install apache2 nut-cgi
```

MarvinMynx, 2023-02-27:
> > * > docs for "pollonly"
> > 
> > See usbhid-ups manpage, e.g. https://github.com/networkupstools/nut/blob/master/docs/man/usbhid-ups.txt It is just a flag setting (nothing to equate to) in an `ups.conf` section for the device, e.g.
> > ```
> > [ups]
> >     driver = usbhid-ups
> >     port = auto
> >     pollonly
> >     # other options
> > ```
> 
> I'm trying this now.

But this seems to Work out. Since I added this to my Config, I didn't see Data Stale or any Relevant Errors so far anymore.

MarvinMynx, 2023-02-27:
> Sounds promising! And that's with 2.7.4 remaining the installed version?

yes.

jimklimov, 2023-03-01:
Regarding in-place builds, I hope PR #1865 will fix the deficiencies found while investigating this thread - thanks for bringing that up.

MarvinMynx, 2023-03-01:
No Problem :)

jimklimov, 2023-03-02:
Can this ticket be closed then - as far as the original scope was to monitor more reliably with what NUT you have deployed, it is achieved by `pollonly`?

Ideas to build a newer NUT that may be even more resilient became a new and separate issue of sorts, and ultimately not an unavoidable and required step to solution "here and now"?

MarvinMynx, 2023-03-02:
Yes, this Ticket can be Closed.
Thank you very much, for all your Support.

closed 2023-03-02 by jimklimov