networkupstools/nut #1856: Salicru SPS 1500 AVD T supported by nutdrv_qx

DEADSEC-SECURITY, 2023-02-22:
I have my SPS connected to my NUT server with a SERIAL to USB adaptor and I'm using the nutdrv_qx driver which is the only one able to detect and connect to the UPS, unfortunately even though it can get some data, the driver is not able to get everything from the UPS.

The SPS works properly with their recommended software (PowerMaster+). Maybe there is a driver or sub driver with better support with PowerMaster+ protocol?

My config:
```
driver: nutdrv_qx
name: ups
port: /dev/ttyUSB0
```

Data being able to read:

- [X] Input Voltage
- [X] Load
- [X] Output Voltage
- [X] UPS Status

Output from log:
[image]

Let me know if you guys need more information.

jimklimov, 2023-02-22:
Try NUT 2.8.0, some salicru sub-drivers were added/updated there.

DEADSEC-SECURITY, 2023-02-23:
Will do, I was using Home Assistant Addon but ill run it on a Raspberry so I get the newest version @jimklimov

jimklimov, 2023-02-28:
Note there were wiki articles and community posts about building HA plugins with current NUT. Never tried it myself, so passing "as is" :)

DEADSEC-SECURITY, 2023-03-01:
I tested in my raspberry PI and getting the same data:
```
Init SSL without certificate database
battery.voltage: 27.20
device.type: ups
driver.name: nutdrv_qx
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 2
driver.parameter.port: /dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_DECBb11A920-if00-port0
driver.parameter.synchronous: no
driver.version: 2.7.4
driver.version.data: Q1 0.07
driver.version.internal: 0.28
input.frequency: 50.1
input.voltage: 224.5
input.voltage.fault: 224.5
output.voltage: 224.9
ups.beeper.status: enabled
ups.delay.shutdown: 30
ups.delay.start: 180
ups.load: 28
ups.status: OL
ups.temperature: 25.0
ups.type: offline / line interactive
```

DEADSEC-SECURITY, 2023-03-01:
It still complains about not being able to get battery data. @jimklimov

DEADSEC-SECURITY, 2023-03-01:
I just noticed Im using 2.7.4 @jimklimov but I cant seem to update it to 2.8.0. Is it the apt repository yet?

jimklimov, 2023-03-01:
That's a question to the distro. Some iterations of 2.8.0 were in debian-experimental (and maybe sid), not sure about other debianish forks - and how long it takes for experiments to trickle down there. Or which repos to add/enable if they did.

It should not be hard to build NUT, more so for the compatibility tests that can be done without replacing anything in the OS: https://github.com/networkupstools/nut/wiki/Building-NUT-for-in%E2%80%90place-upgrades-or-non%E2%80%90disruptive-tests

Also, just in case check if the device also talks standard protocols - e.g. if usbhid-ups would recognize it better?

DEADSEC-SECURITY, 2023-03-02:
I installed version 2.8.0 from debian SID but im getting permission erros and when I run nut-scanner it can't find devices:

Running `sudo upsdrvctl start`
```
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

Driver failed to start (exit status=1)
```

The same when I use the generic driver.

Also I have nut user as root:
[image]

... so im kinda lost on whats causing the permission error.

That was with port auto and vendorid specified. 
When I put the serial port `/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_DECBb11A920-if00-port0`
I get this error:
```
Network UPS Tools - UPS driver controller 2.8.0
Network UPS Tools - Generic Q* USB/Serial driver 0.32 (2.8.0)
USB communication driver (libusb 1.0) 0.43
WARNING: warn_if_bad_usb_port_filename(): port argument specified to
  the driver is "/dev/serial/by-id/usb-Prolific_Technology_Inc._USB-Serial_Controller_DECBb11A920-if00-port0" but USB drivers do not use it and rely on
  libusb walking all devices and matching their identification metadata.
  NUT documentation recommends port="auto" for USB devices to avoid confusion.
libusb1: Could not open any HID devices: insufficient permissions on everything
No supported devices found. Please check your device availability with 'lsusb'
and make sure you have an up-to-date version of NUT. If this does not help,
try running the driver with at least 'subdriver', 'vendorid' and 'productid'
options specified. Please refer to the man page for details about these options
(man 8 nutdrv_qx).

Driver failed to start (exit status=1)
```

Output from lsusb:
[image]

Any thing I might've missed @jimklimov?

DEADSEC-SECURITY, 2023-03-02:
ups.conf looks like this:
[image]

and the file permissions of the configs:
[image]

jimklimov, 2023-03-02:
Typically config files with passwords should not be readable by world, some versions and tools of NUT may complain and ignore these. 

Screenshot with ups.conf only has a vendorid but not a productid - both are needed to select a subdriver.

jimklimov, 2023-03-02:
Also, are you sure no other driver runs that already took the device?

DEADSEC-SECURITY, 2023-03-03:
Will try the productID too and check systemd but I dont think so.

DEADSEC-SECURITY, 2023-03-03:
Tried changing permissions to all usb_devices using udev but no luck:
[image]

PS: Reloaded the plugin and restarted the raspberry after creating rules but nothing changed when trying to start nut.

DEADSEC-SECURITY, 2023-03-03:
Also not sure if relevant but I'm using a serial to USB adaptor plugged into the USB3 port of the raspberry

DEADSEC-SECURITY, 2023-03-03:
This is the adaptor: https://www.amazon.es/gp/product/B00QUZY4UG/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

DEADSEC-SECURITY, 2023-03-03:
Kinda running out of ideas rn. Let me know what you think @jimklimov

jimklimov, 2023-03-03:
Really odd. Can you try running the driver with higher debug verbosity? I think in 2.8.0 `upsdrvctl -DDDDDD -d start` could be done to pass debug flags to driver. If not, it should show the exec line as its own debug, and you copy-paste that with `-DDDDDD` added.

My guess so far is that another copy of the driver runs and has grabbed the USB port exclusively. You mentioned "restarting the plugin" - could that be it? Or some other serial/USB program on it?..

jimklimov, 2023-03-03:
Also, if the UPS is a serial-port one, then NUT monitoring the serial port adapter does not make sense and it probably does not serve UPS USB HID anyway. I kinda missed that part at first.

Try setting the `port=/dev/ttyUSB0` or similar, as *that* is a serial-port shim over USB link.

DEADSEC-SECURITY, 2023-03-03:
I dont have any tty though @jimklimov 
[image]

DEADSEC-SECURITY, 2023-03-03:
Well I got it connected @jimklimov, good call on the port btw didn't think of that.

I think its fixed. Battery voltage now available, will have to test the full functionality but seems good for now.

[image]

DEADSEC-SECURITY, 2023-03-03:
Just the battery pack number that it still cant detect but might be because of the UPS it self.

DEADSEC-SECURITY, 2023-03-03:
Well I got 2.8.0 running everything seems fine and working. Would be cool to bump HA version to 2.8.0 and also would be great to add runtime to the driver.

jimklimov, 2023-03-04:
Thanks for the report!

As for HA - that's up to them. Unrelated project (to NUT). From what I gather, recipes to roll your own are out there (with a copy on our wiki).

closed 2023-10-25 by DEADSEC-SECURITY

AppleCarmine, 2024-10-16:
> Bene, l'ho collegato@jimklimov, bella chiamata sul porto, tra l'altro non ci avevo pensato.
> 
> Penso che sia stato risolto. Ora è disponibile la tensione della batteria, dovrò testare la funzionalità completa, ma per ora sembra tutto a posto.
> 
> [image]

Hi, how did you finally solve it? What driver did you use?