networkupstools/nut #1867: IPv4/ipv6 locahost conflict on the same UPS device ? really?

cypa, 2023-03-05:
"What happens is that NUT tries to LISTEN on both ipv4 and ipv6. It can only LISTEN on one of them so put out the error message for the one it can't LISTEN to." https://forums.linuxmint.com/viewtopic.php?t=97487 

got confusing error messages on single UPS device on single server 
`мар 05 13:40:59 server2 upsmon[865527]: Communications with UPS 1USB@localhost established
мар 05 13:40:54 server2 upsmon[865527]: Login on UPS [1USB@localhost] failed - got [ERR ACCESS-DENIED]` 
it's both success and error, really? dear developers, it is very confusing!

jimklimov, 2023-03-05:
> on UPS ***@***.*** failed - got [ERR ACCESS-DENIED]

Actually it should be able to listen on both, NIT tests check for that
nowadays. Some systems aliased `localhost` string to both 127.0.0.1 and ::1
and fail half the connections if only one is listened to. So later
`localhost6` appeared.

In the email log, failure and success were 5 seconds apart.

In a recent thread (maybe on github) we discussed issues with NUT 2.7.4
packaging, which is old but still shipped by many distros, and how service
dependencies are (not) spelled out there. One guess was that there (by log
timestamps) everything started at once - drivers, upsd and upsmon. So the
driver took a few seconds to talk to device and make itself known by socket
file - only then upsd became aware of it. First attempts of upsmon to log
in for management of the UPS were before upsd knew it exists (ok that is
known from ups.conf) and is really handled. Re-attempts later succeeded.

Maybe something similar happens here too?

Hope this helps,
Jim Klimov

cypa, 2023-03-05:
> Re-attempts later succeeded. Maybe something similar happens here too?

well, maybe, but there is no connect success message after fail message - this is misleading anyway 

and how do I check if upsmon connected successfully?

jimklimov, 2023-03-05:
> how do I check if upsmon connected successfully?

Probably by bumping debug verbosity of upsmon and/or upsd. Also by the
message it emits, if you are there to catch it :)

You can also query clients logged in for each UPS (there is no listing of
"all" connections though, so you have to list the devices and go one by one
for their clients), via protocol (note: I added the blanks for readability,
but really it is just a wall of text):

$ telnet localhost 3493
Trying 127.0.0.1...
Connected to localhost.localdomain.
Escape character is '^]'.

LIST UPS
BEGIN LIST UPS
UPS dummy "Description unavailable"
UPS eco650 "Eaton Ellipse ECO 650"
END LIST UPS

LIST CLIENT eco650
BEGIN LIST CLIENT eco650
CLIENT eco650 127.0.0.1
END LIST CLIENT eco650

LIST CLIENT dummy
BEGIN LIST CLIENT dummy
END LIST CLIENT dummy

LOGOUT
OK Goodbye
Connection closed by foreign host.

> there is no connect success message after fail message

Not quite so: there *are* two messages in your message, one from 13:40:54
(ERR) and another from 13:40:59 (conn. established); no idea why your log
reported that so weirdly:

> мар 05 13:40:59 server2 upsmon[865527]: Communications with UPS
***@***.*** established мар 05 13:40:54 server2 upsmon[865527]: Login on
UPS ***@***.*** failed - got [ERR ACCESS-DENIED]

Jim