networkupstools/nut #1868: Smart UPS 1500(SMT1500RM2UC) does not report input.*, output.*, ups.load, or ups.realpower.nominal

eamathw, 2023-03-07:
Smart UPS 1500, Model: SMT1500RM2UC
Nut v2.8.0

upsc does not report input.\*, output.\*, ups.load, or ups.realpower.nominal values.  There is no way to measure the load on the UPS.

/etc/nut/ups.conf:

```
[upsr3n1]
	driver = "usbhid-ups"
	port = "auto"
	vendorid = "051D"
	productid = "0003"
	product = "Smart-UPS_1500 FW:UPS 06.0 / ID=1027"
	serial = "3S2231X16739"
	vendor = "American Power Conversion"
	bus = "002"
```

```
$ upsc upsr3n1
Init SSL without certificate database
battery.charge: 100
battery.charge.low: 10
battery.charge.warning: 50
battery.runtime: 2844
battery.runtime.low: 150
battery.type: PbAc
battery.voltage: 26.3
battery.voltage.nominal: 24.0
device.mfr: American Power Conversion 
device.model: Smart-UPS_1500
device.serial: 3S2231X16739    
device.type: ups
driver.name: usbhid-ups
driver.parameter.bus: 002
driver.parameter.pollfreq: 30
driver.parameter.pollinterval: 15
driver.parameter.port: auto
driver.parameter.product: Smart-UPS_1500 FW:UPS 06.0 / ID=1027
driver.parameter.productid: 0003
driver.parameter.serial: 3S2231X16739
driver.parameter.synchronous: auto
driver.parameter.vendor: American Power Conversion
driver.parameter.vendorid: 051D
driver.version: 2.8.0
driver.version.data: APC HID 0.98
driver.version.internal: 0.47
driver.version.usb: libusb-1.0.23 (API: 0x1000107)
ups.beeper.status: enabled
ups.delay.shutdown: 20
ups.firmware: UPS 06.0 / ID=1027
ups.mfr: American Power Conversion 
ups.mfr.date: 2022/08/05
ups.model: Smart-UPS_1500
ups.productid: 0003
ups.serial: 3S2231X16739    
ups.status: OL
ups.timer.reboot: -1
ups.timer.shutdown: -1
ups.vendorid: 051d

```
This was also a problem on Nut v2.7.4. I compiled and installed v2.8.0 to ensure it was not fixed in the latest release.

Please let me know if you need any other information, logs, debug output, etc.

Thank you.

jimklimov, 2023-03-12:
I wonder if it is "too new" so might just not serve it? (APCs after 2010 apparently moved most abilities from USB HID to Modbus) See #139 et al for more context.

Just in case, see docs/ for hid-subdriver development - try to get a data walk of your UPS and compare the generated subdriver to existing one. Maybe new data points would be seen...

eamathw, 2023-03-16:
I tried APCUPSD and it also does not report the data I was looking for. Even if I tried setting UPSTYPE to "modbus".
Closing.

closed 2023-03-16 by eamathw

jimklimov, 2023-03-16:
Thanks for the heads-up! If you can, as a customer try asking APC for more info?