networkupstools/nut #1869: driver to ACPI interface to use NUT services on laptops

cypa, 2023-03-08:
I have no UPS device but I need to setup and test NUT-services - what should I do? 
If there are NUT-driver to ACPI battery of my laptop then I can run and test NUT services ...

jimklimov, 2023-03-08:
This is something I wondered about over time; I think it is currently lacking. In early 2.6.x a HAL driver existed but was dropped since contemporary GNOME releases lost the ability. MATE is a GNOME2 fork still popular so maybe that driver would still make sense...

cypa, 2023-03-09:
could you please hint how to make the driver?

jimklimov, 2023-03-09:
Generally there are "hints" in `docs/` directory, and the example "skel" driver (can search for it as a keyword in docs I guess).

Currently there is a new driver being PR'ed in #1855, with a new access medium (and third-party lib dependency) at that - you can take a look there at a practical example of what this involves.

aquette, 2023-03-14:
Starting from my mentioned HAL work base, it should not be that hard ri reconnect to nowadays UPower, since it's its child

aquette, 2023-03-14:
Going the /dev/acpi way is another, more straight option.
And back to the basic nut, this is why I develop dummy-ups 😉
Check it out, along With the DDL website

aquette, 2023-03-14:
@jimklimov I once wanted to put more light on dummy-ups, renaming it, listing in HCL, ... Not sure I made a move

AT-StephenDetomasi, 2025-01-19:
I just went down the NUT rabbit hole and found myself in a similar predicament. I am using a Lenovo Thinkpad an a compact Proxmox hypervisor and I can't seem to get NUT to interface with the internal battery. Some sort of integration with ACPI would be really appreciated.

jimklimov, 2025-01-19:
Pull requests would also be appreciated :)

Following the recent `hw_ina219` driver as an example, that parses Linux sysfs files, can be another quick and dirty option at least for one platform...

cbrunnkvist, 2025-08-13:
> I am using a Lenovo Thinkpad an a compact Proxmox hypervisor and I can't seem to get NUT to interface with the internal battery. Some sort of integration with ACPI would be really appreciated.

LOL I am in **precisely** the same situation! I wouldn't even bother about ACPI, I'd be fine taking the output of a cron:ed shellscript for all I care... This can't be THAT rare of a use case, it ought to be documented somewhere.

Like, I can get this far:

```sh
#!/bin/bash

PATH=$PATH:/usr/bin
# This script is called by the NUT 'exec' driver.
# It must output variables in the format: VAR <variable> "<value>"

# Find the primary battery path from upower
BATTERY_PATH=$(upower -e | grep 'battery_BAT' | head -n 1)

if [ -z "$BATTERY_PATH" ]; then
    # If no battery, output a sensible default state
    echo 'VAR ups.status "OL"'
    echo 'VAR battery.charge "100"'
    exit 0
fi

# Get battery percentage
CHARGE=$(upower -i "$BATTERY_PATH" | grep -E "percentage" | awk '{print $2}' | sed 's/%//')

# Get battery state
STATE=$(upower -i "$BATTERY_PATH" | grep -E "state" | awk '{print $2}')

# Get time to empty in minutes, then convert to seconds
TIME_TO_EMPTY_MIN=$(upower -i "$BATTERY_PATH" | grep -E "time to empty" | awk '{print $4}')
if [[ "$TIME_TO_EMPTY_MIN" =~ ^[0-9.]+$ ]]; then
    RUNTIME=$(echo "$TIME_TO_EMPTY_MIN * 60" | bc | cut -d. -f1)
else
    RUNTIME=0
fi

# Translate the system state to NUT status codes
NUT_STATUS="OL" # Default to "On Line"
if [ "$STATE" = "discharging" ]; then
    NUT_STATUS="OB" # "On Battery"
elif [ "$STATE" = "charging" ]; then
    NUT_STATUS="OL CHRG" # "On Line" and "Charging"
fi

# --- Output for NUT ---
# Print the variables in the required format. We now use standard
# NUT variable names, not the "dummy.*" ones.
echo "VAR ups.status \"$NUT_STATUS\""
echo "VAR battery.charge \"$CHARGE\""
echo "VAR battery.runtime \"$RUNTIME\""
```

but how in the world I get that into a nut ups driver, I just can't figure out.

AT-StephenDetomasi, 2025-08-13:
Just an FYI, I ended up using upower for this. Works fine. My VMs shutdown thanks to QEMU tools so no real need for a full NUT setup. Here are my notes: 

Install upower & config

apt update -y && apt install upower && nano /etc/UPower/UPower.conf

Adjust the config accordingly. Keep in mind it can take several minutes for Proxmox to shutdown, and you have to consider that batteries degrade. Below are some suggested configs:

PercentageLow=45
PercentageCritical=40
PercentageAction=40
CriticalPowerAction=PowerOff

Restart the config

systemctl restart upower

Check the status

upower -d

Done!

cbrunnkvist, 2025-08-13:
@AT-StephenDetomasi cool, yeah in fact I started on a upower adapter driver a while back but got sidetracked ... 
I'm playing around with (ab)using dummy-ups driver for this purpose instead: 
https://gist.github.com/cbrunnkvist/6ad2f44787048de268281a0e01dcf621

It kind of works.