networkupstools/nut #1871: Baytech Driver Requires Additional Outlet States

iunderwood, 2023-03-13:
I am using this project through its integration with the OpenGear commercial terminal server product.  Through some trial and error I have found that there is a bug in the driver which prevents it from starting.

The source of the bug is in the following in drivers/baytech-mib.c:

`static info_lkp_t baytech_outlet_status_info[] = {
	{ -1, "error", NULL, NULL },
	{ 0, "off", NULL, NULL },
	{ 1, "on", NULL, NULL },
	{ 2, "cycling", NULL, NULL }, /* transitional status */
	{ 0, NULL, NULL, NULL }
};`

I was able to determine this based upon where a diagnostic stopped when running the driver with debugging:

`$ sudo -u ups /bin/snmp-ups -D -a rpc_huba02rpdu3
Network UPS Tools - Generic SNMP UPS driver 0.56 (2.6.2)
   0.000000     debug level is '1'
   0.007640     SNMP UPS driver : entering upsdrv_initups()
   0.201460     load_mib2nut: trying classic method with 'baytech' mib
   0.217200     load_mib2nut: using baytech mib
   0.230692     Detected MMP101 on host huba02rpdu3 (mib: baytech 4031)
...
   0.488313     getting data: outlet.6.status (.1.3.6.1.4.1.4779.1.3.5.3.1.3.1.6)
   0.500882     su_find_infoval: found on (value: 1)
   0.501368     entering su_setinfo(outlet.6.status)
   0.501952     getting data: outlet.7.status (.1.3.6.1.4.1.4779.1.3.5.3.1.3.1.7)
$ `

From there, a specific get revealed the value returned:

` % snmpget -v 2c -c rw -m ALL huba02rpdu3 .1.3.6.1.4.1.4779.1.3.5.3.1.3.1.7
Baytech-MIB-503-1::sBTAModulesRPCOutletState.1.7 = INTEGER: lockoff(4)`

This led me to the values that exist in the MIB, which may have been expanded since the driver was first implemented:

`sBTAModulesRPCOutletState OBJECT-TYPE
   SYNTAX INTEGER {
        off	(0),
        on	(1),
        reboot	(2),
        lockon	(3),
        lockoff	(4),
        unlock	(5),
        unknown	(6)
   }`

I'll not use the lock for the time being, but it would be handy to have these values added if time allows.

jimklimov, 2023-03-14:
Can you please re-check with current NUT (e.g. build it on PC and run the driver right from that build to probe the device)? https://github.com/networkupstools/nut/blob/v2.6.2/drivers/snmp-ups.c is 12 years old... maybe the issue was addressed since?

SNMP-UPS should not have many issues building and running from anywhere you can get libsnmp, nowadays including Windows (MSYS2/MinGW64).

If modern NUT works, gotta get OpenGear to update their integration...

iunderwood, 2023-03-14:
Thanks for the direction.  I'll spin up a new VM and let you know what I find out.

jimklimov, 2023-03-14:
Note https://github.com/networkupstools/nut/blob/master/docs/config-prereqs.txt for package names you'd need on those several OSes who are tested by NUT CI. Generally Linux would be easiest, but a lot more platforms are well supported.

jimklimov, 2023-03-23:
By the way, at least in current code *I think* the lack of mapping value would not have caused the driver to crash like it seems to have been for your 2.6.2.

Out of curiosity, can you verify if a recent build of NUT survives this situation properly (effectively just skipping the value/reading)?

iunderwood, 2023-03-24:
I apologize for the late response, but I surmise you are correct in that the original NUT version I am working with has the crash and cannot handle the state.  I verifed the debugs on my Ubuntu test host simply state that there is no information to match:

'''
   0.792526     getting data: outlet.7.status (.1.3.6.1.4.1.4779.1.3.5.3.1.3.1.7)
   0.805774     su_find_infoval: found on (value: 1)
   0.805811     entering su_setinfo(outlet.7.status)
   0.805829     getting data: outlet.8.status (.1.3.6.1.4.1.4779.1.3.5.3.1.3.1.8)
   0.819415     su_find_infoval: no matching INFO_* value for this OID value (4)
   0.819453     entering su_setinfo(outlet.8.status)
   0.819470     outlet template definition found (outlet.%i.desc)...
   0.819483     instantiate_info(outlet.%i.desc)
'''

I'll submit a request with the vendor to upgrade the version of NUT they are using so this can be avoided.

Thang you again, for adding the missing values.

jimklimov, 2023-03-24:
Thanks for confirmation. Note that the latest released NUT is 2.8.0 a year ago. I expect to cut 2.8.1 after investigating some outstanding issues - probably vendors would want to package a tagged release (though with CI in place, master branch is rarely broken and "worse" than a release).

jimklimov, 2023-03-29:
Did you have a chance to build and try PR sources from https://github.com/jimklimov/nut/tree/issue-1871 to see if now it would also recognize those outlet states for your device?

closed 2023-03-29 by jimklimov

iunderwood, 2023-03-31:
I have not had time this week to build from the PR.  I do trust the additions will work as defined.  Additionally, I did submit a feature request to OpenGear to update their version of NUT that is packaged with the system.

Thank you the help!