networkupstools/nut #1878: Riello UPS compatibility list HCL

mzampieri70, 2023-03-24:
Dear maintainers please update the Riello UPS compatibility list (HCL) on the web site:

with riello_usb driver
IDG 400/600/800/1200/1600
USB

IPG 600/800
USB

NPW 600/800/1000/1500/2000
USB

VST 800/1100/1500/2000
USB

VSR 800/1100
USB

VSD 1100/1500/2200/3000
USB

SEP 700/1000/1500/2200/3000
USB

SDH 1000/1500/2200/3000
USB

SDU 4000/5000/6000/8000/10000
USB

STW 5000/6000/8000/10000
USB

S3M
USB

S3T
USB

with riello_ser

VST 800/1100/1500/2000

VSR 800/1100

VSD 1100/1500/2200/3000

SEP 700/1000/1500/2200/3000

SDH 1000/1500/2200/3000

SDU 4000/5000/6000/8000/10000

STW 5000/6000/8000/10000

S3M

S3T

MPM

MPT

MHT

MHE

NXE

with snmp-ups
All UPS with NetMan 204

All UPS with NetMan 208

Thank you
Massimo Zampieri

jimklimov, 2023-08-30:
Cheers @mzampieri70, and sorry this chore took a while to get started :\

One question: you mentioned "All UPS with NetMan 204/208" and the `data/driver.list.in` file lists several instances of "Netman Plus XYZ SNMP Card". So, should these two also have a `Plus` in the name?

closed 2023-09-30 by jimklimov

ha11oga11o, 2024-03-07:
Hello all,

im sorry for hijacking this thread, if its not ok i will delete it, but @mzampieri70 can you please check this post out? It might need your insight. Thank you in advance.

[https://github.com/networkupstools/nut/issues/2254](url)