networkupstools/nut #1879: FDs 0,1,2 are not always STDIO

kkalganov, 2023-03-25:
[background()](https://github.com/networkupstools/nut/blob/ff16dabca191e5fd8ddc20137317bdebee554d8d/common/common.c#L175) closes the first 3 FDs assuming they are allocated for STDIO, which is not always the case.

While keeping _upsd_ in foreground I observe:

```shell
 6148 root       908 S    /usr/sbin/upsd -4 -D 
 6149 root      1400 R    ps 
$ ls -l /proc/6148/fd/
lrwx------    1 root     root           64 Mar 25 09:31 0 -> socket:[6218498]
lrwx------    1 root     root           64 Mar 25 09:31 1 -> socket:[6218500]
``` 
 
which when going into background leads to both SYSLOG and Server sockets being dropped and replaced by _/dev/null_ which makes _uspd_ to silently eat 100% CPU, due to
```shell
write(1, "<31>Mar 23 22:32:35 upsd[4348]: [D2] mainloop: polling 1 filedescriptors\0", 73) = 73
rt_sigaction(SIGPIPE, {SIG_DFL, [RT_69 RT_71 RT_72 RT_75 RT_77 RT_80 RT_82 RT_83 RT_84 RT_86 RT_88 RT_90 RT_92 RT_94], SA_NOCLDSTOP}, NULL, 16) = 0
poll([{fd=2, events=POLLIN}], 1, 2000)  = 1 ([{fd=2, revents=POLLIN}])
accept(2, 0x7fff7b18, [128])            = -1 ENOTSOCK (Socket operation on non-socket)
```

The patch below solves the issue for me (I only use _upsd_ and a couple of drivers, so those are the only parts of the project I have verified the fix for).
```
--- a/server/upsd.c     2023-03-23 23:19:07.000000000 +0100
+++ b/server/upsd.c     2023-03-23 23:21:16.000000000 +0100
@@ -1412,6 +1412,13 @@

        setup_signals();

+       /* we are going to blindly re-allocate FDs 0,1,2 in backgound() */
+       /* yet there is no guarantee they are allocated for STDIO */
+       /* make sure they are allocated for something to avoid using them for important stuff */
+       do {
+               i = open("/dev/null", O_RDWR);
+       } while ((0 <= i) && (i < STDERR_FILENO));
+
        open_syslog(progname);

        /* send logging to the syslog pre-background for later use */
--- a/drivers/main.c    2023-03-23 23:44:39.000000000 +0100
+++ b/drivers/main.c    2023-03-23 23:45:54.000000000 +0100
@@ -685,6 +685,14 @@
        group = xstrdup(RUN_AS_GROUP);  /* xstrdup: this gets freed at exit */

        progname = xbasename(argv[0]);
+
+       /* we are going to blindly re-allocate FDs 0,1,2 in backgound() */
+       /* yet there is no guarantee they are allocated for STDIO */
+       /* make sure they are allocated for something to avoid using them for important stuff */
+       do {
+               i = open("/dev/null", O_RDWR);
+       } while ((0 <= i) && (i < STDERR_FILENO));
+
        open_syslog(progname);

        upsdrv_banner();
--- a/clients/upsmon.c  2023-03-24 12:37:11.000000000 +0100
+++ b/clients/upsmon.c  2023-03-24 12:38:53.000000000 +0100
@@ -2217,6 +2217,13 @@
        argc -= optind;
        argv += optind;

+       /* we are going to blindly re-allocate FDs 0,1,2 in backgound() */
+       /* yet there is no guarantee they are allocated for STDIO */
+       /* make sure they are allocated for something to avoid using them for important stuff */
+       do {
+               i = open("/dev/null", O_RDWR);
+       } while ((0 <= i) && (i < STDERR_FILENO));
+
        open_syslog(prog);

        loadconfig();
--- a/clients/upssched.c        2023-03-24 12:43:40.000000000 +0100
+++ b/clients/upssched.c        2023-03-24 12:45:21.000000000 +0100
@@ -954,6 +954,7 @@
 int main(int argc, char **argv)
 {
        const char      *prog = NULL;
+       int i;
        /* More a use for argc to avoid warnings than a real need: */
        if (argc > 0) {
                xbasename(argv[0]);
@@ -963,6 +964,13 @@

        verbose = 1;            /* TODO: remove when done testing, or add -D */

+       /* we are going to blindly re-allocate FDs 0,1,2 in backgound() */
+       /* yet there is no guarantee they are allocated for STDIO */
+       /* make sure they are allocated for something to avoid using them for important stuff */
+       do {
+               i = open("/dev/null", O_RDWR);
+       } while ((0 <= i) && (i < STDERR_FILENO));
+
        /* normally we don't have stderr, so get this going to syslog early */
        open_syslog(prog);
        syslogbit_set();
--- a/clients/upslog.c  2023-03-24 12:55:48.000000000 +0100
+++ b/clients/upslog.c  2023-03-24 12:57:52.000000000 +0100
@@ -459,6 +459,15 @@
        if ((argc == 1) || (argc == 2))
                help(prog);

+       /* we are going to blindly re-allocate FDs 0,1,2 in backgound() */
+       /* yet there is no guarantee they are allocated for STDIO */
+       /* make sure they are allocated for something to avoid using them for important stuff */
+       do {
+               i = open("/dev/null", O_RDWR);
+       } while ((0 <= i) && (i < STDERR_FILENO));
+
+       open_syslog(prog);
+
        /* see if it's being called in the old style - 3 or 4 args */

        /* <system> <logfn> <interval> [<format>] */
@@ -511,8 +520,6 @@
        /* now drop root if we have it */
        new_uid = get_user_pwent(user);

-       open_syslog(prog);
-
        if (foreground < 0) {
                if (logfile == stdout) {
                        foreground = 1;

```

P.S. I also noticed several places in the code where logging is used prior to opening syslog, that does not affect me so I won't bother to file a bug.

jimklimov, 2023-03-25:
Very odd, which OS is that? stdin, stdout and stderr are 0-1-2 everywhere for decades... They can be non-interactive, closed, redirected from/to /dev/null etc. 

I agree that POSIX standard tokens are better than magic numbers however :)

Would you care to open this as a PR in your name, so we can be sure all OSes in NUT CI matrix have the macros defined?

jimklimov, 2023-03-25:
Also, to avoid repetitions, especially of the comment, makes sense to define this as a method in common.c/h

kkalganov, 2023-03-26:
I agree that is is uncommon, yet I've seen other (embedded) projects (like [dnsmasq](https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=blob;f=src/dnsmasq.c;h=bd3dcf5ee455d6abd834108163701dbbf179e513;hb=HEAD#l160)) to have similar mitigations.
The OS I am seeing on is a heavily customized OpenWrt - no SYSV, no systemd, no udev. I get this behavior when upsd is run from within the kernel event handler, I have regular STDIO FDs occupied when run upsd from shell. Supposedly it has to do with the way the kernel launches the userspace handler.

It is your project, you are free to pick the way you want to fix this, I just shared one of possible solutions.

I filed https://github.com/networkupstools/nut/issues/1881#issue-1640890557 per your request.

jimklimov, 2023-03-29:
The idea was about a pull request (with git commits attributing you with the fix), not another issue :)
If that's too cumbersome/undesirable, just let me know and I'll adjust your patch from the original post.

jimklimov, 2023-03-31:
Hello again, can you please check if PR #1889 (e.g. your build of branch https://github.com/jimklimov/nut/tree/issue-1879) addresses the issue on that platform?

commit 04a4c8e5 (2023-03-31): common/common.c: background(): use STDIN_FILENO/STDOUT_FILENO/STDERR_FILENO macros instead of hard-coded FD 0-1-2 numbers [#1879, #1881]

Closes: #1879
Closes: #1881
closed 2023-03-31 by jimklimov