networkupstools/nut #1883: Cross build on i386 to armv7

idabsent, 2023-03-30:
Hello. Sorry for my English. This text is written with a web translation tool and with my knowledge of English.

I have a Debian (i386) based system built with debootstrap. Try to generate a configuration(**configure** executable script) file for sources (I build binaries and libraries from sources) on this system  **autom4te**, **aclocal**, **autoreconf** tools give errors. 

```
(chroot wheezy_deb) root@pop-os:/projects/nut# ./autogen.sh
Regenerating Augeas ups.conf lens with 'python'...
Regenerating the USB helper files...
Calling autoreconf...
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal -I m4
/usr/bin/m4:configure.ac:89: Warning: too few arguments to builtin `ifdef'
autom4te: /usr/bin/m4 failed with exit status: 1
aclocal: /usr/bin/autom4te failed with exit status: 1
autoreconf: aclocal failed with exit status: 1
FAILED: did not generate an executable configure script!

# Note: on some systems "autoreconf", "automake" et al are dispatcher
# scripts, and need you to explicitly say which version you want, e.g.
#    export AUTOCONF_VERSION=2.65 AUTOMAKE_VERSION=1.13
# If you get issues with AC_DISABLE_STATIC make sure you have libtool.

(chroot wheezy_deb) root@pop-os:/projects/nut# m4 --version
m4 (GNU M4) 1.4.16
Copyright (C) 2011 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>.
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Rene' Seindal.

(chroot wheezy_deb) root@pop-os:/projects/nut# autom4te --version
autom4te (GNU Autoconf) 2.69
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+/Autoconf: GNU GPL version 3 or later
<http://gnu.org/licenses/gpl.html>, <http://gnu.org/licenses/exceptions.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

(chroot wheezy_deb) root@pop-os:/projects/nut# aclocal --version
aclocal (GNU automake) 1.11.6
Copyright (C) 2011 Free Software Foundation, Inc.
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Tom Tromey <tromey@redhat.com>
       and Alexandre Duret-Lutz <adl@gnu.org>.
       
(chroot wheezy_deb) root@pop-os:/projects/nut# autoreconf --version
autoreconf (GNU Autoconf) 2.69
Copyright (C) 2012 Free Software Foundation, Inc.
License GPLv3+/Autoconf: GNU GPL version 3 or later
<http://gnu.org/licenses/gpl.html>, <http://gnu.org/licenses/exceptions.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by David J. MacKenzie and Akim Demaille.       
```

When I try to generate the files on the target machine (armv7), the configuration(**configure**) file was created successfully and the binaries with the libraries were successfully created.

Debootstrap build command:
```
debootstrap --no-check-gpg \
--include vim,locales,gcc,build-essential,flex,bison,automake,\
gcc-multilib,libdbus-1-dev,libdbus-1-3,tree,file,qt4-linguist-tools,libudev-dev,\
ccache,time,python,curl,perl,make,autoconf,automake,libtool,\
valgrind,cppcheck,pkg-config --arch i386 wheezy ${CHROOT_DIR} ${MIRROR}
```

Thank you. If you need more information, I will share the it

jimklimov, 2023-03-30:
Those values are only relevant on some OSes, where `autoconf` in PATH is a shared script that dispatches to a particular implementation version. Which Debian version is yours?

I do not think cross-build selection should impact autogen.sh (that would be configure script parameters), so would investigate (poke more debugging verbosity into your copy) to check why it complains? Maybe it is not a standard version of something, but patched by distro, etc.

Otherwise for an example of one known working cross-build scenario, check linux+mingw builds for Windows, `scripts/Windows` that are part of NUT sources now.

idabsent, 2023-03-30:
> I do not think cross-build selection should impact autogen.sh (that would be configure script parameters), so would investigate (poke more debugging verbosity into your copy) to check why it complains? Maybe it is not a standard version of something, but patched by distro, etc.

Do I understand correctly that you need debug verbose output of some executable program? If it's correct, tell me the name of the program and its option to include details (only if the option name is anything other than **-v** or **--verobose**). Trying to include **-v** or **--versobe** for autogen.sh does not show any other information.

> Which Debian version is yours?

Wheezy is name for Debian 7

jimklimov, 2023-03-30:
Not for autogen itself (good idea though), but for tools it calls.

I suppose the culprit is 
````
/usr/bin/m4:configure.ac:89: Warning: too few arguments to builtin `ifdef'
````
but so far have little idea why it decides that.

jimklimov, 2023-03-30:
Containers can probably be a decent way, NUT CI uses some for non-native architectures - see https://github.com/networkupstools/nut/blob/master/docs/ci-farm-lxc-setup.txt and keep in mind https://github.com/networkupstools/nut/blob/master/docs/config-prereqs.txt

FWIW I've also started installation of a Debian 7 i386 container, this puzzle piqued my interest. But if there is no low-hanging fruit like a missed dependency package in your build host, an ARM container (e.g. libvirt/lxc + qemu-static) may be a faster solution for you.

jimklimov, 2023-03-30:
Also, the comment at `configure:89` suggests the line may be commented away with little loss:
````
dnl Default to `configure --enable-silent-rules` or `make V=1` for details?
dnl This feature seems to require automake-1.13 or newer (1.11+ by other info)
dnl On very old systems can comment it away with little loss (then automake-1.10
dnl is known to suffice):
ifdef(`AM_SILENT_RULES', `AM_SILENT_RULES([yes])')
dnl, `AC_MSG_NOTICE([Silent Rules feature not defined in this automake version, skipped])' )
````

I wonder if the commented-away third argument (on next line) is the missing bit for the tooling in Wheezy?

jimklimov, 2023-03-30:
Note there also seem to be issues with C++11 intermediate support (gcc-4.6 on Wheezy) , you may want to disable it in configure options.

jimklimov, 2023-03-31:
Checked to not enable C++11 on Debian 7 builds (despite its GCC claiming to support the language version by macro) anymore, and to still enable it with newer environments by default.

jimklimov, 2023-03-31:
By the way, if you move forward with a cross-build, it would be nice of you to post a PR to update NUT documentation :)

Probably `config-prereq.txt` with more detail on installing packages needed for e.g. ARM build on x86 platform - there is a bit of "multilib" hints in the Debian section already, but not too evolved... And parameters to `configure` script for the build itself.

I guess currently the mingw notes for Windows builds on Linux do a better job in this direction, although there many components have to be custom-built.

idabsent, 2023-04-01:
> By the way, if you move forward with a cross-build, it would be nice of you to post a PR to update NUT documentation :)

Okay. When I solve the problem, I'll see what can be added to the documentation

jimklimov, 2023-06-04:
Of the PRs above, I hope that https://github.com/networkupstools/nut/pull/1888/files#diff-49473dca262eeab3b4a43002adb08b4db31020d190caaad1594b47f1d5daa810R89-R90 in particular fixes the situation about `configure` script with both older and newer platforms.

Can you please check if this issue is now fixed (you original build approach succeeds) with current master-branch NUT code?

idabsent, 2023-06-06:
I restarted the configuration creation script after cloning master branch. The error has changed
```
(chroot wheezy_deb) root@pop-os:/projects/nut# ./autogen.sh 
Regenerating Augeas ups.conf lens with 'python'...
Regenerating the USB helper files...
Calling autoreconf...
autoreconf: Entering directory `.'
autoreconf: configure.ac: not using Gettext
autoreconf: running: aclocal -I m4
autoreconf: configure.ac: tracing
autoreconf: configure.ac: not using Libtool
autoreconf: running: /usr/bin/autoconf
configure.ac:2580: error: possibly undefined macro: AC_DISABLE_STATIC
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
autoreconf: /usr/bin/autoconf failed with exit status: 1
FAILED: did not generate an executable configure script!
```

jimklimov, 2023-06-06:
Ouch. The two lines about `AC_DISABLE_STATIC` did not change since 2010, and line 2580 is a comment :\

Probably that line can be reworded, and the use of the macro on line 2634 safe-guarded by `m4_ifdef` like the `AM_SILENT_RULES` precedent earlier, but that feels very odd now. As a guess, do you have `libtool(-dev?)` installed there?

idabsent, 2023-06-06:
Waw! Yes. The installed package `libtool` solved the problem. Thank you friend. Can i close issue?

jimklimov, 2023-06-06:
If the build all works now (and runs on target HW) - yes. Congrats!

closed 2023-07-03 by idabsent