networkupstools/nut #1890: APC Back-UPS RS 1000MS

chriseivert, 2023-04-01:
Hey all! I followed the guide [https://www.kreaweb.be/diy-home-server-2021-software-proxmox-ups/#3_DIY_HOME_SERVER_-_PROXMOX_-_NUT_Installation](url) and it seems like the ups won't play nice with the proxmox server I'm running. Simply copy/paste all of the configs over, changing usernames/passwords. In testing, it will shutdown ~30 seconds after removing power from the ups like it's ignoring upssched-cmd maybe? It certainly is ignoring something.
It then shuts down 210 seconds later, as I was messing with final delay. which didn't help anything. Still shutdown. I suppose the issue lies within a script somewhere, I'm just scratching my head because I don't know where to check to ensure proper functionality.
Below are my config files. If there is anything else required, please let me know.

**nut.conf**
`MODE=netserver`

**ups.conf**
```
pollinterval = 15
maxretry = 3

offdelay = 120
ondelay = 240

[apc]
# APC
driver = usbhid-ups
port = auto
desc = "APC"
```
**upsd.conf**
```
LISTEN 0.0.0.0 3493
LISTEN :: 3493
```
**upsd.users**
```
[upsadmin]
# Administrative user
password = *********
# Allow changing values of certain variables in the UPS.
actions = SET
# Allow setting the "Forced Shutdown" flag in the UPS.
actions = FSD
# Allow all instant commands
instcmds = ALL
upsmon master
```
**upsmon.conf**
```
RUN_AS_USER root
MONITOR apc@localhost 1 upsadmin ************ master

MINSUPPLIES 1
SHUTDOWNCMD "/sbin/shutdown -h"
NOTIFYCMD /usr/sbin/upssched
POLLFREQ 4
POLLFREQALERT 2
HOSTSYNC 15
DEADTIME 24
MAXAGE 24
POWERDOWNFLAG /etc/killpower

NOTIFYMSG ONLINE "UPS %s on line power"
NOTIFYMSG ONBATT "UPS %s on battery"
NOTIFYMSG LOWBATT "UPS %s battary is low"
NOTIFYMSG FSD "UPS %s: forced shutdown in progress"
NOTIFYMSG COMMOK "Communications with UPS %s established"
NOTIFYMSG COMMBAD "Communications with UPS %s lost"
NOTIFYMSG SHUTDOWN "Auto logout and shutdown proceeding"
NOTIFYMSG REPLBATT "UPS %s battery needs to be replaced"
NOTIFYMSG NOCOMM "UPS %s is unavailable"
NOTIFYMSG NOPARENT "upsmon parent process died - shutdown impossible"

NOTIFYFLAG ONLINE   SYSLOG+WALL+EXEC
NOTIFYFLAG ONBATT   SYSLOG+WALL+EXEC
NOTIFYFLAG LOWBATT  SYSLOG+WALL+EXEC
NOTIFYFLAG FSD      SYSLOG+WALL+EXEC
NOTIFYFLAG COMMOK   SYSLOG+WALL+EXEC
NOTIFYFLAG COMMBAD  SYSLOG+WALL+EXEC
NOTIFYFLAG SHUTDOWN SYSLOG+WALL+EXEC
NOTIFYFLAG REPLBATT SYSLOG+WALL
NOTIFYFLAG NOCOMM   SYSLOG+WALL+EXEC
NOTIFYFLAG NOPARENT SYSLOG+WALL

RBWARNTIME 43200
NOCOMMWARNTIME 600

FINALDELAY 210
```
**upssched-cmd**
```
#!/bin/sh
 case $1 in
       onbatt)
          logger -t upssched-cmd "UPS running on battery"
          ;;
       earlyshutdown)
          logger -t upssched-cmd "UPS on battery too long, early shutdown"
          /usr/sbin/upsmon -c fsd
          ;;
       shutdowncritical)
          logger -t upssched-cmd "UPS on battery critical, forced shutdown"
          /usr/sbin/upsmon -c fsd
          ;;
       upsgone)
          logger -t upssched-cmd "UPS has been gone too long, can't reach"
          ;;
       *)
          logger -t upssched-cmd "Unrecognized command: $1"
          ;;
 esac
```
**upssched.conf**
```
CMDSCRIPT /etc/nut/upssched-cmd
PIPEFN /etc/nut/upssched.pipe
LOCKFN /etc/nut/upssched.lock

AT ONBATT * START-TIMER onbatt 30
AT ONLINE * CANCEL-TIMER onbatt online
AT ONBATT * START-TIMER earlyshutdown 30
AT LOWBATT * EXECUTE onbatt
AT COMMBAD * START-TIMER commbad 30
AT COMMOK * CANCEL-TIMER commbad commok
AT NOCOMM * EXECUTE commbad
AT SHUTDOWN * EXECUTE powerdown
AT SHUTDOWN * EXECUTE powerdown
```
**/var/syslog**
```
Mar 31 19:30:01 proxmox systemd[1]: Started Network UPS Tools - power device driver controller.
Mar 31 19:30:03 proxmox upsd[175587]: Connected to UPS [apc]: usbhid-ups-apc
Mar 31 19:32:47 proxmox upsmon[175599]: UPS apc@localhost on battery
Mar 31 19:32:47 proxmox upssched[268112]: Timer daemon started
Mar 31 19:32:48 proxmox upssched[268112]: New timer: onbatt (30 seconds)
Mar 31 19:32:48 proxmox upssched[268112]: New timer: earlyshutdown (30 seconds)
Mar 31 19:33:18 proxmox upssched[268112]: Event: onbatt 
**Mar 31 19:33:18 proxmox upssched-cmd: UPS running on battery**
Mar 31 19:33:18 proxmox upssched[268112]: Event: earlyshutdown 
**Mar 31 19:33:18 proxmox upssched-cmd: UPS on battery too long, early shutdown**
Mar 31 19:33:18 proxmox upsmon[175599]: Signal 10: User requested FSD
Mar 31 19:33:18 proxmox upsd[175587]: Client upsadmin@::ffff:127.0.0.1 set FSD on UPS [apc]
Mar 31 19:33:18 proxmox upsmon[175599]: Executing automatic power-fail shutdown
**Mar 31 19:33:18 proxmox upsmon[175599]: Auto logout and shutdown proceeding**
Mar 31 19:33:18 proxmox upssched[269603]: Executing command: powerdown
Mar 31 19:33:18 proxmox upssched-cmd: Unrecognized command: powerdown
Mar 31 19:33:18 proxmox upssched[269603]: Executing command: powerdown
Mar 31 19:33:18 proxmox upssched-cmd: Unrecognized command: powerdown
Mar 31 19:33:33 proxmox upssched[268112]: Timer queue empty, exiting
Mar 31 19:34:50 proxmox usbhid-ups[259390]: Signal 15: exiting
```

jimklimov, 2023-04-01:
It looks like it is doing what you told it to:

This starts two timers with same 30-sec timeout:
````
AT ONBATT * START-TIMER onbatt 30
AT ONBATT * START-TIMER earlyshutdown 30
````

This asks to shutdown, twice:
````
AT SHUTDOWN * EXECUTE powerdown
AT SHUTDOWN * EXECUTE powerdown
````

The posted `upssched-cmd` does not have a case entry for `powerdown` so that fails. Probably it should ditto your SHUTDOWNCMD?

chriseivert, 2023-04-01:
> This starts two timers with same 30-sec timeout:
So where does the communication with the ups come in? What is the point of this upssched.conf script? If it's starting these timers, and runs the onbatt and earlyshutdown messages for 30 seconds (is that how it works?) what is the point of communicating with UPS at all? It should run until it's been on battery for minimum 15 minutes OR the battery is at 15% as in the linked instructions (which I set)

```
We will set the parameter for minimum runtime to 15 minutes and the minimum charge to 40% :

upsrw -s battery.runtime.low=900 apc@localhost 
upsrw -s battery.charge.low=40 apc@localhost
```

chriseivert, 2023-04-01:
Is there any way of tracing what made this action? Not sure where the full shutdown command is ran and how that works

> Mar 31 19:33:18 proxmox upsmon[175599]: Signal 10: User requested FSD

jimklimov, 2023-04-01:
Did you read NUT's documentation and not a derivative excerpt of "this worked for me" style?

In short, NUT has 3 layers - drivers (media and vendor protocol), data server (upsd, common data model) and clients for different purposes (upsc, upsmon, half a dozen more just in the project). The upssched is one of several ways upsmon can do its job, by starting a timer to absorb small outages as irrelevant. Without it, upsmon effectively relies on UPS to say "all is lost, abandon ship!" which not all UPS models can physically do.

jimklimov, 2023-04-01:
For tracing you can enable debug verbosity. 

With NUT 2.8.0+ (e.g. a custom build from master branch) you can use `debug_min` options in config files for the daemons. 

For older (2.7.4-) packages you'd edit systemd units or init scripts to add `-D` and possibly fiddle with backgrounding or unit type (debug means foregrounded).

Or just stop persistently defined services and start the parts manually from command line with debug and foregrounding in several terminals, as you experiment.

chriseivert, 2023-04-04:
awe man!!! Just reading the manual made it so much easier... Goes to show just following a guide is NOT the answer 100% of the time. READ THE MANUAL

closed 2023-04-04 by chriseivert

VeryFunnyMonkey, 2023-06-06:
@chriseivert did you still have this config? I unfortunately am struggling to set NUT up, seeing your configs will help a lot. Thanks :)